jsk-ros-pkg/jsk_demos #140: jsk_maps cannot build with catkin build?

garaemon, 2015-01-26:
1. `catkin build jsk_maps` works
2. `catkind build` failed on building jsk_maps
```
[jsk_maps] ==> '/home/leus/ros/hydro/build/jsk_maps/build_env.sh /usr/bin/make -j40 -l40' in '/home/leus/ros/hydro/build/jsk_maps'
[100%] Generating /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng2-2f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng2-7f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng6-1f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng6-3f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng6-b1f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng8-6f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng2-1f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng2-3f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng2-8f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng6-2f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng6-4f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng8-1f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps, eng8-b1f-0.05.yaml, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/launch/start_map_eng2_main.launch, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/launch/start_map_eng6_main.launch, /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/launch/start_map_eng8_main.launch
configuring by "/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0.0
/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  too many streams for selection in (init-xwindow)X:E: make[2]: *** [/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_demos/jsk_maps/raw_maps/eng2-2f-0.05.yaml] 
make[1]: *** [CMakeFiles/jsk_maps_install.dir/all] 
make: *** [all] 
[jsk_maps] <== '/home/leus/ros/hydro/build/jsk_maps/build_env.sh /usr/bin/make -j40 -l40' failed with return code '-2'

```

k-okada, 2015-01-26:
it works on my computer, does irteusgl work on same configuration?

garaemon, 2015-01-26:
irteusgl works, only parallel compilation is problem

garaemon, 2015-02-11:
solved

closed 2015-02-11 by garaemon