jsk-ros-pkg/jsk_demos #155: git clone is heavy

garaemon, 2015-02-10:
it takes 22 minutes to clone
```
$ time git clone https://github.com/jsk-ros-pkg/jsk_demos.git
Cloning into 'jsk_demos'...
remote: Counting objects: 7667, done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 7667 (delta 13), reused 0 (delta 0)
Receiving objects: 100% (7667/7667), 50.30 MiB | 22 KiB/s, done.
Resolving deltas: 100% (5285/5285), done.

real	22m28.156s
user	0m6.976s
sys	0m2.651s
```

k-okada, 2015-02-10:
ok, I'll cleanup old heavy files, do not send PR or commit until I say ok

k-okada, 2015-02-10:
ok, it ok, 
here is what I did, and if you find any trouble, please let me know asap, I'll keep back under /tmp directory so if my laptop reboot, all of data is gone.
``` 
2010  java -jar ~/Downloads/bfg-1.12.1.jar --delete-files '*.{bag,pgm,jpg}' .
 2011  git reflog expire --expire=now --all && git gc --prune=now --aggressive
 2012  git push -f origin master 
```

garaemon, 2015-02-10:
thanks!

✉︎ Ryohei Ueda

closed 2015-02-11 by garaemon