jsk-ros-pkg/jsk_demos #183: drc_task_common/operator_station_main.launchがエラー無く立ち上がるようにする

garaemon, 2015-02-13:
@mmurooka @YuOhara 新規の環境でやると真っ赤になるので対応お願いします．

なにか特殊な設定などがある場合はREADMEとかに設定方法を書いておいてください

mmurooka, 2015-02-13:
どのPCで何を打ったか再現方法を教えていただけますでしょうか，確認してみます．

garaemon, 2015-02-13:
sagamiです。再現方法はサラからjsk.rosbuildでインストールして、.desktopを実行です

✉︎ Ryohei Ueda

YuOhara, 2015-02-13:
はい。
#182 とかも、多分その一部が問題だったんですよね・・・
多分python のライブラリ(fysomなど)のinstallとかも必要な気がします。

garaemon, 2015-02-13:
はい、その通りです。fysomはpipかな、だとしたらrosdistroにリクエストを出して、、、というステップがあるけどそこまでやってとは言わないので、何か特殊なことが必要かをまずリストアップしましょう。

✉︎ Ryohei Ueda

mmurooka, 2015-02-14:
https://github.com/jsk-ros-pkg/jsk_demos/pull/184 https://github.com/jsk-ros-pkg/jsk_demos/pull/185 https://github.com/jsk-ros-pkg/jsk_demos/pull/186 で、インストール必要なパッケージの説明、エラーの原因の説明、細かいエラーをなくす修正をしました。

k-okada, 2015-02-14:
> fysomはpipかな、だとしたらrosdistroにリクエストを出して、、、というステップがあるけどそこまでやってとは言わないので

僕はそこまでやろう，と言いたいところです．ものすごく簡単ですよ．
https://github.com/ros/rosdistro/pull/6993/files

mmurooka, 2015-02-14:
そこまでやってみました． https://github.com/ros/rosdistro/pull/7262

garaemon, 2015-02-16:
今日中に計算機を分けて動くようにしましょう。

第一ステップ
1. /tfと/joint_statesをremap
2. 別マシンでやる
  1. robot_state_publisherが動いていない問題 -> 植田

garaemon, 2015-02-18:
できた @ sagami

closed 2015-02-18 by garaemon