jsk-ros-pkg/jsk_demos #465: ar_poseの依存関係

orikuma, 2015-04-16:
srvにdrc_task_common::SetValueが追加されたのでdrc_task_commonをbuildし直そうとすると,
ar_poseの依存関係が解決できずにビルドが失敗します.

rosdep installすると以下でだめそうだったので
```bash
$ rosdep install drc_task_common
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
drc_task_common: Cannot locate rosdep definition for [ar_pose]
```
https://github.com/ar-tools/ar_tools.git からsrcを落としてきてビルドしようとしたのですが,
以下のエラーでartoolkitのビルドが通りません.
```bash
iori@iori-x1c:~/ros/hydro/src/ar_tools/ar_pose
$ catkin bt
==> Expanding alias 'bt' from 'catkin bt' to 'catkin b --this'
==> Expanding alias 'b' from 'catkin b --this' to 'catkin build --this'
-----------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/iori/ros/hydro_parent/devel:/opt/ros/hydro
Workspace:                   /home/iori/ros/hydro
Source Space:       [exists] /home/iori/ros/hydro/src
Build Space:        [exists] /home/iori/ros/hydro/build
Devel Space:        [exists] /home/iori/ros/hydro/devel
Install Space:     [missing] /home/iori/ros/hydro/install
DESTDIR:                     None
-----------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-----------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
-----------------------------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------------------------
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
Found '298' packages in 0.4 seconds.
Starting ==> artoolkit

[artoolkit] ==> '/home/iori/ros/hydro/build/artoolkit/build_env.sh /usr/bin/make -j4 -l4' in '/home/iori/ros/hydro/build/artoolkit'
[ 12%] Performing download step (download, verify and extract) for 'ARToolkit'
-- downloading...
     src='http://csc.kth.se/~fevb/downloads/ARToolKit-SVN-r311.tar.gz'
     dst='/home/iori/ros/hydro/build/artoolkit/ARToolkit-prefix/src/ARToolKit-SVN-r311.tar.gz'
     timeout='none'
CMake Error at ARToolkit-stamp/download-ARToolkit.cmake:6 (file):
  file DOWNLOAD MD5 mismatch

    for file: [/home/iori/ros/hydro/build/artoolkit/ARToolkit-prefix/src/ARToolKit-SVN-r311.tar.gz]
      expected MD5 sum: [2f3537f782b5506133bcc79f2daba3bb]
        actual MD5 sum: [d41d8cd98f00b204e9800998ecf8427e]

make[2]: *** [ARToolkit-prefix/src/ARToolkit-stamp/ARToolkit-download]  1
make[1]: *** [CMakeFiles/ARToolkit.dir/all]  2
make: *** [all]  2
[artoolkit] <== '/home/iori/ros/hydro/build/artoolkit/build_env.sh /usr/bin/make -j4 -l4' failed with return code '2'
                                                                                                                                                                                                                                                                                                                              
Failed   <== artoolkit [ 2 minutes and 7.5 seconds ]
[build] There were '1' errors:

Failed to build package 'artoolkit' because the following command:

# Command to reproduce:
cd /home/iori/ros/hydro/build/artoolkit && /home/iori/ros/hydro/build/artoolkit/build_env.sh /usr/bin/make -j4 -l4; cd -

# Path to log:
cat /home/iori/ros/hydro/build/build_logs/artoolkit.log

Exited with return code: 2

[build] Runtime: 2 minutes and 8.6 seconds
```
ar_poseが追加されたのは4/8のようですが, それ以降どのようにビルドを通しているのでしょうか.

garaemon, 2015-04-17:
1. ar_toolsのコンパイルは普通に通るはず。今は一部サイトが落ちてる
2. ar_poseがなくてもdrc_task_commonはコンパイルできるはず

garaemon, 2015-04-17:
ar_toolsにプルリクエスト出しておいたので、このブランチ使えばコンパイルできます

https://github.com/ar-tools/ar_tools/pull/14

closed 2015-04-18 by garaemon