jsk-ros-pkg/jsk_demos #466: drc_task_commonのroseus-add-srvsについて

orikuma, 2015-04-17:
roseusから
```lisp
(ros::roseus-add-srvs "drc_task_common")
```
すると以下のエラーが出ます.
```lisp
1.irteusgl$ (ros::roseus-add-srvs "drc_task_common")
/home/iori/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package JSK_RECOGNITION_MSGS in (apply #'load-org-for-ros fullname args)
```
```bash
rosrun roseus generate-all-msg-srv.sh jsk_recognition_msgs
```
はしていて, ~/ros/hydro/devel/share/roseus/ros/drc_task_common/manifest.lは存在しています.
この状態で以下をするとjsk_interactive_markerで同様のエラーが出て,
```lisp
2.E1-ROS:irteusgl$ (ros::load-ros-package "jsk_recognition_msgs")
nil
3.E1-ROS:irteusgl$ (ros::roseus-add-srvs "drc_task_common")
/home/iori/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package JSK_INTERACTIVE_MARKER in (apply #'load-org-for-ros fullname args)
```
再度以下をするとエラーが出なくなります.
```lisp
4.E2-ROS:irteusgl$ (ros::load-ros-package "jsk_interactive_marker")
nil
5.E2-ROS:irteusgl$ (ros::roseus-add-srvs "drc_task_common")
nil
```
依存関係の順序があるのでしょうか?

k-okada, 2015-04-17:
> roseusから
>
> (ros::roseus-add-srvs "drc_task_common")
>
> すると以下のエラーが出ます.
>
> 1.irteusgl$ (ros::roseus-add-srvs "drc_task_common")
> /home/iori/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package JSK_RECOGNITION_MSGS in (apply #'load-org-for-ros fullname args)
>
> rosrun roseus generate-all-msg-srv.sh jsk_recognition_msgs
>
> はしていて,
> ~/ros/hydro/devel/share/roseus/ros/drc_task_common/manifest.lは存在しています.
> この状態で以下をするとjsk_interactive_markerで同様のエラーが出て,
>
> 2.E1-ROS:irteusgl$ (ros::load-ros-package "jsk_recognition_msgs")nil3.E1-ROS:irteusgl$ (ros::roseus-add-srvs "drc_task_common")
> /home/iori/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package JSK_INTERACTIVE_MARKER in (apply #'load-org-for-ros fullname args)
>
> 再度以下をするとエラーが出なくなります.
>
> 4.E2-ROS:irteusgl$ (ros::load-ros-package "jsk_interactive_marker")nil5.E2-ROS:irteusgl$ (ros::roseus-add-srvs "drc_task_common")nil
>
> 依存関係の順序があるのでしょうか?

(ros::roseus-add-srvs "drc_task_common")
を使っチエルコードの前に
(ros::load-ros-package "jsk_interactive_marker")
を追加しておいて下さい．

orikuma, 2015-04-22:
他のパッケージを追加しないといけない仕様とのことだったので
(ros::roseus-add-msgs "jsk_recognition_msgs")
(ros::roseus-add-msgs "jsk_interactive_marker")
を追加して解決しました

closed 2015-04-22 by orikuma

k-okada, 2015-04-22:
FYI, この問題ですが，geneus 2.2.0で和田くんと直したはずです．
https://github.com/jsk-ros-pkg/geneus/pull/18
hydroだともうshadow-fixedでつかえると思います．