jsk-ros-pkg/jsk_demos #544: OCSのrqtのGUIがときどきフリーズする

mmurooka, 2015-04-30:
OCSのrqtのGUIがときどき30秒くらいグレーになってかたまり自然に復帰します．
復帰するときに以下のエラーが出ています．

画像のtopicを受け取れずタイムアウトしている？
別PCでmultisense_remoteを上げているのでこれが原因？
（ただ，昨日までも別PCであげていて起きていなかった）

```
Traceback (most recent call last):
  File "/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_visualization/jsk_rqt_plugins/src/jsk_rqt_plugins/image_view2_wrapper.py", line 134, in updateTopics
    for (topic, topic_type) in rospy.get_published_topics():
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/client.py", line 370, in get_published_topics
    code, msg, val = get_master().getPublishedTopics(namespace)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/msproxy.py", line 101, in wrappedF
    return f(*args, **kwds)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1224, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1578, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1264, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1292, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1439, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 954, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 814, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 776, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 757, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
```

garaemon, 2015-04-30:
了解、見ます

garaemon, 2015-04-30:
solved but it's a problem of rosmaster

closed 2015-04-30 by garaemon