jsk-ros-pkg/jsk_demos #669: gazebode

mmurooka, 2015-05-14:


k-okada, 2015-05-14:
創造だけど両方のマシン(roscore)で時間をわわせて、あとuse_sum_timeも伝搬させないと行けないとか？

garaemon, 2015-05-14:
どちらも同じマシンを使っています．

もしかしたら外部からshutdown requestがきて，timeoutになっているだけかも

garaemon, 2015-05-14:
とおもったら一部transvmをmachineタグで指定しているのがダメかも

garaemon, 2015-05-14:
と思いましたがきのせいでした

garaemon, 2015-05-14:
https://github.com/ros/ros_comm/blame/indigo-devel/clients/rospy/src/rospy/impl/masterslave.py#L439

rospyがtimeoutを指定していますね

k-okada, 2015-05-14:
roscoreが2つあって片方でuse_sim_timeで片方が違うとだめな気がします。あるいは両方のuse_sim_timeだと/clockを伝搬させないとだめじゃないかな。

こちらか他の人でも再現できることなら投げておいて現地の人は実機でしかできないことに注力できるといいですんr。

garaemon, 2015-05-14:
@mmurooka https://github.com/jsk-ros-pkg/jsk_common/pull/925 これで治ったはずです。

> こちらか他の人でも再現できることなら投げておいて現地の人は実機でしかできないことに注力できるといいですんr。
はい、もちろんそうしたいです

closed 2015-05-14 by garaemon

mmurooka, 2015-05-15:
ありがとうございました。

実機はかなり人気者で使える時間が限られているので
プログラムをひと通り回して、タイポとかがないことを事前に確認しておくのに使おうと思います。
本番直前にプログラムを書き換えて、実機テストはできないまま本番みたいなときにも、
さらっとシミュレーション上で確認できたら安心です。