jsk-ros-pkg/jsk_demos #869: DRC stairのやり方とトラブルシューティング

yuki-asano, 2015-05-28:
黒岩くんからissueあげとくといいんじゃないですか、
ということなので、メモを書いておきます。

<階段>
階段のぎりぎり(1cmくらい)に立たせる

・door through pose

・別guiで前かがみ姿勢
start stair
go stair pose

・階段を上る
フットステップを，いい感じのところに移動する(最大はx300mm, z 250mm)
高さの調整が必要で，ポイントクラウドの下面をかすめるくらい．上から見て，
フットステップ全体がなんとなく隠れるくらいが目安

rvizのmultisenseは
/ocs/communication/laser_cloud
に合わせる

・終わったあと
黒岩が作ってくれた
asano-san.lをロード
(setup-asano-san) ;;jaxonとかのロード
(owatta-asano-san) ;;階段の上で，膝を曲げて，クレーンで釣れるようにする
(kaishi-asano-san) ;;階段をはじめる姿勢になる

<トラブルシューティング>
とりあえず，
rosrun jaxon_ros_bridge check_jaxon_sanity.py 
か
rosrun drc_task_common check_drc_task_sanity.py
を実行して，何が落ちているか確認．
出てきたコマンドをそのまま実行すればok

・multisenseが見えない

jaxonvisionで
multisense_local的なやつを実行

local pcで
multisense_remote的なやつを実行

・multisenseのフラッシュがguiから操作できない
xボタンで消して，rpt_reconfigureを再起動
rosrun rqt_reconfigure rqt_reconfigure

snozawa, 2015-05-28:
メモ
```
cd ~/ros/hydro_parent/src/hrpsys/ && git pull && roscd hrpsys &&
catkin bt --start-with-this --force-cmake && roscd hrpsys_ros_bridge
&& git pull && catkin bt --start-with-this --force-cmake
```

garaemon, 2015-05-28:
こら本当に動きますか？
roscd hrpsysだと、develの方に行くけど大丈夫ですか？

snozawa, 2015-05-28:
develでcatkin buildがうごくようなのですが、
hydro_parentのものはどうするのが正解ですか？

k-okada, 2015-05-28:
source /opt/ros/hydro/setup.bash
cd ~/ros/hydro_parent
catkin

garaemon, 2015-05-29:
Updated

```lisp
(warn "(setup-asano-san)~%")
(defun setup-asano-san ()
  (load "package://hrpsys_ros_bridge_tutorials/euslisp/jaxon-interface.l")
  (load "package://drc_task_common/euslisp/test-drc-terrain-walk.l")
  (jaxon-init)
  (setq *robot* *jaxon*))

(warn "(owatta-asano-san)~%")
(defun owatta-asano-san ()
  (walking-pose *robot* :root-link-pitch-offset 0 :root-link-height-offset -300 :default-pose-method :narrow-width-pose)
  (send *ri* :angle-vector (send *robot* :angle-vector) 4000)
  (send *ri* :wait-interpolation))

(warn "(kaishi-asano-san)~%")
(defun kaishi-asano-san ()
  (send *robot* :narrow-width-pose)
  (send *ri* :angle-vector (send *robot* :angle-vector) 4000)
  (send *ri* :wait-interpolation))

```

eisoku9618, 2015-05-29:
updated

```bash
source /opt/ros/hydro/setup.bash && cd ~/ros/hydro_parent/src/hrpsys && git pull && catkin bt --start-with-this --force-cmake && source ~/ros/hydro/devel/setup.bash && roscd hrpsys_ros_bridge && git pull && catkin bt --start-with-this --force-cmake && roscd jsk_pcl_ros && git pull && roscd drc_task_common && git pull && catkin bt --force-cmake
```

からの

```bash
roscd jsk_models
git pull
roscd hrpsys_ros_bridge_tutorials
git clean -xfd
catkin bt --start-with-this --force-cmake --make-args -j1
```

eisoku9618, 2015-05-29:
JAXON_RED体内でhrpsys_ros_bridge_tutorialsのbuildが通らなくて，ログを見たらmin-maxで引っかかっていて，roscd euslispしたら /optにいきました．

ので，aptで入っていたjaxon_redのeuslisp / jskeus などをpurgeしておきました．

closed 2015-07-23 by garaemon