jsk-ros-pkg/jsk_demos #934: DRC task 一気通貫手順

yuki-asano, 2015-06-03:
# 運転
熊谷、寺沢

野田にチェンジ

# 降車

室岡にチェンジ

# ドア
ビークル消す
ドアを探す（右回り90度回転）
ドア前に進む
ドアノブ確認（カメラ下げる）
ドア開け
後ろに下がる
ドアスルーポーズ
ドアの途中まで進む
ドアの最後まで進む
ドアから離れてリセットマニプポーズ

# バルブ
バルブの初期位置の確認
バルブ終了後にバルブから多めに下がる。

小原にチェンジ

# ドリル
首下げてドリル位置確認
ドリルグラスプ
ドリルマーカーの位置合わせ（奥行き、反転）
ドリルウォッチ
ドリルマーカーの位置合わせ（回転）

植田さんにチェンジ

# テレイン
fixed frame -> Ground

# 階段
/ocs/communication/laser_cloudに変更
StairMarkerをつける
階段の目の前まで移動
ドアスルーポーズ
微調整
ゴーステアポーズ(かがむ姿勢)
階段がすねで隠れない程度まで前進
垣内さんの（アールビスなんちゃらが出たらオーケー） -> 登る(1)
垣内さんの -> 登る(2)
垣内さんの -> 登る(3)
垣内さんの -> 登る(4)
垣内さんの -> エンドステアポーズ

# 復帰手順
オートバランサー
リムーブフォースオフセット
エス・ティー
地面につける
現地の人に揺すって確認してもらう
ゴーポス（000）

eisoku9618, 2015-06-04:
[image]

yuki-asano, 2015-06-04:
唐沢君(他、フィールドの方々)

開始/再開時、バッテリ切れ時は、
今のところ、上のチェックリストの手順で確認を行っていく予定です。
印刷する予定ですが、確認よろしくお願いします。

yuki-asano, 2015-06-05:
開始手順を、T-xにしたがって並び替えました。

# T-45 ～ T-20
- ロボットを自動車に載せて準備完了にする必要がある
## 開始準備
### フィールド
- hrpsysの確認
   - [ ] 基板確認(5, 12, 24V入れる)
   - [ ] リレーon
   - [ ] キャリブジョイント
   - [ ] リレーoff
   - [ ] サーボon
   - [ ] サーボoff
   - [ ] リレーon
   - [ ] servo with reset pose
   - [ ] リセットポーズの確認
   - [ ] abcをいれる
   - [ ] リムーブフォースオフセット
   - [ ] start-defaultを全部する(ST, impedance)
   - [ ] 空中go-pos
   - [ ] 地面にロボットを下ろす
   - [ ] ロボットを揺すって確認(st)
   - [ ] ロボットを吊る
   - [ ] abc, ST, impedanceを切る
   - [ ] 運転時の姿勢を送る
   - [ ] 運転時の姿勢になったら練習通りに車に乗せる

## オペレータ
- [ ] ros_bridgeの確認, fcプログラムの確認をする
- [ ] ocsプログラムの確認をする
- [ ] Egress用の野田GUIをあげる
- ocsの確認
   - [ ] hcfからのgo-pos(0 0 0)
   - [ ] ocsからのgo-pos(0, 0, 0)
   - [ ] リセットマニプポーズ
   - [ ] ハンドキャリブ
   - (fi)ハンドキャリブの終了(ハンド開閉)を確認
   - [ ] ハンド開閉動作確認
   - [ ] インピーダンス切る(手に触らないように)
   - [ ] リセットフォースオフセット(ハンド)
   - [ ] インピーダンス入れ直す
   - [ ] ocsから再度go-pos(0 0 0)
   - [ ] 再度リセットポーズ
   - [ ] 垣内さんのやつ
 - [ ] 熊谷、寺澤にチェンジ

# T-20 ～ T-5
### 運転準備
- 無線通信開始(unlimited)
    - [ ] joyを読めているかどうか確認する（USB挿しなおしてvehicleのところを立ち上げ直し）
   - [ ] ストップインピーダンス
   - [ ] reset force offset
   - [ ] rqt_reconfigureでmultisenseおよびprosilicaのexposureなどを調節をする
   - [ ] rvizの表示を運転用に変える(vehicleをつける、PointCloudを0.01->0.05、Laserを消す)
   - [ ] 左上１番目のCommunicationボタンをOn（赤->緑）にする
   - [ ] 左上２番目のRealボタンをOn（赤->緑）にする
- 運転用のテスト開始
   - [ ] 首を下に下げてマーカーを見る
   - [ ] もう一度multisenseのマーカーが見える程度にrqt_reconfigureを調節する->人の目でハンドル位置を確認
   - [ ] Correct Handle Poseボタンを押して、クランクを掴みにいく
   - [ ] neck, accel, handleモードをすべてoperationに切り替える
   - [ ] 左のジョイスティックでハンドルをゆっくり回してみてハンドルの様子を確認する（その際マーカーが動いていることも確認する）
   - (フィールド)アクセルの調整方法を係の人に説明する（【重要！！最新版FAQ参照】）
   - [ ] 右のジョイスティックで右足を動かして（RPMを参考にしながら）アクセルMAX(RPM2000くらい),MIN(ギリギリ踏まない位置)を調整する
   - (フィールド)右足を操作し始めたら、エンジンの回転数（RPM）の変化をオペレータに連絡する
   - (フィールド)もしアクセルを踏めていなかったら、さっとロボットを車に乗せ直して上記の操作をする

# T-5 - T2
- T-3で無線がlimitedに切り替わる
- [ ] バッテリに切り替える 
- [ ] ギアをhighに切り替える

# T-0
- 電話を切る
- 運転スタート

eisoku9618, 2015-06-05:
フィールドメンバーへのメモ

以下を 10:45 - 11:10 でやる．時間には余裕があるはず．

1. @t-karasawa 5V/12Vを入れてJAXONガチャをする
1. @t-karasawa 24Vを入れる
1. @t-karasawa リレーを入れる
1. @fsugai simple_shmで80Vが来ているか確認してからキャリブをする
1. 一息ついて一旦落ち着く **キャリブを忘れるとサーボONでピーッと鳴って焦るので忘れずに**
1. @fsugai ノートPCで体内PCにsshする．``roscd jaxon_ros_bridge/scripts && ipython -i jaxon_hrpsys.py ""``
1. @t-karasawa リレーを落とす
1. @fsugai ``hcf.startDRCInitiSequence(servo=True)``でもろもろを確認し， @orikuma ポーズになっている
1. みんなで車に載せる
1. 一息ついて一旦落ち着く
1. @t-karasawa 時間が余っていたら右足微調整の量を考えとく

### もろもろ

yesが聞かれるタイミングで分けると以下

1. リレーが切れている状態でサーボON/OFFでエラーリセット
1. リレーを入れてサーボON with reset-pose ．終わった瞬間に AutoBalancerが入る
1. @t-karasawa ロボットの腰を掴んで抑えた状態で， @fsugai が y を押してremove-force-sensor-offset がされる．終わった瞬間に，STとICが入る
1. @t-karasawa がロボットを立たせるのを待ってから， @fsugai がgo-posする
   - ガガッとなったらリレーを落としてC-cして ``hcf.servoOff()``をしてやり直す
1. @t-karasawa がロボットを吊るのを待ってから， @fsugai がst / abc / ic を全部切る
1. @fsugai が熊谷さん初期姿勢を送る

closed 2015-06-14 by garaemon