jsk-ros-pkg/jsk_demos #987: pr2 irtデモをgazeboで動かしたい

furushchev, 2015-07-02:
irtデモをｐｒ２実機だけではなくgazeboでも動かしてシミュレーション環境を整えたいです。

- [x] pr2体内でなくても動かせるようにする => https://github.com/jsk-ros-pkg/jsk_demos/pull/986
- [ ] `jsk_2013_04_pr2_610`にworldファイルと６１０環境のurdfを置く
- [ ] gazebo用のデモlaunchファイルを作る
- [ ] readme更新

の順に作業をしていこうと思います。どうでしょうか？

k-okada, 2015-07-02:
いいとおもいます．まずは，pr2_gazeboをあげて（robot startした状態に成るはず）
で，jsk_pr2_startupが動くようにする，というのがベースですかね．これはできているのかな．

furushchev, 2015-07-02:
なるほど、たしかにそうですね。
jsk_pr2_startupはおそらく`openni` vs `head_mount_kinect`問題が発生すると思われます。
見てみた感じsdformatにnamespaceの名前が直接書かれているので、spawn_modelするとそのnamespaceで出してしまうようです。(urdfを変える必要がある？）
jsk_pr2_startupとその依存をgazebo（というかopenni名前空間を使っているのはjskだけ？）に対応させるオプションを追加するのが早いでしょうか・・・？

garaemon, 2015-07-02:
できればjskのモデルを標準のものにしたいところです。

k-okada, 2015-07-02:
そうしましょう．今ならいいタイミングな気がします．

furushchev, 2015-07-02:
@garaemon 日本語の理解不足ですが、つまりjsk-ros-pkgをhead_mount_kinect名前空間に移行するというのでいいでしょうか？

garaemon, 2015-07-02:
実機で使ってるurdfのフレームと、kinect立ち上げのlaunchでnamespaceを変えるということです。
こんな時のためにjsk_topic_toolsにdeprecated_relayなるプログラムを用意してあります

garaemon, 2015-07-02:
手順的には

- [ ] /etc/robotのurdfを標準のframeにする
- [ ] kinectを立ち上げるlaunchのnamespaceを変える
- [ ] 今kinectを使っているlaunchやプログラムのtopicを変える
- [ ] frame_idを直接指定しているものは修正する
- [ ] `/openni`以下にはdeprecated_relayでrelayしておく(トピックを使うと大量のwarningが出るようになる)

ですかね。結構大変ですがキャリブとかが楽になって良いです。

furushchev, 2015-07-02:
@garaemon 詳しい手順をありがとうございます。ｐｒ１０１２で今から試験的に変えていこうと思います。

furushchev, 2015-07-02:
実機で上の２つを行い、openniを`/etc/ros/hydro/openni_head.launch`（デフォルト設定）に戻してみました。
すると、実機では`/kinect_head/rgb/~`になりました。
一方gazeboでは`/head_mount_kinect/rgb/~`になっているのですが、これはpr2_gazebo側を修正する方向ですか・・・？

garaemon, 2015-07-02:
pr2_gazeboに修正のPRを出しましょう。ですが却下される可能性も高いと思うので、とりあえず現状のpr2_gazeboでrelayするようなものをコチラ側に加えておく、というのでどうでしょうか。

furushchev, 2015-07-02:
どこを変えたらいいかわからなかったのでissueを出しました。
https://github.com/PR2/pr2_simulator/issues/111

garaemon, 2015-07-02:
では本家の方はそれで進めていおいて、辻褄が合うようにシミュレーションの方はrelayして行きましょう。

k-okada, 2015-07-02:
> どこを変えたらいいかわからなかったのでissueを出しました。
> PR2/pr2_simulator#111 <https://github.com/PR2/pr2_simulator/issues/111>

https://github.com/PR2/pr2_common/blob/indigo-devel/pr2_description/urdf/sensors/kinect_prosilica_camera.urdf.xacro#L88
では？

わかっていないけど，gazeboではこの部分でkinect情報が出てきて，
実機はurdfはno-kinectであとで，openni.launch をしているのでただしい？

furushchev, 2015-07-05:
https://github.com/jsk-ros-pkg/jsk_robot/pull/364 https://github.com/jsk-ros-pkg/jsk_smart_apps/pull/53
で jsk_pr2_startup/pr2.launchまでは動くようになったはずです。

k-okada, 2015-07-05:
> jsk-ros-pkg/jsk_robot#364
> jsk-ros-pkg/jsk_smart_apps#53
> で jsk_pr2_startup/pr2.launchまでは動くようになったはずです。

実機の状態はそれぞれどうなっている？

furushchev, 2015-07-05:
今はpr1012だけ、urdfをオリジナルに変えて、僕のユーザだけ上のPRの仕様にしています。
/etc/ros/distro/urdf/ 以下に
robot.xml -> robot_calibrated_2014_11_07_21_57.xml.org
robot.xml.jsk -> robot_calibrated_2014_11_07_21_57.jsk.xml
というシンボリックリンクがあり、
robot start ではrobot.xmlを呼ぶようにして有ります。
元の設定に戻すには
mv robot.xml robot.xml.org
mv robot.xml.jsk robot.xml
ですべて戻ると思います。

furushchev, 2015-07-06:
@k-okada 
> わかっていないけど，gazeboではこの部分でkinect情報が出てきて，
> 実機はurdfはno-kinectであとで，openni.launch をしているのでただしい？

僕もコード見ました。それで合ってると思っています。
現状relayで対応しましたが、xacroのmacroを書きなおせばrelayなしでできる気がするので、ひと通り動いたら作業しようと思います。

k-okada, 2015-07-06:
どの機体がどの状態で何が動くか皆で情報共有ちゃんとされているといいですね．研究会の後ぐらいに一同確認会をしますか．