jsk-ros-pkg/jsk_demos #998: DRC関係で、launchしたものが必ず起動する。落ちない。

snozawa, 2015-07-17:
field_computor.sh とocs.sh的なもので、起動したら100%あがる＋落ちないようにするのが必須な気がします。
思い当たるところがある人は、どんどんあげていってください。

- fc-execute?のEUslispのプログラムがおちる
- 点群がでてこないときがある

mmurooka, 2015-07-17:
- Rvizのロボット立ち位置モデルが出てこない
- バルブ，ドアなどの認識に失敗する
- nighthawkの通信が遅いことがある
- silverhammerのプロセスが残ったままのことがある

@snozawaさんが挙げたものも含めてすべて，DRC本番前には直したいとそれなりの時間と真面目さをかけましたが残ったままの問題です．

snozawa, 2015-07-18:
> バルブ，ドアなどの認識に失敗する

認識失敗はアルゴリズムをもっとよくするという方向もありえるけど、
立ち位置マーカがでな、点群がでない、落ちる系の不具合はもう一段階基礎的な部分の
不具合なので、直したいですね。

それぞれ、何がバグの原因と予想してどういうデバッグをしたか覚えているかな？

garaemon, 2015-07-18:
問題は、「subscribeされたらsubscribeする」というのが動かない時があるとこですね。

jsk_topic_toolsの問題です。

mmurooka, 2015-07-18:
> Rvizのロボット立ち位置モデルが出てこない

数回に1回くらいの割合で起きる？
立ち位置モデルのプログラムが起動する瞬間にrobot_descriptionパラメータが渡せていない．
@garaemon とlaunchのパラメータのnamespaceとかfurutaructiveのプログラムとかを見返してそれらしいところを変えたが直りませんでした．．
もう一度頑張るしかないとは思います．
roscoreを分けていないときにも起こる．

> fc-execute?のEUslispのプログラムがおちる

1日実験していると数回くらい起きます．
キャンセルボタンを押したときに途中で関数から抜けられるように，catch, throwみたいなことをたくさんしていてそれがいけないかもしれない．
gdbで見れたことはないので見れたら何かわかるかもしれないです．

garaemon, 2015-07-18:
たしかgdbかけると再現しない系の問題だった気がします。
とりあえず、例外無しのコードに書きなおすべきですね

snozawa, 2015-07-18:
Furutaractive問題は、起動の順番もしくはタイミングの問題かな？
Furutaractiveの最初のほうで、robot_deacriptionがくるまでwaitするかんじにすればいいかな?
数回に一回はかなりの高頻度なので、デバッグはしやすそうですね

YoheiKakiuchi, 2015-07-18:
> Furutaractiveの最初のほうで、robot_deacriptionがくるまでwaitするかんじにすればいいかな?

これは、launchにパラメータが書いてある場合は、ダイナミックに(launchに書いてあるんじゃなくて、nodeでset-paramする)じゃない限り、nodeからパラメータが読めないことはない仕様だと思うのだけど、どうだろうか？

garaemon, 2015-07-18:
roslaunchはかならずパラメータを最初に読み込むはずなので、どんな理由で動いていないかはよくわかっていないです。

その対処方法というよりは、furutaractiveをしっかり理解して何が問題かを理解するのが大事かなという気がしますね

garaemon, 2015-07-27:
> Rvizのロボット立ち位置モデルが出てこない

https://github.com/jsk-ros-pkg/jsk_visualization/pull/429
修正しました

garaemon, 2015-07-27:
現在の進捗

* nodeletが途中でexited cleanlyと言って死ぬ

  jsk_topic_tools/standalone_complexed_nodeletを使えば良い．部分的に移行中
* nodeletがつながらない

  mutexをしっかり取ることで解決
* クリックしないとfurutaractiveが出てこない

  https://github.com/jsk-ros-pkg/jsk_visualization/pull/428 で解決
* furutaractiveが出てこないことがある

  たぶん https://github.com/jsk-ros-pkg/jsk_visualization/pull/429 で解決

* eusが突然死する問題

  未解決. 例外が怪しいことを予想中

garaemon, 2015-07-28:
> furutaractiveが出てこないことがある
> 
> たぶん jsk-ros-pkg/jsk_visualization#429 で解決

まだ完璧には解決してなかったらしい

garaemon, 2015-07-31:
> furutaractiveが出てこないことがある
> 
> たぶん jsk-ros-pkg/jsk_visualization#429 で解決
> まだ完璧には解決してなかったらしい

ついに解決した気がします
https://github.com/jsk-ros-pkg/jsk_demos/pull/1010

garaemon, 2015-08-04:
> silverhammerのプロセスが残ったままのことがある

https://github.com/jsk-ros-pkg/jsk_common/pull/1066
これで解決します．

pythonのmultiprocess使っている部分を明示的にC++で書き直しました

garaemon, 2015-08-04:
launchの書き直しは残っていますが、これで一通りシステムの基礎となる部分の修正が終わり、安定度がかなり増したと思います。

ようやくもう少し楽しいレイヤに入れるかと思います

garaemon, 2015-08-05:
nodeletの見直しでプロセス数が348 -> 246になりました

garaemon, 2015-08-05:
とりあえず目につく問題はひと通り解決したつもりです．

launchを書きなおしたらfc, ocsを上げている状態で
```
$ ping jaxonred
PING jaxonred (192.168.96.52) 56(84) bytes of data.
64 bytes from jaxonred (192.168.96.52): icmp_req=1 ttl=64 time=30.5 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=2 ttl=64 time=24.5 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=3 ttl=64 time=59.2 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=4 ttl=64 time=36.1 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=5 ttl=64 time=23.0 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=6 ttl=64 time=31.1 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=7 ttl=64 time=23.1 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=8 ttl=64 time=41.1 ms
64 bytes from jaxonred (192.168.96.52): icmp_req=9 ttl=64 time=40.9 ms
```

closed 2015-08-05 by garaemon

YuOhara, 2015-08-29:
silverhammerが残ってしまいます。
```
leus@Z640:~/ros$ ps aux | grep silver
root       4870  0.0  0.0  72220  4428 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py __name:=fc_from_ocs_low_speed __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_from_ocs_low_speed-2.log
root       4880  1.1  0.0 669560 33380 ?        Sl   17:00   0:10 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py __name:=fc_from_ocs_low_speed __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_from_ocs_low_speed-2.log
root       7691  0.0  0.0  72220  4428 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_from_fc_low_speed __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_from_fc_low_speed-3.log
root       7692  1.1  0.0 669304 33012 ?        Sl   17:00   0:10 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_from_fc_low_speed __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_from_fc_low_speed-3.log
root       7825  0.0  0.0  72220  4420 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_from_fc_basic_low_speed __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_from_fc_basic_low_speed-4.log
root       7826  1.1  0.0 669304 35192 ?        Sl   17:00   0:11 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_from_fc_basic_low_speed __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_from_fc_basic_low_speed-4.log
root       7902  0.0  0.0  72220  4324 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py __name:=fc_from_ocs_reconfigure __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_from_ocs_reconfigure-11.log
root       7904  1.1  0.0 669560 35140 ?        Sl   17:00   0:10 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py __name:=fc_from_ocs_reconfigure __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_from_ocs_reconfigure-11.log
root       8142  0.0  0.0  72220  4420 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py __name:=fc_from_ocs_vehicle __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_from_ocs_vehicle-12.log
root       8143  1.0  0.0 669304 32820 ?        Sl   17:00   0:10 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py __name:=fc_from_ocs_vehicle __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_from_ocs_vehicle-12.log
root       8181  0.0  0.0  72220  4360 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_to_fc_reconfigure __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_to_fc_reconfigure-6.log
root       8183  1.1  0.0 743036 35012 ?        Sl   17:00   0:10 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_to_fc_reconfigure __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_to_fc_reconfigure-6.log
root       8558  0.0  0.0  72220  4444 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_from_fc_vehicle __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_from_fc_vehicle-7.log
root       8559  1.1  0.0 669304 32936 ?        Sl   17:00   0:10 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_receiver.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_from_fc_vehicle __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/ocs_from_fc_vehicle-7.log
root       8695  0.0  0.0  72220  4424 ?        Ss   17:00   0:00 sudo /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py __name:=fc_to_ocs_eus __log:=/home/leus/.ros/log/fa88a3ee-4e23-11e5-89ea-8cdcd43387e4/fc_to_ocs_eus-14.log
[... cut ...] us-14.log
root      35557  1.0  0.0 743036 34964 ?        Sl   17:06   0:06 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py __name:=fc_to_ocs_eus __log:=/home/leus/.ros/log/c4ca6052-4e24-11e5-94bb-8cdcd43387e4/fc_to_ocs_eus-14.log
leus      50754  0.0  0.0  16928  2488 pts/6    S+   17:16   0:00 grep --color=auto silver
leus@Z640:~/ros$ tmux ls
failed to connect to server
```
jvrc用には、silverhammerは使わないようにいずれlaunchを書き換えるかもしれませんが。。

garaemon, 2015-08-29:
親プロセスは分かりますか？

YuOhara, 2015-08-29:
```
4  1000   1703   1696  20   0  41276  4660 poll_s Ss   ?          0:00 init --user
```
一応ps alx結果を全部載せると
```
F   UID    PID   PPID PRI  NI    VSZ   RSS WCHAN  STAT TTY        TIME COMMAND
4     0      1      0  20   0  33900  4464 poll_s Ss   ?          0:02 /sbin/init
1     0      2      0  20   0      0     0 kthrea S    ?          0:00 [kthreadd]
1     0      3      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/0]
1     0      5      2   0 -20      0     0 worker S<   ?          0:00 [kworker/0:0H]
1     0      6      2  20   0      0     0 worker S    ?          0:00 [kworker/u288:0]
1     0      8      2  20   0      0     0 rcu_gp S    ?          0:28 [rcu_sched]
1     0      9      2  20   0      0     0 rcu_gp S    ?          0:00 [rcu_bh]
1     0     10      2  20   0      0     0 rcu_no S    ?          0:07 [rcuos/0]
1     0     11      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/0]
1     0     12      2 -100  -      0     0 smpboo S    ?          0:00 [migration/0]
5     0     13      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/0]
5     0     14      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/1]
1     0     15      2 -100  -      0     0 smpboo S    ?          0:00 [migration/1]
1     0     16      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/1]
1     0     17      2  20   0      0     0 worker S    ?          0:00 [kworker/1:0]
1     0     18      2   0 -20      0     0 worker S<   ?          0:00 [kworker/1:0H]
1     0     19      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/1]
1     0     20      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/1]
5     0     21      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/2]
1     0     22      2 -100  -      0     0 smpboo S    ?          0:00 [migration/2]
1     0     23      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/2]
1     0     25      2   0 -20      0     0 worker S<   ?          0:00 [kworker/2:0H]
1     0     26      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/2]
1     0     27      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/2]
5     0     28      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/3]
1     0     29      2 -100  -      0     0 smpboo S    ?          0:00 [migration/3]
1     0     30      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/3]
1     0     31      2  20   0      0     0 worker S    ?          0:00 [kworker/3:0]
1     0     32      2   0 -20      0     0 worker S<   ?          0:00 [kworker/3:0H]
1     0     33      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/3]
1     0     34      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/3]
5     0     35      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/4]
1     0     36      2 -100  -      0     0 smpboo S    ?          0:00 [migration/4]
1     0     37      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/4]
1     0     38      2  20   0      0     0 worker S    ?          0:00 [kworker/4:0]
1     0     39      2   0 -20      0     0 worker S<   ?          0:00 [kworker/4:0H]
1     0     40      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/4]
1     0     41      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/4]
5     0     42      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/5]
1     0     43      2 -100  -      0     0 smpboo S    ?          0:00 [migration/5]
1     0     44      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/5]
1     0     46      2   0 -20      0     0 worker S<   ?          0:00 [kworker/5:0H]
1     0     47      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/5]
1     0     48      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/5]
5     0     49      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/6]
1     0     50      2 -100  -      0     0 smpboo S    ?          0:00 [migration/6]
1     0     51      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/6]
1     0     52      2  20   0      0     0 worker S    ?          0:00 [kworker/6:0]
1     0     53      2   0 -20      0     0 worker S<   ?          0:00 [kworker/6:0H]
1     0     54      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/6]
1     0     55      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/6]
5     0     56      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/7]
1     0     57      2 -100  -      0     0 smpboo S    ?          0:00 [migration/7]
1     0     58      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/7]
1     0     59      2  20   0      0     0 worker S    ?          0:00 [kworker/7:0]
1     0     60      2   0 -20      0     0 worker S<   ?          0:00 [kworker/7:0H]
1     0     61      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/7]
1     0     62      2  20   0      0     0 rcu_no S    ?          0:00 [rcuob/7]
5     0     63      2 -100  -      0     0 smpboo S    ?          0:00 [watchdog/8]
1     0     64      2 -100  -      0     0 smpboo S    ?          0:00 [migration/8]
1     0     65      2  20   0      0     0 smpboo S    ?          0:00 [ksoftirqd/8]
1     0     67      2   0 -20      0     0 worker S<   ?          0:00 [kworker/8:0H]
1     0     68      2  20   0      0     0 rcu_no S    ?          0:00 [rcuos/8]
[... cut ...]
4     0 126013 126003  20   0 448108 35016 poll_s Sl   ?          0:41 python /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py /tf:=/ocs/tf /joint_states:=/ocs/joint_states /robot_description:=/ocs/robot_description __name:=ocs_to_fc_eus __log:=/home/leus/.ros/log/5da01aa4-4e27-11e5-876f-8cdcd43387e4/ocs_to_fc_eus-10.log
1     0 136671      2   0 -20      0     0 worker S<   ?          0:00 [kworker/19:1H]
1     0 143194      2   0 -20      0     0 worker S<   ?          0:00 [kworker/16:1H]

```

garaemon, 2015-08-29:
ゾンビになってますね。

立ち上げ直後はroslaunchが親になっていて、roslaunchはSIGTERMを子供におくるはずなので、SIGTERMで死なないプロセスということになります.

◉  ryohei

2015-08-29 19:05 GMT+09:00 Yu Ohara <notifications@github.com>:

> 4  1000   1703   1696  20   0  41276  4660 poll_s Ss   ?          0:00 init --user
>
> 一応ps alx結果を全部載せると
>
> F   UID    PID   PPID PRI  NI    VSZ   RSS WCHAN  STAT TTY        TIME COMMAND
> 4     0 [...]