jsk-ros-pkg/jsk_demos #1004: LOGが必ず残るようにする

snozawa, 2015-07-26:
ログが必ず残るようにしたいです。
@garaemon 

- 体内（RTC）は、servoOffのときにログがはかれるようにする
@YoheiKakiuchi さん、https://github.com/fkanehiro/hrpsys-base/issues/490でemergency servo signalがあったらDataLoggerのログが残されるようになってるでしょうか。
- 体外PCは、ROSbagをとる。最低限hrpsys ros bridgeのstate関係がとれればOK？rosbridge起動と同時にrosbag record開始はどうでしょうか。

garaemon, 2015-07-26:
`rosbag record /lfsensor /lhsensor /rfsensor /rhsensor /joint_states /motor_states --duration 60`で32Mなので，一日だと
`32 * 60 * 24`で46Gでしょうか

garaemon, 2015-07-26:
とりあえず10Gくらい体内でログを取るようにしましょうか．それを超えたら古いものから消すようなスクリプトはすでに有ります．

ちなみに，JAXONREDは今は9.2Gしかありません

```
~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/sda1        94G   80G  9.2G  90% /
udev            3.9G  8.0K  3.9G   1% /dev
tmpfs           800M  820K  799M   1% /run
none            5.0M  8.0K  5.0M   1% /run/lock
none            4.0G  240K  4.0G   1% /run/shm
```

```
$ du -h . -d 1
884K	./.gstreamer-0.10
1.8G	./Desktop
80K	./.compiz-1
8.0K	./.remmina
36K	./.gnome2
23G	./tmp
24K	./.anthy
75M	./jaxon_gopos_000
308K	./.gconf
4.0K	./Public
40K	./.gnome
111M	./fsharp
9.2M	./src
40K	./.ssh
2.2G	./.ros
6.4G	./ros
123M	./.config
16K	./.aptitude
4.0K	./Documents
141M	./nozawa_log
110M	./sonoba_2
514M	./kuroiwa-0530
4.0K	./Pictures
80K	./.matplotlib
113M	./sonoba_log
4.0K	./Videos
19M	./.mozilla
0	./.gvfs
48K	./.subversion
5.9G	./prog
3.4G	./intel
292K	./ookubo
56K	./.pulse
2.2M	./.local
5.1G	./ros.bak
21M	./.emacs.d
168K	./.fontconfig
8.4G	./kuroiwa
4.0K	./Templates
36K	./.byobu
200M	./.cache
440K	./.mozc
4.0K	./Music
110M	./sonoba_3
36K	./.pki
66M	./jaxon_gopos_015
24K	./.dbus
1.7M	./.launchpadlib
8.0K	./.mplayer
8.0K	./.vim
12K	./.mission-control
356K	./.thumbnails
4.0K	./.gnome2_private
4.0K	./Downloads
58G	.
```

garaemon, 2015-07-26:
jaxonの場合，やはり~/tmp以下にすべてのログをすべて残しているのがやり過ぎかも

YoheiKakiuchi, 2015-07-26:
> 体内（RTC）は、servoOffのときにログがはかれるようにする @YoheiKakiuchi さん、

statusがエラーになったときですが、use_owned_ecをTrueにするとそうなるとおもいます。
https://github.com/k-okada/hrpsys-base/blob/master/python/hrpsys_config.py#L237
https://github.com/k-okada/hrpsys-base/commit/867a7dadca4509c1eaeacf90da7304d1e1b61af6

snozawa, 2015-07-26:
> とりあえず10Gくらい体内でログを取るようにしましょうか．それを超えたら古いものから消すようなスクリプトはすでに有ります．

これはいいですね。今すぐにでも全ロボットで使いましょう。
緊急時のログの代わりなので、数時間分くらいでも良いかもしれません。
その代わり、もっと他の情報を残すと良いかも。

> jaxonの場合，やはり~/tmp以下にすべてのログをすべて残しているのがやり過ぎかも

tmpに今はなにがのこってますか？

snozawa, 2015-07-26:
> statusがエラーになったときですが、use_owned_ecをTrueにするとそうなるとおもいます。
> https://github.com/k-okada/hrpsys-base/blob/master/python/hrpsys_config.py#L237
k-okada/hrpsys-base@867a7da

これは体外の別ExecutionContextのDataLoggerでとるという話でしたっけ？
emergency signalでservoOffがされたときなど、ログをとるようにするのは別でしたでしょうか？

YoheiKakiuchi, 2015-07-26:
> emergency signalでservoOffがされたときなど、ログをとるようにするのは別でしたでしょうか？

意図した動作だとおもいます。servoOffではなくて、statusにエラーが出たらという条件ですが。

> これは体外の別ExecutionContextのDataLoggerでとるという話でしたっけ？

説明が正しいか分かりませんが、体内のrtcdの多分プロセスも同じで別のECがDataLoggerのメソッドを呼びます。

garaemon, 2015-07-27:
つねにrosbagとれるスクリプト
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_data/scripts/rosbag_always.py

eisoku9618, 2015-07-27:
> jaxonの場合，やはり~/tmp以下にすべてのログをすべて残し ているのがやり過ぎかも

毎日午前四時に一週間前のログをariesにuploadしてから体内から消すようにしていたつもりですが、そのcronが動いていない or
一週間が長すぎたのかもしれません。

あとで確認します。

garaemon, 2015-07-27:
> cronが動いていない

cronは止めています。通信帯域を使うと、DRCはもちろんですが、画像の取得などにも影響が有りそうなので。

garaemon, 2015-08-04:
https://github.com/jsk-ros-pkg/jsk_common/pull/1063

bagとりスクリプトを修正しました．

garaemon, 2015-08-04:
> 毎日午前四時に一週間前のログをariesにuploadしてから体内から消すようにしていたつもりですが、そのcronが動いていない or
> 一週間が長すぎたのかもしれません。

修正しました
https://github.com/jsk-ros-pkg/trans_system/pull/431

garaemon, 2015-08-05:
> 体内（RTC）は、servoOffのときにログがはかれるようにする @YoheiKakiuchi さん、
> statusがエラーになったときですが、use_owned_ecをTrueにするとそうなるとおもいます。
> https://github.com/k-okada/hrpsys-base/blob/master/python/hrpsys_config.py#L237
> k-okada/hrpsys-base@867a7da

何も考えずにtrueにしたら良いでしょうか?

garaemon, 2015-08-05:
bagに関しては体外マシンで長時間安定して取ることが出きるようになってきたので、
可視化と、サーボを落としたら自動的にbagを切るというのをやったら終わりかなという気がしています。

YoheiKakiuchi, 2015-08-05:
> 何も考えずにtrueにしたら良いでしょうか?

そうですね．まずは，ちゃんとログがとれているか確認してみて下さい．

garaemon, 2015-08-05:
> そうですね．まずは，ちゃんとログがとれているか確認してみて下さい．

これは何をするとログが取られるのでしょうか?

YoheiKakiuchi, 2015-08-05:
> これは何をするとログが取られるのでしょうか?

サーボがONされているときに，statusが0x800（正常はサーボoff)以外になって,status_monitorで赤くなってサーボが落ちた時にログがとられるはずです．

garaemon, 2015-08-05:
つまり、リレーを落とせばよいのでしょうか?

YoheiKakiuchi, 2015-08-05:
> つまり、リレーを落とせばよいのでしょうか?

リレーのOn/Offは直接見ていませんが，リレーがoffされると，refとcurrent angleの差が大きくなって落ちる場合が多いので，ログがとられるはずです．

garaemon, 2015-08-06:
エラーになりました

```diff
$ git diff urata_hrpsys_config.py
diff --git a/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py b/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py
index b6af882..3cc9f60 100755
--- a/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py
+++ b/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py
@@ -10,6 +10,7 @@ import OpenHRP
 class URATAHrpsysConfigurator(HrpsysConfigurator):
     def __init__(self, robotname=""):
         self.ROBOT_NAME = robotname
+        self.log_use_owned_ec = True
         HrpsysConfigurator.__init__(self)
         self.defJointGroups()
 
@@ -152,8 +153,10 @@ class URATAHrpsysConfigurator(HrpsysConfigurator):
             stp.eefm_rot_damping_gain=20*1.6*1.1*1.5*1.2
             stp.eefm_pos_damping_gain=[3500*1.6*6, 3500*1.6*6, 3500*1.6*1.1*1.5*1.2]
         elif self.ROBOT_NAME == "JAXON_RED":
-            stp.eefm_rot_damping_gain=20*1.6*1.1*1.5
-            stp.eefm_pos_damping_gain=[3500*1.6*6, 3500*1.6*6, 3500*1.6*1.1*1.5]
+            # stp.eefm_rot_damping_gain=20*1.6*1.1*1.5
+            # stp.eefm_pos_damping_gain=[3500*1.6*6, 3500*1.6*6, 3500*1.6*1.1*1.5]
+            stp.eefm_rot_damping_gain=20*1.6*1.1*1.5*1.2
+            stp.eefm_pos_damping_gain=[3500*1.6*6, 3500*1.6*6, 3500*1.6*1.1*1.5*1.2]
         stp.eefm_rot_time_const=1.5/1.1
         stp.eefm_pos_time_const_support=[3.0/1.1, 3.0/1.1, 1.5/1.1]
         stp.eefm_wrench_alpha_blending=0.7
```

```
[hrpsys.py] activating components
Traceback (most recent call last):
  File "/home/leus/ros/hydro/src/jsk-ros-pkg/trans_system/jaxon_ros_bridge/scripts/jaxon_red_setup.py", line 13, in <module>
    JAXON_REDRealHrpsysConfigurator().init(sys.argv[1], sys.argv[2])
  File "/home/leus/ros/hydro/src/jsk-ros-pkg/trans_system/trans_ros_bridge/src/trans_ros_bridge/urata_real_hrpsys_config.py", line 30, in init
[hrpsys-2] process has died [pid 3349, exit code 134, cmd sudo env -i LANG=C ORBgiopMaxMsgSize=2147483648 PATH=/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus//Linux/bin:/home/leus/ros/hydro/devel/bin:/home/leus/ros/hydro_parent/devel/bin:/opt/ros/hydro/bin:/opt/intel/composer_xe_2015.2.164/bin/ia32:/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus//Linux/bin:/opt/intel/composer_xe_2015.2.164/bin/ia32:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/leus/fsharp/FSharp-2.0.0.0/bin:/home/leus/fsharp/FSharp-2.0.0.0/bin LD_LIBRARY_PATH=/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus//Linux/lib:/home/leus/ros/hydro/devel/lib:/home/leus/ros/hydro_parent/devel/lib:/opt/ros/hydro/lib:/opt/intel/composer_xe_2015.2.164/compiler/lib/ia32:/opt/intel/composer_xe_2015.2.164/mpirt/lib/ia32:/opt/intel/composer_xe_2015.2.164/ipp/../compiler/lib/ia32:/opt/intel/composer_xe_2015.2.164/ipp/lib/ia32:/opt/intel/composer_xe_2015.2.164/ipp/tools/ia32/perfsys:/opt/intel/composer_xe_2015.2.164/compiler/lib/ia32:/opt/intel/composer_xe_2015.2.164/mkl/lib/ia32:/opt/intel/composer_xe_2015.2.164/tbb/lib/ia32/gcc4.4:/opt/intel/composer_xe_2015.2.164/debugger/ipt/ia32/lib /home/leus/ros/hydro_parent/devel/lib/openrtm_aist/bin/rtcd -o manager.is_master:YES   -o corba.nameservers:localhost:15005   -o corba.endpoint:192.168.96.52:  -o naming.formats:%n.rtc   -o logger.file_name:/tmp/rtc%p.log,stdout   -o logger.enable:YES -o logger.log_level:NORMAL -o logger.file_name:stdout -o exec_cxt.periodic.rate:200 -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/leus/ros/hydro/src/jsk-ros-pkg/trans_system/hrpsys_trans_bridge/lib,/home/leus/ros/hydro_parent/devel/share/hrpsys/lib -o manager.modules.preload:RobotHardware.so,hrpEC.so -o manager.components.precreate:RobotHardware -o exec_cxt.periodic.type:hrpExecutionContext -o example.SequencePlayer.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.ForwardKinematics.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.ImpedanceController.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.AutoBalancer.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.StateHolder.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.TorqueFilter.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.TorqueController.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.ThermoEstimator.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.ThermoLimiter.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.VirtualForceSensor.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.AbsoluteForceSensor.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.conf -o example.RemoveForceSensorLin [... cut ...]
    print("[rtm.py] \033[31m   error in serialize %s of %s %s\033[0m" % (rtc.name(),  [[r, r.name()] for r in rtcs], str(e)))
  File "/home/leus/ros/hydro_parent/devel/lib/python2.7/dist-packages/hrpsys/rtm.py", line 169, in name
    cprof = self.ref.get_component_profile()
  File "/home/leus/ros/hydro_parent/devel/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1344, in get_component_profile
    return _omnipy.invoke(self, "get_component_profile", _0_RTC.RTObject._d_get_component_profile, args)
omniORB.CORBA.TRANSIENT: CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
```