jsk-ros-pkg/jsk_demos #1027: could not find pacakge [jskpointcloud]

himlen1990, 2015-08-09:
could not find pacakge [jskpointcloud] when  running jsk_demos/detect_cans_in_fridge_201202/euslisp/color_histogram_creater.l

k-okada, 2015-08-09:
please describe
1 what did you do: which command did you pinput
1 what did you get: copy and paste all output including this error

himlen1990, 2015-08-09:
sorry, and here is the result 

himlen@himlen-ThinkPad-W550s:~/ros/hydro/src/jsk-ros-pkg/jsk_demos/detect_cans_in_fridge_201202/euslisp$ roseus color_histogram_creater.l 
configuring by "/home/himlen/ros/hydro_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0.0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x4edfde0[16374] --> 0x533c320[32748] top=37ba
irtgl irtviewer 
EusLisp 9.12(7777529 1.0.8) for Linux64 created on himlen-ThinkPad-W550s(Wed Jun 24 13:42:57 JST 2015)
roseus ;; loading roseus("0709095") on euslisp((9.12 himlen-ThinkPad-W550s Wed Jun 24 13:42:57 JST 2015 7777529 1.0.8))
eustf roseus_c_util 

;; load-ext-lib: jskeus 	(Sun Aug  9 13:31:07 2015)
ckdtree not loaded
fftlib not loaded
svm not loaded
jsk matlib graph kdtree galib neurolib eusrapid rapid kalmanlib lmeds icp optimization octave preview queue searchlib dplib backprop bayesian qlearn soundlib sendmail linearproblem pfilter euscorba galatea jskc win32api cvoxel jskgeoc regexp festival listener talk unittest eusjasper jasper 
;; load-ext-lib: jskeusunix 	(Sun Aug  9 13:31:08 2015)
jskunix sericom uptime 
;; load-ext-lib: jskeusgeo 	(Sun Aug  9 13:31:17 2015)
jskgeo pickport kdraw primtpatch primtapp vclip pfitting sample_poisson 
;; load-ext-lib: jskeusimage 	(Sun Aug  9 13:31:09 2015)
jpeg_compress_struct jpeg_decompress_struct jskimage 
;; load-ext-lib: jskeusx 	(Sun Aug  9 13:31:24 2015)
jskx pickview winspect browse gnuplotlib graphwin 
;; load-ext-lib: jskeusgl 	(Sun Aug  9 13:31:29 2015)
jskgl nurbs voxel jskglc [rospack] Error: stack/package jskpointcloud not found
;; could not find pacakge [jskpointcloud]
/home/himlen/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  0x5805f58 in (error "file ~s not found" fname)E:

k-okada, 2015-08-10:
please copy and paste `~/.bashrc`, basically you do not need any custom
settings.

k-okada, 2015-08-10:
Sorry it will not your fault,we're working on https://github.com/jsk-ros-pkg/jsk_demos/commit/2725e0e81b595fdf01db2b88f4c3611b474ab0ac#commitcomment-12609242 and https://github.com/jsk-ros-pkg/jsk_demos/pull/1030

Please ask them directory if there'are anything you can help

k-okada, 2015-08-10:
@himlen1990 , now you can run color_histogram_creater.l with https://github.com/jsk-ros-pkg/jsk_demos/pull/1032, @YoheiKakiuchi or @aginika will explain how to use them

himlen1990, 2015-08-10:
thank you for reminding me, it helps a lot

closed 2015-08-10 by himlen1990