jsk-ros-pkg/jsk_demos #1042: osc_rviz のfurutaractiveまわりでtfga

YuOhara, 2015-08-28:


garaemon, 2015-08-28:
urdf_control_markerの出力を見てみると良いかも。あとはdynamic_tf_publisherかな。

YuOhara, 2015-08-28:
rqt_consoleで
```
Node: /urdf_control_marker
Time: 19:59:33.181482102 (2015-08-28)
Severity: Error
Published Topics: /rosout, /urdf_control_marker/pose, /urdf_control_marker/selected_pose, /urdf_control_marker/update, /urdf_control_marker/update_full

Failed to call dynamic_tf: /odom_on_ground => robot_marker_root

Location:
/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_visualization/jsk_interactive_markers/jsk_interactive_marker/src/urdf_control_marker.cpp:callDynamicTf:150
```
何か他に確認の仕方ありましたでしょうか

garaemon, 2015-08-28:
この部分、一回目はset失敗して、二回目に成功するのが通常の挙動です。
二回目が出てなければ大丈夫なはずだが、、、
dynamic tfが二ついたりしないかな？

YoheiKakiuchi, 2015-08-28:
関係ないとおもうけど、てもと環境では以下になるのですが、これは何をインストールするのかな？

~~~
process[task_state_manager-18]: started with pid [2770]
Traceback (most recent call last):
  File "/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_demos/jsk_2015_06_hrp_drc/drc_task_common/scripts/task_state_manager.py", line 4, in <module>
    import fysom
ImportError: No module named fysom
~~~

YoheiKakiuchi, 2015-08-28:
/ocs/tf  が出ていないようなんだけど、remap忘れみたいなことないかな？

YuOhara, 2015-08-28:
fysomについては、　#183 で対応されたとおもったのですが、
```
sudo pip install fysom
```
が自分の環境でも必要でした

YoheiKakiuchi, 2015-08-28:
~~~
/odom_on_ground => robot_marker_root
~~~
のtfはあって、robot_marker_root -> /jsk_model_marker_interface/robot/BODYのtfがnan。
robot_marker_rootという名前が見えるのがおかしい？？
~~~
  - 
    header: 
      seq: 0
      stamp: 
        secs: 3227
        nsecs: 313000000
      frame_id: /robot_marker_root
    child_frame_id: /jsk_model_marker_interface/robot/BODY
    transform: 
      translation: 
        x: nan
        y: nan
        z: nan
      rotation: 
        x: nan
        y: nan
        z: nan
        w: nan

~~~

YuOhara, 2015-08-28:
[image]
ちゃんと動いたらこう出るはずで，roboto_marker_root と/jsk_model_marker_interface/robot/BODYはあるようです

YoheiKakiuchi, 2015-08-28:
すいません、素人過ぎてへんなところをさまよってます。

先頭の/があったりなかったりで違ったりしないかな？

YuOhara, 2015-08-28:
なるほど。見てみます

YuOhara, 2015-08-28:
そもそもの原因はわかりませんが、とりあえず
```
[ERROR] [1440758263.957239529]: [void UrdfModelMarker::resetRootForVisualization()] Failed to lookup transformation from /jsk_model_marker_interface/robot/lleg_end_coords to /jsk_model_marker_interface/robot/BODY: "jsk_model_marker_interface/robot/BODY" passed to lookupTransform argument target_frame does not exist.
[ERROR] [1440758263.957352874]: [void UrdfModelMarker::resetRootForVisualization()] Failed to lookup transformation from /jsk_model_marker_interface/robot/rleg_end_coords to /jsk_model_marker_interface/robot/BODY: "jsk_model_marker_interface/robot/BODY" passed to lookupTransform argument target_frame does not exist.
```
と出ているのはまずそうです

YuOhara, 2015-08-28:
ひとまず治ったかもしれないです。取り急ぎdiffを
```
leus@Z640:~/ros/indigo/src/jsk-ros-pkg/jsk_visualization/jsk_interactive_markers/jsk_interactive_marker/src$ git diff urdf_model_marker.cpp 
diff --git a/jsk_interactive_markers/jsk_interactive_marker/src/urdf_model_marke
index 38af0f4..f5b6aa6 100644
--- a/jsk_interactive_markers/jsk_interactive_marker/src/urdf_model_marker.cpp
+++ b/jsk_interactive_markers/jsk_interactive_marker/src/urdf_model_marker.cpp
@@ -351,7 +351,7 @@ void UrdfModelMarker::resetRootForVisualization(){
   if(fixed_link_.size() > 0 && (mode_ == "visualization" || mode_ == "robot")){
     string marker_name =  tf_prefix_ + model->getRoot()->name;
     tf::StampedTransform st_offset;
-
+    sleep(5.0);
     bool first_offset = true;
     for(int i=0; i<fixed_link_.size(); i++){
       std::string link = fixed_link_[i];
(END)
```

YoheiKakiuchi, 2015-08-28:
はい、それは怪しいですね。
その下の、

closed 2015-08-28 by YuOhara