jsk-ros-pkg/jsk_demos #1100: Why travis test for interactive_behaivor_201409 passes?

wkentaro, 2015-10-29:
without the deb release of face_recognition?

do you have any idea?

I have a build problem in a package which depends on interective_behavior_201409 implicitly. 
see https://github.com/jsk-ros-pkg/jsk_smart_apps/issues/60#issuecomment-152045390

wkentaro, 2015-10-29:
this is because of this line.
https://github.com/jsk-ros-pkg/jsk_demos/blob/master/interactive_behavior_201409/package.xml#L25

wkentaro, 2015-10-29:
this will be fixed by https://github.com/jsk-ros-pkg/jsk_travis/pull/180

k-okada, 2015-10-29:
> this will be fixed by jsk-ros-pkg/jsk_travis#180

removing https://github.com/fkanehiro/hrpsys-base/pull/854 may fix this
problem, (why we have both face_detection and faece_recognition )

having missing deb on build_depend may ignored by rosdep with "ignore
error" option, but find_package(face_detection) failed during catkin b,
if you did not use face_detection on build time, such as C source code
requires them, you can remove such package names from find_package

wkentaro, 2015-10-29:
I don't know about why `interactive_behavior_201409` depends on `face_recognition`.
@furushchev may know about it.

furushchev, 2015-10-29:
main.l in `interactive_behavior_201409` subscribes `face_recognition/FaceRecognitionAction`.

furushchev, 2015-10-29:
Current progress:
- [ ] fix travis.sh: jsk-ros-pkg/jsk_travis#180 -> this causes test fail on `interactive_behavior_201409`
- [ ] test `USE_DEB=true` since `face_recognition` is not yet released
- [ ] apply new `.travis` to `jsk_demos` repo
- [ ] add `face_recognition` to `.travis.rosinstall`

wkentaro, 2015-11-03:
done with https://github.com/jsk-ros-pkg/jsk_demos/pull/1104

closed 2015-11-03 by wkentaro