jsk-ros-pkg/jsk_demos #1116: 610 demo does not work on indigo

furushchev, 2015-11-25:
error occurs on smach_viewer.py

```bash
$ rosrun smach_viewer smach_viewer.py
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/ros/indigo/lib/smach_viewer/smach_viewer.py", line 848, in _update_graph
    self.set_dotcode(dotstr,zoom=False)
  File "/opt/ros/indigo/lib/smach_viewer/smach_viewer.py", line 866, in set_dotcode
    if self.widget.set_dotcode(dotcode, None):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/wxxdot.py", line 455, in set_dotcode
    self.set_xdotcode(xdotcode)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/wxxdot.py", line 483, in set_xdotcode
    self.graph = parser.parse()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 1146, in parse
    DotParser.parse(self)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 951, in parse
    self.parse_graph()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 960, in parse_graph
    self.parse_stmt()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 993, in parse_stmt
    self.parse_subgraph()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 974, in parse_subgraph
    self.parse_stmt()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 993, in parse_stmt
    self.parse_subgraph()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 974, in parse_subgraph
    self.parse_stmt()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 1009, in parse_stmt
    self.handle_node(id, attrs)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 1120, in handle_node
    shapes.extend(parser.parse())
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 608, in parse
    points = self.read_polygon()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 503, in read_polygon
    x, y = self.read_point()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 486, in read_point
    x = self.read_number()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xdot/xdot.py", line 480, in read_number
    return int(self.read_code())
ValueError: invalid literal for int() with base 10: '310.92'
```

k-okada, 2015-11-25:
see  https://github.com/ros-visualization/executive_smach_visualization/issues/7 for issue and https://github.com/jbohren/xdot/pull/12 for PR

k-okada, 2015-11-25:
https://github.com/jbohren/xdot/pull/7/files is short version

closed 2015-11-26 by k-okada