jsk-ros-pkg/jsk_demos #1149: 冷蔵庫デモをgazeboで動かす

furushchev, 2015-12-07:
### pr2.launchが動くレベル
- [ ] 依存パッケージが結構インストールされない => https://github.com/jsk-ros-pkg/jsk_robot/issues/512
- [ ] tilt_shadow_filterが落ちる @aginika 
- [ ] move_baseがsegfaultする => https://github.com/jsk-ros-pkg/jsk_robot/pull/513

### detect_canデモlaunchレベル

aginika, 2015-12-07:
https://github.com/jsk-ros-pkg/jsk_3rdparty/issues/11 ?
https://github.com/ros-perception/laser_filters/pull/27 ?

aginika, 2015-12-07:
rgb -> bgr プロブレム
https://github.com/PR2/pr2_common/pull/244

aginika, 2015-12-07:
Depth nameプログラム
https://github.com/PR2/pr2_common/pull/243

furushchev, 2015-12-07:
@aginika ありがとう

aginika, 2015-12-07:
https://github.com/euslisp/EusLisp/issues/126

aginika, 2015-12-07:
@furushchev tilit_shadow_filterが落ちません・・・

aginika, 2015-12-07:
```
export ROBOT=sim
roslaunch jsk_pr2_startup pr2_gazebo_73b2.launch
```
を行っただけなのですが、これ以外にも必要でしょうか

furushchev, 2015-12-07:
確かに落ちてます。
配列のindexに-2147483648を入れるときがあるみたいです。

-1490571
by @aginika

furushchev, 2015-12-07:
@aginika https://github.com/furushchev/fridge これです。

furushchev, 2015-12-08:
冷蔵庫がバラバラになってしまう時がある。

[image]

furushchev, 2015-12-08:
冷蔵庫を固定してみると、ドアの把持まではうまくいきそう。

[image]

aginika, 2015-12-08:
- rvizでのinitial pose をinitするnodeを用意する

aginika, 2015-12-08:
- remove this 
[image]

aginika, 2015-12-08:
- pr2の把持が遅い
https://github.com/PR2/pr2_common/issues/245

仮fixには
https://github.com/aginika/pr2_common/blob/speed-up-grasp/pr2_description/urdf/gripper_v0/gripper.transmission.xacro#L45
?で行けるかも。把持できるか試していないので、怪しい

furushchev, 2015-12-09:
- pr2のstart-graspが終わらない

[image]

aginika, 2015-12-09:
stop-graspは終わる?

furushchev, 2015-12-09:
終わる。おそらくドアハンドルに阻まれて目標に到達しないんだと思う。

aginika, 2015-12-09:
http://answers.ros.org/question/10382/problem-picking-items-up-in-gazebo-with-pr2/
これとはべつ？
https://github.com/PR2/pr2_controllers/blob/hydro-devel/pr2_gripper_action/src/pr2_gripper_action.cpp#L212

aginika, 2015-12-10:
start-grasp問題は上記ので解決しました。

k-okada, 2015-12-11:
上記ので，というのは
```
<!-- Set params for sim -->
<param
name="r_gripper_controller/gripper_action_node/stall_velocity_threshold"
value="0.01" type="double"/>
<param
name="l_gripper_controller/gripper_action_node/stall_velocity_threshold"
value="0.01" type="double"/>
```
?

aginika, 2015-12-11:
はい
その変更をhttps://github.com/jsk-ros-pkg/jsk_robot/pull/519/files
で直してくれてるはずです

furushchev, 2015-12-11:
@aginika @k-okada みんな困ると思うのでupstreamにもPR送りました。
https://github.com/PR2/pr2_simulator/pull/115

TheDash, 2015-12-14:
@furushchev 

If anyone else from your lab needs maintainer admin access.. let me know and I can give it to you. You guys are doing great work.

furushchev, 2016-11-06:
Closed this issue. All issues were now solved.

closed 2016-11-06 by furushchev