jsk-ros-pkg/jsk_demos #1217: [jsk_maps] 実行の仕方について

kochigami, 2017-04-13:
``` roslaunch jsk_maps start_map_eng8.launch ```の実行について、
machineがlocalhostというのは分からないというエラー（バイナリとソースコードで確認）
が起きてしまうのですが、正しい実行方法をどなたかご存知の方はご教示ください。
自力で分からずすみません。

ubuntu 14.04
ros indigo
echo $MACHINE
=> (returned none)

１．バイナリで試した
１−１. 古いバージョン
```
apt-show-versions ros-indigo-jsk-maps
=> ros-indigo-jsk-maps:amd64/trusty 0.0.4-0trusty-20170315-012702-0700 upgradeable to 0.0.4-0trusty-20170328-012950-0700
roslaunch jsk_maps start_map_eng8.launch
=> ERROR: unknown machine [localhost]
```
1-2. 新しいバージョン
```
sudo apt-get update
sudo apt-get upgrade
apt-show-versions ros-indigo-jsk-maps
=> ros-indigo-jsk-maps:amd64/trusty 0.0.4-0trusty-20170328-012950-0700 uptodate
roslaunch jsk_maps start_map_eng8.launch
=> ERROR: unknown machine [localhost]
```
2. ソースで試した
.rosinstallで指定されたversion: 0.0.4 
```
roslaunch jsk_maps start_map_eng8.launch （コンパイル前）
... logging to /home/kochigami/.ros/log/20170412-205613_06eb07e6-1f77-11e7-95d9-00306444a934/roslaunch-nori-3433.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

while processing /home/kochigami/ros/indigo/src/jsk-ros-pkg/jsk_demos/jsk_maps/launch/include/start_map_eng8_main.xml:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/kochigami/ros/indigo/src/jsk-ros-pkg/jsk_demos/jsk_maps/launch/include/start_map_eng8_main.xml'
The traceback for the exception was written to the log file

roslaunch jsk_maps start_map_eng8.launch （コンパイル後）
ERROR: unknown machine [localhost]
```

k-okada, 2017-04-13:
>    1. ソースで試した

これはちょっと難しくて、http://wiki.ros.org/roslaunch/XML/machine にかいてあるんだけど、

```
<launch>
 <machine name="localhost" address="localhost" default="true"/>
 <incldue file="$(find jsk_map)/launch/start_map_eng8.launch" />
</launch>
```
として使うことが期待されています．

kochigami, 2019-05-30:
こちら２年も放置してしまいすみませんでした．
```
<launch>
  <include file="$(find jsk_maps)/launch/start_map_eng2.launch">
    <arg name="use_machine" default="true" />
    <arg name="use_pictogram" default="true" />
  </include>
</launch>
```
で実行することができました．
ありがとうございました．

closed 2019-05-30 by kochigami