jsk-ros-pkg/jsk_demos #1226: 2017 年ゼミ

k-okada, 2017-10-18:
大村 @dyson8910
室岡 @takayuki5168
古賀 @koga-yu
 平岡 @Naoki-Hiraoka

k-okada, 2017-10-18:
10/25 までの宿題。　ロボットシステムの宿題(talker/listner, 画像処理結果を受け取り描画する／cmd_vel でロボットを動かす）をeuslispで書く。

Naoki-Hiraoka, 2017-10-18:
```
mkidr -p ~/semi_ws/src
cd semi_ws/src
git clone http://github.com/jsk-ros-pkg/jsk_robot
 rosdep install -r --from-paths . --ignore-src
ros-kinetic-pepper-meshes
ros-kinetic-nao-meshes
git clone https://github.com/fetchrobotics/fetch_ros
git clone https://github.com/ros-naoqi/naoqi_interaction
cd ..
catkin b peppereus
source devel/setup.bash
roscd peppereus
roseus pepper.l
$ (pepper)
$ (objects (list *pepper*))
$ (send *pepper* :angle-vector)
$ (send *pepper* :head :neck-p :joint-angle 30) ;; https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_fetch_robot
```

koga-yu, 2017-10-24:
上のコマンドを打ち込んだところ、catkin build peppereusでエラーが出ます
ターミナルの出力は以下のとおりです

====================================================================
---------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/kogatti/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/kogatti/semi_ws
---------------------------------------------------------------------------
Source Space:       [exists] /home/kogatti/semi_ws/src
Log Space:          [exists] /home/kogatti/semi_ws/logs
Build Space:        [exists] /home/kogatti/semi_ws/build
Devel Space:        [exists] /home/kogatti/semi_ws/devel
Install Space:      [unused] /home/kogatti/semi_ws/install
DESTDIR:            [unused] None
---------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------------
[build] Found '20' packages in 0.0 seconds.                                                                                                  
[build] Package table is up to date.                                                                                                         
Starting  >>> naoqieus                                                                                                                       
_____________________________________________________________________________________________________________________________________________
Errors     << naoqieus:cmake /home/kogatti/semi_ws/logs/naoqieus/build.cmake.007.log                                                         
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "nao_interaction_msgs" with any of the following names:

    nao_interaction_msgsConfig.cmake
    nao_interaction_msgs-config.cmake

  Add the installation prefix of "nao_interaction_msgs" to CMAKE_PREFIX_PATH
  or set "nao_interaction_msgs_DIR" to a directory containing one of the
  above files.  If "nao_interaction_msgs" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "nao_interaction_msgs" with any of the following names:

    nao_interaction_msgsConfig.cmake
    nao_interaction_msgs-config.cmake

  Add the installation prefix of "nao_interaction_msgs" to CMAKE_PREFIX_PATH
  or set "nao_interaction_msgs_DIR" to a directory containing one of the
  above files.  If "nao_interaction_msgs" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/kogatti/semi_ws/build/naoqieus; catkin build --get-env naoqieus | catkin env -si  /usr/bin/cmake /home/kogatti/semi_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/kogatti/semi_ws/devel/.private/naoqieus -DCMAKE_INSTALL_PREFIX=/home/kogatti/semi_ws/install; cd -
.............................................................................................................................................
Failed     << naoqieus:cmake           [ Exited with code 1 ]                                                                                
Failed    <<< naoqieus                 [ 0.5 seconds ]                                                                                       
Abandoned <<< peppereus                [ Unrelated job failed ]                                                                              
[build] Summary: 0 of 2 packages succeeded.                                                                                                  
[build]   Ignored:   18 packages were skipped or are blacklisted.                                                                            
[build]   Warnings:  None.                                                                                                                   
[build]   Abandoned: 1 packages were abandoned.                                                                                              
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 0.5 seconds total.                         
                               
=========================================================================

k-okada, 2017-10-24:
workspaceに
```
git clone https://github.com/ros-naoqi/naoqi_interaction
```
がはいっているかな？
```
cd ~/semi_ws/src
wstool init .
wstool  scrape -y
```
としてワークスペースをwstool対応にしてから
```
wstool info
```
とすると，いまのworkspaceの情報を得ることができます．
https://github.com/jsk-ros-pkg/jsk_demos/issues/1226#issuecomment-337496943
をもう一回確認してみましょう．

koga-yu, 2017-10-25:
もう一度コマンドを実行したところうまく行きました
お騒がせしました

koga-yu, 2017-10-25:
fetch15
```
in shell
$rossetip 133.11.216.82
$rossetmaster fetch15
$source devel/setup.bash
$roseus

in roseus
$roslaunch "fetch-interface.l"
$fetch-init
$(send *ri* :move-to (make-coords :pos #f(0 7000 0) :rpy (float-vector pi 0 0)) :frame-id "/map")

$load "package://jsk_maps/src/eng2-scene.l"
$setq *scene* (make-eng2-scene)
$objects (list *scene*)
$send *scene* :spots
$(send-all (send *scene* :spots) ;:name) print all spots name
$send *scene* :spot "/eng2/7f/room73B2-trashbox-front"
$send *ri* :move-to (send *scene* :spot "/eng2/7f/room73B2-trashbox-front") :frame-id "/map"
$send *ri* :move-to (send *scene* :spot "/eng2/7f/room73B2-door-front")
```

k-okada, 2017-10-26:
rvizはcommon_msgs を持ってきてブランチをindigo-develにいｓて，全部コンパイルすると大変だから，visualization_msgsだけをsrcからシンボリックリンクをはって，buildするひつようがありますね．

で伝わるかな．．．

k-okada, 2017-10-26:
scene の spot を手元で確認する方法．

```
(load "package://jsk_maps/src/eng2-scene.l")
(setq *scene* (make-eng2-scene))
(objects (list *fetch* *scene*))
(send *fetch* :move-to (send (send *scene* :spot "/eng2/7f/room73B2-door-front") :worldcoords) :world)
(send *irtviewer* :look-all *fetch*)
```
[image]

map相対の位置を知る方法
```
(send *fetch* :move-to (send (make-coords :pos #f(-1000 7000 0)  :rpy (float-vector pi/2 0 0)) :transform (car (send *scene* :find-object "/eng2/7f"))) :world)
```
spotの位置を知る方法
```
(send *fetch* :move-to (send (send (send (send *scene* :spot "/eng2/7f/room73B2-front-of-kitchenboard") :copy-worldcoords) :transform (send (send (car (send *scene* :find-object "/eng2/7f")) :copy-worldcoords) :inverse-transformation)) :transform (car (send *scene* :find-object "/eng2/7f"))) :world)
または
(send *fetch* :move-to (send (send *scene* :spot "/eng2/7f/room73B2-front-of-kitchenboard") :worldcoords) :world)
```
実機を動かすときは
```
(send *ri* :move-to  (send (send *scene* :spot "/eng2/7f/room73B2-front-of-kitchenboard") :worldcoords))
```
でよい．(たぶん）

実機の場合`:move-to` は `:frame-id "world"` となっている．`world` は建物原点 `map`はフロア原点．
```
$ (send (send *scene* :spot "/eng2/7f/room73B2-front-of-kitchenboard")  :worldcoords)
#<coordinates #X14e726e0  1350.0 8750.0 30002.0 / 3.142 0.0 0.0>
```
となって，`z`座標に注目してこれは建物原点．

Cc: @furushchev

dyson8910, 2017-11-01:
@koga-yu 傘を掴んでかさ立てに立てる
1. 傘と傘立てのeusモデルを作る
2. 傘の柄に手を伸ばす
3. 傘立てに傘を立てる
4. 実機で実行する
5. 傘を発見するプログラムを作る
６．実機で実行する
7. 傘立てを発見するプログラムを作る
8.実機で実行する

@Naoki-Hiraoka 
1.エレベータの前に移動する
2.写真を撮ってツイッターに流す
3.画像をセーブしておく
4.ポスターを一個追加する
5.画像をセーブしておく
6.２つの画像の差分を計算する
7.差分をツイッターに流す

@takayuki5168
1.電子レンジのモデルを作る
2.レンジの取っ手に手を伸ばす
3.ドアの開閉に沿ってドアを開ける
4.弁当を中に入れる
5.ドアを閉める
6.ボタンを押す
7.実機で実行する
8.ボタンを押して中が光るのを認識する
9.中が光るまでボタンを何回も押す
10.光が消えたらドアを開ける
11.弁当を取り出す
12.弁当を持ってくる

@dyson8910 お湯を沸かして注ぐ
1.コンロのボタンを押す
2.3分待つ
3.やかんをもつ動作を作る
4.やかんを持ったまま移動する
5.カップラーメンに注ぐ
6.お湯が湧いた音を聞く
7.カップラーメンの蓋を開ける

Naoki-Hiraoka, 2017-11-01:
pr2の場合
```
roseus
load "package://pr2eus/pr2-interface.l"
pr2-init
```
 kineticに対応していないというエラーが出てきたら、
```
cd semi_ws/src
git clone https://github.com/pr2/pr2_controllers
cd ..
catkin b pr2_controllers_msgs
source ./devel/setup.bash
```

Naoki-Hiraoka, 2017-11-01:
今日は https://github.com/Naoki-Hiraoka/jsk_demos/commit/2ccb426ed7df2d2f092aa4528d301fdb03ba5492 までできました。
- 写真をとって保存するところまでできました。

dyson8910, 2017-11-01:
今日は
https://github.com/dyson8910/jsk_demos/commit/9591e39936ae92d9587beefd7e823b61e39a3618
まで進みました。
1番のコンロのボタンを押すところを座標で指定して行うところまでできました。

takayuki5168, 2017-11-01:
今日は 
https://github.com/takayuki5168/jsk_demos/commit/5e3c8d792d6960879f0cdc3a243a5054e1c9893e
までやりました。
電子レンジのモデル(扉の開閉のみ)を作ったのとロボットが移動する位置をシミュレータ上で確認しました。

koga-yu, 2017-11-01:
koga-yu/jsk_demos@a4ae5d3
までやりました
傘のモデルを作り、ロボットの前におかれた傘の柄にハンドを伸ばすところまでシミュレータで動かしました

takayuki5168, 2017-11-08:
https://github.com/takayuki5168/jsk_demos/commit/7879152b8d51d04f70dabfd011f56e6bad386578
までやりました

テンプレートマッチングの実装、realsenseを用いた動作確認(jskのライブラリを使うことになったのでこれは使わなさそうですが)、pr2の環境構築をしました。

Naoki-Hiraoka, 2017-11-08:
roskineticで、pr2(実機)をrvizで見たい場合。
rviz.launchを使ったり、jsk_pr2_startupをビルドしようとしてはいけないそうです。
rvizを普通に起動し、見たい情報を一つ一つ選択していくのが良いとのことです。

koga-yu, 2017-11-08:
https://github.com/koga-yu/jsk_demos/commit/2b6337ede0d464e109d2f4abaa479d49be1c1f8c
までやりました
傘を掴んで下ろす一連の動作を実装しシミュレータで確認、
またpr2用のコードも準備しました

Naoki-Hiraoka, 2017-11-08:
入口付近に移動
(load "package://jsk_maps/src/eng2-scene.l")
(setq *scene* (make-eng2-scene))
(send *ri* :move-to  (send (send *scene* :spot "init-spot") :worldcoords) :frame-id "eng2")
(send *ri* :state :potentio-vector)
(send *ri* :angle-vector #f(0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0) 5000)
(send *ri* :start-grasp)
(send *ri* :stop-grasp)
(send *ri* :start-grasp :larm)
(send *pr2* :reset-pose)  (send *pr2* :init-pose)
(send *ri* :angle-vector (send *pr2* :angle-vector))

dyson8910, 2017-11-08:
https://github.com/dyson8910/jsk_demos/commit/7a6bf9b72028dec8e49d76c0f922cad68c020691
までやりました。
前回のもののpr2版が作成途中で、やかんを掴んで持ち上げ、運んでくるプログラムを作成しました。

Naoki-Hiraoka, 2017-11-08:
今日は、https://github.com/Naoki-Hiraoka/jsk_demos/commit/c9021dcde20006a80254cf3f1784172090dd614e
までやりました。
作成した写真から掲示を認識するプログラムの精度が非常に悪いので、これを改良することが課題です。

dyson8910, 2017-11-15:
今日は
https://github.com/dyson8910/jsk_demos/commit/8d707a8484bf4c18197dea05c1141cf2f50df31c
までやりました。
テンプレートマッチング自体の実装はできたと思うので、今後はそこから座標を計算してコンロのボタンを押せるようにしていく予定です。

takayuki5168, 2017-11-16:
今日は
https://github.com/takayuki5168/jsk_demos/commit/20086026ea54758e43d17c08ff3c40e6a4635edd
までやりました。

PR2の実機を用いて電子レンジの目の前まで移動、電子レンジを認識、電子レンジの適当な位置にアームを動かす、またeusモデル上で認識した位置に電子レンジを配置しました。
次回はドアの開閉やボタンを押す動作を作るのと、まだ精度的にボタンを押したりドアを開けたりできそうにないので何か策を考える予定です。

Naoki-Hiraoka, 2017-11-17:
11月15日は、
https://github.com/Naoki-Hiraoka/jsk_demos/commit/6b3cfd914f7d1eddae17599ee1969b8b0559aa81
までやりました。
2枚の写真が全く同じ位置から撮影され、差分が掲示1枚だけであれば、うまく動くのではないかと思います。

Naoki-Hiraoka, 2017-11-18:
(質問)
ros::advertiseしてすぐにros::publishした時に、メッセージが送信されない場合があります。プロセス間通信の関係上すぐに送信できないのは仕方ない気はするのですが、ros::advertiseしてからどのくらい待てば確実に送信できるのか知るよい方法はありますでしょうか。
```
(ros::load-ros-manifest "roseus")
(ros::roseus "hoge")
(ros::advertise "/image_time_diff/start" std_msgs::Header 10)
(setq msg (instance std_msgs::Header :init))

(ros::rate 10)
(while (not (ros::get-topic-publisher "/image_time_diff/start"));;ここで少し工夫してみたが効果なし
  (ros::sleep))

(setq i 0)
(while (< i 10)
  (send msg :seq i)
  (send msg :stamp (ros::time-now))
  (ros::publish "/image_time_diff/start" msg)
  (incf i)
  (ros::sleep)
  )

(exit)
```

結果seq0とseq1が送信されていない。
```
$ rostopic echo /image_time_diff/start
seq: 2
stamp: 
  secs: 1510983851
  nsecs: 699853984
frame_id: ''
---
seq: 3
stamp: 
  secs: 1510983851
  nsecs: 799852981
frame_id: ''
---
seq: 4
stamp: 
  secs: 1510983851
  nsecs: 899822602
frame_id: ''
---
seq: 5
stamp: 
  secs: 1510983851
  nsecs: 999822598
frame_id: ''
---
seq: 6
stamp: 
  secs: 1510983852
  nsecs:  99817629
frame_id: ''
---
seq: 7
stamp: 
  secs: 1510983852
  nsecs: 199854365
frame_id: ''
---
seq: 8
stamp: 
  secs: 1510983852
  nsecs: 299850264
frame_id: ''
---
seq: 9
stamp: 
  secs: 1510983852
  nsecs: 399848354
frame_id: ''
---
```

furushchev, 2017-11-18:
@Naoki-Hiraoka 経験的には`(unix:sleep 1)`をすると大体大丈夫だと思います． (参考: https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L1226-L1228 )
が，正しい記述をするなら，

```lisp
(ros::advertise "/image_time_diff/start" std_msgs::Header 10)

(ros::rate 10)
(while (zerop (ros::get-num-subscribers "/image_time_diff/start"))                                               
  (ros::sleep))
```

のようになるのかなと思います．

Naoki-Hiraoka, 2017-11-18:
ありがとうございます。
```
(while (not (ros::get-topic-publisher "/image_time_diff/start"))
   (ros::sleep))
```
の代わりに
```
(while (zerop (ros::get-num-subscribers "/image_time_diff/start"))                                               
  (ros::sleep))
```
を使うとうまく行きました。

takayuki5168, 2017-11-22:
今日は
https://github.com/takayuki5168/jsk_demos/commit/889b81bf3b4c678733ed7e02cd1a849ed8c1c66c
までやりました

PR2のハンドが電子レンジの扉の先に追従するようにinverse-kinematicsが解けるようになったので次回実機で電子レンジの開閉を試します。

koga-yu, 2017-11-22:
今日は
https://github.com/koga-yu/jsk_demos/commit/032c2471a486831a6be14472b150d26c9c60670d
までやりました

傘を掴む・持ち上げる・下ろす・放すの一連の動作を実機で動かしました
rgb画像を読み込めるところまで来たので、傘を発見するプログラムの準備に移ります

Naoki-Hiraoka, 2017-11-22:
11/22は、
https://github.com/Naoki-Hiraoka/jsk_demos/commit/a12e4e63cb5210bb31d7261fc26aa8ba92447c46
までやりました。
同じ位置で写真を撮ったとしても、ノイズの影響でうまく差分が取れないということがわかったので、次はopencvのBackground Subtractionあたりを使ってみようと思います。

dyson8910, 2017-11-22:
今日は
https://github.com/dyson8910/jsk_demos/commit/a0228992388b6fac478aba1ea92fd8492563a232
までやりました。
実機を動かしたところ、シミュレーターと位置が異なっていたので、次回はpr2のpositionとの違いなどを確認してこれからの動かし方を考えていきたいと思います。

koga-yu, 2017-11-29:
https://github.com/koga-yu/jsk_demos/commit/9214dd17f00b6038b711c73c007ee6642cc47ea2
までやりました
opencvのmatchTemplateを使って傘の柄を見つけることができました
大きさが変わることに対応していないので、画像の大きさをかえつつマッチングし最も近いものから
傘への距離を計算するようにします

dyson8910, 2017-11-29:
https://github.com/dyson8910/jsk_demos/commit/aa3118365ba2b6df818c2a7afdae0cb3fba599bc
までやりました。
座標のみでの移動は再現性があまりなかったので、画像認識を使うことにしました。
画像から座標を求めるやり方を @takayuki5168 に教えてもらったので、それを利用したいと思います。

takayuki5168, 2017-11-29:
https://github.com/takayuki5168/jsk_demos/tree/54f4e64551ebebaee825331493c37a48257166f0
までやりました。

inverse-kinematicsを解くときにハンドの向きも動作に合うように直しました。
電子レンジからの相対位置に移動してロバスト性を高める方法を教えてもらったので来週それを試します。
電子レンジに入れるもの以外の動作は作り終わりまた見やすくしました。

以下は電子レンジのボタンの押して扉を開けている様子です。
[image]

Naoki-Hiraoka, 2017-12-02:
遅くなりました。11/29は、
https://github.com/Naoki-Hiraoka/jsk_demos/commit/82b8c674c44e67e8229ec017be00aeaaa095a055#diff-22a427a8819b6daaf6aa11895be7b08f
までやりました。
cv_bridge::toCvCopyの引数encodingについて、pr2-interface.lとnamespaceの関係についてを学びました。

Naoki-Hiraoka, 2017-12-06:
12/6は、
https://github.com/Naoki-Hiraoka/jsk_demos/commit/2fcdd15f4e2d9870da019c028d3f8f9511c06822
までやりました。

twitter関連では、どのような経路でtopic等が流れてtweetがなされているのかをある程度理解し、
jsk_robot_common/tweet_client.l内のtweet-string関数を参考に、
speak-jp関数がうまく動かないのでコメントアウトし、
muxにservice-callするタイミングとimage_saverにservice_callするタイミングが通信状況によって入れ替わってしまうことがあるようなので間にunix::sleepして十分に時間をあけるようにしました。

差分を計算するプログラムでは、新たに追加された物体がカメラに近いと物体に反射した光が原因で画像全体が大きく変化してしまうようなので、追加する物体はカメラから十分離すか小さくする必要があるとわかりました。

デバッグで使ったusb_camと実機のカメラではbgrとrgbが反対になっていることもあり、cv_bridge::toCvCopyの引数encodingは、opencvの保存・読み込み機能(imread,imwrite,videoCaptureなど)と合わせて使用する場合はmsg->encodingではなく"bgr8"と指定してやらないと、topicの画像がbgrかrgbかでうまく行かないことが多いと学びました。

koga-yu, 2017-12-06:
https://github.com/koga-yu/jsk_demos/commit/22f9041b8746f53d13f8cf49bcb33e4879406854
までやりました
match templateで傘の柄を見つけ、そこにハンドを伸ばす動作ができました
３次元空間内の位置計算精度が悪いのでそこを見直し、一連の動作ができるようにします

dyson8910, 2017-12-06:
https://github.com/dyson8910/jsk_demos/commit/80e0715eacb9c7a9aba0e2654a7b44fafe067c50
までやりました
画像を用いて座標を取り出すところはできたので、それを用いて動作を作っていきます。

takayuki5168, 2017-12-07:
https://github.com/takayuki5168/jsk_demos/commit/e5b574583cb5923e1cede3361e4d4e2cabf318cb
までやりました。
電子レンジを認識して電子レンジからある一定距離離れて電子レンジの扉に手を移動させるinverse-kinematicsを試しました。
point_pose_extractorの精度が良くないので次回テンプレート画像を変更します。

furushchev, 2017-12-07:
@takayuki5168 おそらく使っているカメラをもっと狭角に変えると精度がよくなるかなと思います．
来週までにはPR2のnarrow_stereoという狭角カメラが治っているはうなので，これを使ってテンプレート画像を取り直すと精度が良くなると思います．

takayuki5168, 2017-12-07:
@furushchev 
アドバイスありがとうございます。次回試してみます。

takayuki5168, 2017-12-11:
@furushchev 
iPhoneで撮影した電子レンジのパネルの写真をもとに今まで同様自己位置補正(電子レンジの位置からある相対位置にPR2を移動させる)をしたのですが、そこそこ精度がよくなったものの電子レンジを開閉できるほどの精度はありませんでした。特に複数回試した時に奥行き方向のズレの幅が6cmほどあります。error_thresholdの値を2くらいにしてもあまり効果は見られませんでした。
考えている策としてはz方向のみ何らかの方法でさらに補正をかける、シフト特徴量がよく出る(?)ARマーカーを電子レンジに貼り付ける、ですが何かアドバイスをいただけると幸いです。

参考までに、使用している画像認識用launch ファイルは以下です。
https://github.com/takayuki5168/jsk_demos/blob/master/jsk_2017_10_semi/launch/microwave-detection.launch

furushchev, 2017-12-12:
@takayuki5168 （前に聞いたことあるかも）この6cmのズレというのは認識結果そのもののズレかな？それともその認識結果をもとにeusモデルの原点を修正して、そのモデルのどこかの部位とロボットとのズレを指しているかな？
もし認識結果そのもののズレだったとして、
1. 認識結果を見てみたいです。(debug_imageという名前で出ているはず）
2. リンクのlaunchファイルを見ると`narrow_stereo`でも試した形跡があるけど、これ試してみたとしたらどれくらい変わったかな？
(0. ここにあまりこだわる理由がなさそうなら、ここは後で取り組んで、例えばレンジをもう少し高く（してカメラとの相対姿勢変化を小さく）して先に進むというのも個人的にはありだと思います。)

takayuki5168, 2017-12-12:
@furushchev 認識結果そのもののずれであっています。
では次回研究室を伺った時に認識結果を見ていただきたいです。
narrow_stereoで行った時は狭角であるため近くに来ると電子レンジを捉えられないのと誤差が大きかったのでやめました。
はい、画像認識を高度なものにするつもりはないので比較的楽な方法を積極的に考えたいです。

koga-yu, 2017-12-13:
https://github.com/koga-yu/jsk_demos/commit/7419467257ae9aabfb6226e13077967485f46125
までやりました
指の長さ分の奥行き誤差が許容できるなら大方つかめるようになりました
掴んだあとドアの前まで行き、傘立てに差し込むイメージで傘を放す動作を実装しました
次回は傘立ての位置など最終調整をします

Naoki-Hiraoka, 2017-12-13:
今日は、
https://github.com/Naoki-Hiraoka/jsk_demos/commit/5bb17efb6c0797f5c349bc9973bbf6c8108dcc1c
までやりました。
これまでは2つの画像の間で単に数十秒間待機するだけだったのですが、待機時間を自分で判断するようにしようと試みました。具体的には、beforeの画像からある程度以上変化していたら新規掲示ありとみなして待機終了するようにしようと考えました。
このとき、ヒトが新しい掲示を貼るために手や体が画面に写った瞬間に待機終了してしまってはいけないので、ヒトの体は立体物であるのに対し掲示は平面であることを利用し、rgbの画像がもとの画像からある程度以上変化しているが、depthの画像はほとんど変化していないという状態になったときに待機終了するようにしようとしました。
そのために、beforeのdepthの動画を保存しようといろいろと調べてみたのですが、videowriterのiscolor引数がwindowsでしか対応していない?からか、うまく保存できませんでした。
動画の保存はもともとbefore画像とafter画像の間で日をまたぐことを想定していたため行っていたのですが、今回の発表の場合では日をまたがないので、そもそも動画を保存せずに進めることを検討しようと思います。

708yamaguchi, 2017-12-13:
@Naoki-Hiraoka @koga-yu @dyson8910 @takayuki5168 

B4 山口です。

ゼミ机に管理番号040のThinkPadの充電器が忘れられていました。
心当たりのある方は時間のあるときにでも取りに来てください。ゼミ机に置いています。

[image]

dyson8910, 2017-12-14:
https://github.com/dyson8910/jsk_demos/commit/c8d666d69fc363f5be25282347dd8994ec1e6d06
までやりました。
画像認識でズレがあったのですが、オフセットで合わせることでコンロのボタンを押すことができました。
次回までにやかんを取る動作やお湯を注ぐ動作も確認したいと思います。

takayuki5168, 2017-12-14:
https://github.com/takayuki5168/jsk_demos/tree/592cb9d6d31fb7d8aa5ddca8c54c37076c9cd34b
までやりました。

認識カメラをkinect_headからnarrow_stereoに変更した結果認識精度がよくなりました。
電子レンジ関連のタスク(電子レンジを見つける、近づく、開ける、コップを入れる、閉める、ボタンを押す、開ける、コップを取り出す、閉める)はだいたいできるようになりました。
主にコップを取り出すところと扉を開けるときにまだミスることがあります。

次回までにこれらのバグを取り除き再現性を上げたいです。
コップに何を入れるか(飲み物またはこぼれても大丈夫なダミーのもの)悩み中です。

Naoki-Hiraoka, 2017-12-20:
https://github.com/Naoki-Hiraoka/jsk_demos/commit/b020ee98ee113709ebaeabcef7b3d12219325a19
フローチャートとシステム構成図を添付しました。

1.廊下の掲示板の前に移動する。
2.掲示板の画像を100枚記憶する。
3.「新しい掲示を待っています」という。
4.ヒトが掲示を一個追加する。
5.自動で掲示が追加されたこととヒトが去ったことを検知する。
6.新規掲示をツイッターに流す。
7.2〜6を繰り返す。
8.PCから指示を送ると帰ってくる。

Naoki-Hiraoka, 2018-01-15:
ロボットシステム課題提出のメールアドレス宛てに、ゼミのレポートを送りました。
件名は「12/20 少人数ゼミレポート (学生証番号) (名前)」です。
遅くなってしまい申し訳ありません。
よろしくお願いいたします。

koga-yu, 2018-01-30:
ゼミのスライドをアップします
大変遅れてしまい申し訳ありません
[semi_slide.odp.zip](https://github.com/jsk-ros-pkg/jsk_demos/files/1675837/semi_slide.odp.zip)

takayuki5168, 2018-02-01:
ゼミのスライドを [ここ](https://github.com/takayuki5168/jsk_demos/blob/911612c0fd576f084c04d71259cfbce5ca8db835/jsk_2017_10_semi/slide/%E5%B0%91%E4%BA%BA%E6%95%B0%E3%82%BC%E3%83%9F%20%E3%82%B9%E3%83%A9%E3%82%A4%E3%83%89.pdf) にpushしました。
遅れてしまい申し訳ありません。よろしくお願いします。

dyson8910, 2018-02-04:
@Naoki-Hiraoka 同様メールでゼミのレポートを提出しました。
遅くなってしまい、申し訳ありません。
よろしくお願いします。