jsk-ros-pkg/jsk_demos #1236: Kinetic jsk_maps deb not found

knorth55, 2018-04-12:
We cannot find `ros-kinetic-jsk-maps`...

k-okada, 2018-04-12:
here is the list of packages which is not released under kinetic
```
Could not resolve rosdep key 'jsk_robot_utils'
Failed to resolve jsk_robot_utils on ubuntu:xenial with: Error running generator: Failed to resolve rosdep key 'jsk_robot_utils', aborting.
jsk_robot_utils is depended on by these packages: ['drc_com_common']
<== Failed
Could not resolve rosdep key 'pr2_gripper_sensor_msgs'
Failed to resolve pr2_gripper_sensor_msgs on ubuntu:xenial with: Error running generator: Failed to resolve rosdep key 'pr2_gripper_sensor_msgs', aborting.
pr2_gripper_sensor_msgs is depended on by these packages: ['jsk_demo_common']
<== Failed
Could not resolve rosdep key 'pr2_navigation_self_filter'
Failed to resolve pr2_navigation_self_filter on ubuntu:xenial with: Error running generator: Failed to resolve rosdep key 'pr2_navigation_self_filter', aborting.
pr2_navigation_self_filter is depended on by these packages: ['jsk_demo_common']
<== Failed
Could not resolve rosdep key 'roseus_mongo'
Failed to resolve roseus_mongo on ubuntu:xenial with: Error running generator: Failed to resolve rosdep key 'roseus_mongo', aborting.
roseus_mongo is depended on by these packages: ['jsk_demo_common']
<== Failed
Could not resolve rosdep key 'roseus_remote'
Failed to resolve roseus_remote on ubuntu:xenial with: Error running generator: Failed to resolve rosdep key 'roseus_remote', aborting.
roseus_remote is depended on by these packages: ['drc_com_common']
<== Failed
Some of the dependencies for packages in this repository could not be resolved by rosdep.
```

k-okada, 2018-04-12:
@furushchev are we ready to release roseus_mongo on kinetic?
we're just skip releasing at https://github.com/tork-a/jsk_roseus-release/blob/master/kinetic.ignored#L2

furushchev, 2018-04-12:
@k-okada No issue from my side. I can use the package on kinetic built from source without any problem.