jsk-ros-pkg/jsk_demos #1248: 2018年度ゼミ

k-okada, 2018-10-17:
```
catkin b peppereus
```
emacs で `M-x shell`したあと，
```
source ~/semi_ws/devel/setup.bash
roscd peppereus
roseus pepper-init.l
```
EusLisp上で
```
(load "package://peppereus/pepper-interface.l")
(pepper-init)

(objects (list *pepper*))
(dotimes (i 100)
  (send *pepper* :angle-vector (float-vector 2.0 -2.0 (+ -5.0 i) 85.0 10.0 (+ -70.0 i) -20.0 -40.0 85.0 -10.0 70.0 20.0 40.0 0.0 0.0))
  (objects (list *pepper*)))

(send *ri* :angle-vector (send *pepper* :angle-vector))
```

`pepper` , `nao`, `pr2`, `fetch` などが利用可能．
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_fetch_robot も参考になる．

@ykawamura96
@konoha4303
@Yu1313321
@nakaotatsuya

ykawamura96, 2018-10-18:
@k-okada 
前回のゼミと同様に`catkin build naoeus` は成功して, naoのモデルを表示させ関節の角度を指定できましたが, `catkin build fetcheus`は以下のように失敗してしました. 
`source ~/../../opt/ros/kinetic/setup.bash `としてもダメでした. 
```
# license of fetch_description is CreativeCommons-Attribution-NonCommercial-NoDerivatives-4.0
fetch.l
fetch.dae

``````
mech-user@test1-pc:~/semi_ws/src/jsk_robot/jsk_fetch_robot$ catkin build fetcheus
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/mech-user/semi_ws
------------------------------------------------------------
Source Space:       [exists] /home/mech-user/semi_ws/src
Log Space:          [exists] /home/mech-user/semi_ws/logs
Build Space:        [exists] /home/mech-user/semi_ws/build
Devel Space:        [exists] /home/mech-user/semi_ws/devel
Install Space:      [unused] /home/mech-user/semi_ws/install
DESTDIR:            [unused] None
------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------
[build] Found '46' packages in 0.0 seconds.                                                                                                                                    
[build] Updating package table.                                                                                                                                                
Starting  >>> fetcheus                                                                                                                                                         
_______________________________________________________________________________________________________________________________________________________________________________
Errors     << fetcheus:cmake /home/mech-user/semi_ws/logs/fetcheus/build.cmake.000.log                                                                                         
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "fetch_description"
  with any of the following names:

    fetch_descriptionConfig.cmake
    fetch_description-config.cmake

  Add the installation prefix of "fetch_description" to CMAKE_PREFIX_PATH or
  set "fetch_description_DIR" to a directory containing one of the above
  files.  If "fetch_description" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "fetch_description"
  with any of the following names:

    fetch_descriptionConfig.cmake
    fetch_description-config.cmake

  Add the installation prefix of "fetch_description" to CMAKE_PREFIX_PATH or
  set "fetch_description_DIR" to a directory containing one of the above
  files.  If "fetch_description" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)

cd /home/mech-user/semi_ws/build/fetcheus; catkin build --get-env fetcheus | catkin env -si  /usr/bin/cmake /home/mech-user/semi_ws/src/jsk_robot/jsk_fetch_robot/fetcheus --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mech-user/semi_ws/devel/.private/fetcheus -DCMAKE_INSTALL_PREFIX=/home/mech-user/semi_ws/install; cd -
...............................................................................................................................................................................
Failed     << fetcheus:cmake          [ Exited with code 1 ]                                                                                                                   
Failed    <<< fetcheus                [ 1.4 seconds ]                                                                                                                          
[build] Summary: 0 of 1 packages succeeded.                                                                                                                                    
[build]   Ignored:   45 packages were skipped or are blacklisted.                                                                                                              
[build]   Warnings:  None.                                                                                                                                                     
[build]   Abandoned: None.                                                                                                                                                     
[build]   Failed:    1 packages failed.                                                                                                                                        
[build] Runtime: 1.6 seconds total.                                                                                                                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
mech-user@test1-pc:~/semi_ws/src/jsk_robot/jsk_fetch_robot$ 
``

k-okada, 2018-10-19:
そうでした．fetchの場合は
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_fetch_robot#setup-environmentの
をみてcatkin_wsをsemi_wsに読み替えて実行してみてください．
srcの下にfetch_descriptionみたいなパッケージがダウンロードされるはずです

708yamaguchi, 2018-10-19:
@ykawamura96 
基本的には以下のURLのSetup Environmentを参考にしてもらったらいいのですが、kineticの場合は`jsk_fetch.rosinstall`(indigo用)の代わりに`jsk_fetch.rosinstall.kinetic`を使ってもらうといいと思います。
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_fetch_robot#setup-environment

ykawamura96, 2018-10-19:
できました。ありがとうございます。
[image]

konoha4303, 2018-10-24:
幕内です。
naoとpepperのファイティングポーズを作りました。
[image]
[image]

```
(load "package://peppereus/pepper-interface.l")
(pepper-init)

(objects (list *pepper*))
(dotimes (i 100)
  (send *pepper* :angle-vector (float-vector 2.0 -2.0 (- -5.0 i) 85.0 10.0 (- -70.0 i) -20.0 -40.0 85.0 -10.0 70.0 20.0 40.0 0.0 0.0))
  (objects (list *pepper*)))

(send *pepper* :angle-vector (float-vector 15 -5 0 20 -45 -89.5 -20 -40 85 -10 70 20 -30 -20 10))
  (objects (list *pepper*)))

(send *ri* :angle-vector (send *pepper* :angle-vector))

(send *pepper* :rarm :shoulder-p :joint-angle 5)
(send *pepper* :rarm :shoulder-r :joint-angle -45)
(send *pepper* :rarm :elbow-p :joint-angle 30)
(send *pepper* :larm :shoulder-p :joint-angle 30)
(send *pepper* :larm :shoulder-r :joint-angle 0.5)
(send *pepper* :larm :elbow-p :joint-angle -89.5)
(send *pepper* :head :neck-y :joint-angle -20)
(send *pepper* :head :neck-p :joint-angle 10)
(objects (list *pepper*))

(send *ri* :angle-vector (send *pepper* :angle-vector))
```

Yu1313321, 2018-10-24:
戒田です。
pepperは腕振りをします。
[image]

https://gist.github.com/Yu1313321/d006518517d35c9f742ad2847b2025b5

naoはスクアットをします。
[image]

https://gist.github.com/Yu1313321/00e91509ea352e1dd783beaf5be3b7a5

fetchは首振りをします。
[image]

https://gist.github.com/Yu1313321/6b086923385c2d7cffabf172c748a241

ykawamura96, 2018-10-24:
河村です
*試合に勝って喜ぶ動作 
*ごろねする動作(横向きに見てください
*握手をする動作
を作りました

[image]
)
[image]
[image]

```
(load "fetch-interface.l")
(fetch-init)

(objects (list *fetch*))

(defun shake-hand ()
  (dotimes (i 300)
    (setq dest-y (* (cos(/ i 20.0)) 20.0))
    (setq dest-z (+ 900 (* (cos (/ i 20.0)) 100.0)))
    (setq dest-pos (float-vector 800 dest-y dest-z))
    (send *fetch* :rarm  :inverse-kinematics (make-coords :pos dest-pos :degug-view t))
    (objects (list *fetch*))
    (format t "dest-pos = ~A ~%" dest-pos)
    (format t "angle-vector = ~A ~%" (send *fetch* :angle-vector))
    )
  )

  ;;(send *fetch* :torso :waist-z :joint-angle i)

```
```
(load "nao-interface.l") 
(nao-init)

(objects (list *nao*))

(setq goron (float-vector -40.0 10.0 -100.0 -88.0 -60.0  90.0 -40.0 50.0 90.0 0.0  -30.0  0.0 0.0 60.0 -25.0 0.0  -30.0 -00.0 -40.0   90.0  10.0  -0.0 -30.0 5.0))

(setq *default_pose* (float-vector 110.0 10.0 -90.0 -40.0 0.0   110.0 -10.0 90.0 40.0 0.0  0.0 10.0 -30.0 60.0 -30.0 0.0  0.0 0.0 -30.0 60.0 -30.0 0.0  0.0 0.0))

(setq katimashita #f(20.0 0.0 -90.0 -80.0 0.0 20.0 0.0 90.0 80.0 0.0 -30.0 0.0 -40.0 80.0 -20.0 0.0 -30.0 0.0 -40.0 80.0 -20.0 0.0 0.0 0.0))

(dotimes (i 100)
  (send *nao* :angle-vector (float-vector 110.0 10.0 -90.0 -40.0 0.0   110.0 -10.0 90.0 40.0 0.0  0.0 0.0 -30.0 60.0 -30.0 0.0   0.0 0.0 -30.0 60.0 -30.0 0.0  0.0 0.0))
  (unix:usleep 100)
  (objects (list *nao*)))

(send *ri* :angle-vector (send *nao* :angle-vector))

(defun my-anime (pose)
  (dotimes (i 100)
    (send *nao* :angle-vector (v+ *default_pose* (scale  (/ i 100.0) (v- pose *default_pose*))))
    (format t "progress: ~A [%] ~%" i)
    (print  (v+ *default_pose* (scale  (/ i 100.0) (v- pose *default_pose*))))
    (objects (list *nao*))
    )
  )

;;(cond t((send *nao* :legs :crotch-r :joint-angle 0) (objects (list *nao*)))
;;値を表示/更新して描画も更新
(cond (t (print (send *nao* :larm :elbow-p :joint-angle )) (objects (list *nao*)) (format t "angle-vecotr: ~A ~%" (send *nao* :angle-vector))))

```

konoha4303, 2018-10-30:
@k-okada 
明日に向けて課題を進めようと思っていたのですが、```roseus```というコマンドが消えてしまったようです。
``` catkin b peppereus```を試みましたが、
```
==> Expanding alias 'b' from 'catkin b peppereus' to 'catkin build peppereus'

Profile:                     default
Extending:          [cached] /home/mech-user/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/mech-user/semi_ws

Source Space:       [exists] /home/mech-user/semi_ws/src
Log Space:          [exists] /home/mech-user/semi_ws/logs
Build Space:        [exists] /home/mech-user/semi_ws/build
Devel Space:        [exists] /home/mech-user/semi_ws/devel
Install Space:      [unused] /home/mech-user/semi_ws/install
DESTDIR:            [unused] None

Devel Space Layout:          linked
Install Space Layout:        None

Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False

Whitelisted Packages:        None
Blacklisted Packages:        None

Workspace configuration appears valid.

[build] Found '48' packages in 0.0 seconds.
[build] Package table is up to date.
Starting  >>> naoqieus                
No handlers could be found for logger "trollius"

Errors     << naoqieus:check /home/mech-user/semi_ws/logs/naoqieus/build.check.008.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "naoqi_driver" with
  any of the following names:

    naoqi_driverConfig.cmake
    naoqi_driver-config.cmake

  Add the installation prefix of "naoqi_driver" to CMAKE_PREFIX_PATH or set
  "naoqi_driver_DIR" to a directory containing one of the above files.  If
  "naoqi_driver" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "naoqi_driver" with
  any of the following names:

    naoqi_driverConfig.cmake
    naoqi_driver-config.cmake

  Add the installation prefix of "naoqi_driver" to CMAKE_PREFIX_PATH or set
  "naoqi_driver_DIR" to a directory containing one of the above files.  If
  "naoqi_driver" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/mech-user/semi_ws/build/naoqieus; catkin build --get-env naoqieus | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

Failed     << naoqieus:check           [ Exited with code 2 ]
Failed    <<< naoqieus                 [ 0.5 seconds ]
Abandoned <<< peppereus                [ Unrelated job failed ]
[build] Summary: 0 of 2 packages succeeded.
[build]   Ignored:   46 packages were skipped or are blacklisted.
[build]   Warnings:  None.
[build]   Abandoned: 1 packages were abandoned.
[build]   Failed:    1 packages failed.
[build] Runtime: 0.7 seconds total.
```

と出ました。
```naoqieus.driver```周辺のroseusに関するパッケージが消えてしまったようです。

ROSまわりで何かアクションをしたといえば、考える限りは、先日月曜日に行ったロボットシステムの授業にて、私が間違えてEusLisp言語インストールの一般ユーザ編を行ってEusLispをインストールしてしまったことと、それに伴い```.bashrc```に
```
$source ~/semi_ws/devel/setup.bash
$source ~/bashrc.eus
```
を加えたことくらいだと思っています。
```
$mkdir prog
$cd prog
$sudo apt-get install subversion gcc g++ libjpeg62-dev libtext-dev libx11-dev libgl1-mesa-dev libglu1-mesa-dev libpq-dev libpng12-dev xfonts-100dpi xfonts-75dpi msttcorefonts
$git clone http://github.com/euslisp/jskeus
$cd jskeus
$make

$mv bashrc.eus ~/
$cd ~/
$emacs -nw .bashrc
(.bashrc末尾にsource ~/bashrc.eusを追加)
```

k-okada, 2018-10-30:
ワークスペースはどうなっているかな？
wstool info
してみてください
https://github.com/jsk-ros-pkg/jsk_robot/pull/969/files
にあるようにnaoqi_driverをワークスペースに入れる必要があります

ykawamura96, 2018-10-30:
@708yamaguchi 
fetchにボールを投げる動作をさせてみたいと思い、以下のようなコードを書きました。
```
(setq recieve-pos #f(88.1373 39.5951 37.3204 -182.416 85.4712 -228.729 60.863 31.5685 0.0 0.0))
;;投げる動作と時間の設定
(setq thr1 #f(25.3106 68.8275 50.8513 -219.216 127.925 -465.502 107.636 -15.8476 0.0 0.0))
(setq thr2 #f(25.3106 68.8275 50.8513 -219.216 127.925 -465.502 107.636 -15.8476 0.0 0.0))
(setq thr3 #f(118.474 31.4167 29.9995 -206.956 96.9245 -391.023 -101.781 34.9092 0.0 0.0))
(setq throw-list (list thr1 thr2 thr3))
(setq throw-tm-list (list 1000 1000 1000))

;;受け取るポジションに腕を出す
(defun recieve ()
  (send *ri* :stop-grasp)
  (send *ri* :angle-vector recieve-pos)    
  )

;;手を閉じる
(defun grab ()
  (send *ri* :start-grasp)
  )

;;投げる
(defun throw ()
  ;;  (send *ri* :go-pos 0.3 0 0)
  ;;go-posすると、シミュレーター上では進んでから腕を動かし始める  
  (send *ri* :angle-vector thr1)
  (send *ri* :angle-vector-sequence throw-list throw-tm-list)
  (send *ri* :wait-interpolation)
  (objects (list *fetch*))
)
```

708yamaguchi, 2018-10-30:
@ykawamura96 

> 投げる際、前に進みながら腕を動かし、途中で(send *ri* :stop_grasp)をして手を放したいのですが、このような動作は可能でしょうか？

```lisp
(send *ri* :move-to (make-coords :pos #f( ... ) :rpy #f( ... )) :no-wait t)
(send *ri* :angle-vector #f( ... ) 5000)
(unix:sleep 2) ;; (send *ri* :wait-interpolation)を使わない。
(send *ri* :stop-grasp)
```
のようにすれば、前に進みだした直後に5秒かけて腕を動かし始め、2秒たったところでグリッパを開くと思います。

> シミュレーター上だと、動いたあとに腕を動かし、また、(send *ri* :stop_grasp)をどのように挟めばいいのかが分かりません。

シミュレータはkinematic simulatorかな？
シミュレータでも同じコードが動くと思いますが、少し不自然なアニメーションになるかもしれません。

k-okada, 2018-10-31:
@708yamaguchi <https://github.com/708yamaguchi> :wait-interpolation-smooth
というのがあります．
https://euslisp-docs.readthedocs.io/en/latest/pr2eus/robot-interface/
使ったことがなければ，Fetchで動くか，使ってみてください．
--
◉ Kei Okada

708yamaguchi, 2018-10-31:
`:wait-interpolation-smooth`の存在を忘れていました。
fetchでは使ったことがないので、今日時間があれば @ykawamura96 と試してみたいと思います。

nakaotatsuya, 2018-10-31:
来週は移動＋動作を組み合わせてみましょう
```
(load "package://fetcheus/fetch-interface.l")
(fetch-init)

(load "package://jsk_maps/src/eng2-scene.l")
(setq s (make-eng2-scene))
(objects (list s))
(print (mapcar #'(lambda (x) (send x :name)) (send s :spots)))

(send *ri* :move-to (send s :spot "/eng2/7f/room73B2-sink-front"))
;; (send *ri* :move-to (send s :spot "/eng2/7f/room73B2-table-front1"))
```

ykawamura96, 2018-11-07:
相対座標系で移動させるコード
```
(defun goto-tableside()
  (send *ri* :move-to (send (send (send *scene* :spot "/eng2/7f/room73B2-table-side1") :copy-worldcoords)
			       :transform (make-coords :pos #f(-300 -130 0) :rpy (float-vector -1.2 0 0))));;実機の移動
  (send *fetch* :move-to (send (send (send *scene* :spot "/eng2/7f/room73B2-table-side1") :copy-worldcoords)
			       :transform (make-coords :pos #f(-300 0 0) :rpy (float-vector -0.7 0 0))) :world);;irtvewerでの移動
  )
```

708yamaguchi, 2018-11-07:
fetchの頭に付いているマイクとスピーカ用アタッチメントの3DCADファイルです。
僕はDimensionで成形しました。
もっとカッコいいデザインにしてくれると、個人的にとても嬉しいです。

- スタンド:
  - SolidWorks用: https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch_accessories/mic_speaker_stand/mic_speaker_stand_v2.SLDPRT
  - 可視化: https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch_accessories/mic_speaker_stand/mic_speaker_stand_v2.STL

- カバー:
  - SolidWorks用:https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch_accessories/mic_speaker_cover/mic_speaker_cover_v2.SLDPRT
  - 可視化: https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch_accessories/mic_speaker_cover/mic_speaker_cover_v2.STL

708yamaguchi, 2018-11-07:
覚えている範囲でいいのですが、昨日の3年生ゼミの最中にfetchの腕が落ちる問題は発生したでしょうか？
fetch内部のログを見る限り、昨日はあまり発生していないのかな、と思っています。

デバッグの参考になるので、教えてもらえると非常に助かります。

@ykawamura96
@konoha4303
@Yu1313321 
@nakaotatsuya

Yu1313321, 2018-11-13:
fetchが部屋を出たところで忘れ物を思い出して、急いで部屋に戻るという動きを作りました。

(コード)
https://gist.github.com/Yu1313321/5f0c3bc7e5b6e9528ae28faaece1beb9

(様子)
[image]

k-okada, 2018-11-13:
もうそろそろ皆のプログラムが固まってきた気がするので，各自のプログラムを連結して一連の動作を津kつてみましょう．テンプレートは
https://github.com/jsk-ros-pkg/jsk_demos/pull/1252
に作ったので，ココにPullRequestを追加して `roslaunch jsk_2017_10_semi demo.launch` するとロボットがウゴクようにすると良いとおもいます．

明日は試験日なので，ゼミは無いですが，試験がなく時間が余っている人は73B2に来てください．
僕自身は別件でその時間はいないのですが，@mmurooka 先生, @708yamaguchi 先輩が
- [ ] workspaceにbranchを持ってくる方法，
- [ ] そこからPullRequestを送る方法，
- [ ] 自分のプログラムの追加の仕方，
- [ ] 実機がないときにkinemtaics simulatorだけでも動くように，
- [ ] 画像処理の追加方法，たとえば人の顔を見つけたら，でもプログラムが開始する，など
教えてくれるとおもいます．

708yamaguchi, 2018-11-16:
実験でmoveitを使う人もいるかと思いますが、実験が終わったら以下のコードを参考にしてcollision objectを全て消しておいてください。
その後fetchを使うと挙動がおかしくなってしまいます。

https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus_moveit/euslisp/collision-object-publisher.l#L173-L175

ykawamura96, 2018-11-16:
@708yamaguchi 
申し訳ございません。次回以降気を付けます。

konoha4303, 2018-11-21:
fetchが給湯室の前まで行って、ものをとって帰ってくるという動きを作りました。

(コード)
https://gist.github.com/konoha4303/39a7ac4bbff7e430c64502573ed9a606

(様子)
[image]

708yamaguchi, 2018-11-27:
fetchで人認識をするという話がありましたが、外部PCで常に人認識プログラム(https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/node_scripts/people_pose_estimation_2d.py )を走らせるようにしたので、
`/people_poses_to_poses/output`
や
`/people_pose_estimation_2d/output`
をsubscribeすると人認識の結果を使うことが出来ます。

@ykawamura96
@konoha4303
@Yu1313321
@nakaotatsuya

ykawamura96, 2018-11-28:
ストーリー
- 20XX年, 宅配業者は面倒くさがりになり、73B2にまとめて荷物を持ってくるようになった. ロボットは, ７3B2に積まれた荷物を, 然るべき部屋にはこばなくてはいけない. これを任されたのが, Fetchだった. (to be continue...)

@ykawamura96 
- 人に宅配物を渡される(何も認識しない) 
- 机の上においてある宅配物を拾う ->カゴに入れる
    -  rosparamを使って荷物の数を追加する
- シール（赤とか）が貼ってある宅配物を選んで拾う  ->適切なカゴに入れる
    -  rosparamを使って荷物の数を追加する

@Yu1313321 
- よさげなカゴを買う
- 部屋の前に移動する  (rosparamを使って, 持っているものに応じて行く部屋を分ける)
- 適切なカゴを掴む

@nakaotatsuya 
- 認識せずにノックする (何も認識しない)
- 壁を認識してノックする
- ドアが開くまでノックし続ける (開いたら↓へ)

@konoha4303 
- 持っているものを渡す (何も認識しない)
- 人の顔が見えたら渡す
- 人のポーズを認識して、手のところにカゴ渡す

Yu1313321, 2018-11-30:
fetchの台車に板を取り付けて、箱を３つ置けるようにしました。

[image]

ykawamura96, 2018-12-06:
- ハマりポイント
    - 取得したdepthやrgbやpoint cloudのheaderにはtime stampがくっついていて, 処理した結果にもtime sampがくっつている. 複数のtopicをsubscribeして画像処理する際は, これらのtopicのtime stampが完全に一致 (exact sync)していないと, call backが呼ばれないものもある. 複数のtopicがsyncしているかは,  `rosrun jsk_topic_tools is_synchronized honya hoge etc...` で確認できる. デフォルトだと, exact syncしているかを返すが, -a (approximate)オプションをつけると, ゆる〜くsyncしているかが確かめられる. どのくらい緩くかは, [ここらへん](https://github.com/jsk-ros-pkg/jsk_common/blob/878933abd216562c8ae38486131a0d3610dee519/jsk_topic_tools/scripts/is_synchronized#L110)を見れば良くて, 多分引数に何も指定しないと0.1秒になるっぽい. 
   - MultiPlaneExtarcionを用いて, 机の上の箱たちを検出しようと思ったら, なにも出力されなくて, async=trueにしたら直ったのでメモしておきます.
```
mech-user@test1-pc:~/semi_ws/src/jsk_demos/jsk_2018_10_semi/launch$ rosrun jsk_topic_tools is_synchronized /head_camera/depth_registered/points /HSI_color_filter_2/multi_plane_estimate/output /HSI_color_filter_2/multi_plane_estimate/output_polygon /HSI_color_filter_2/multi_plane_estimate/output_coefficients  -a
****************** Test Condition ******************
sync_policy: Approximate
topics: 
  - /head_camera/depth_registered/points [sensor_msgs/PointCloud2]
  - /HSI_color_filter_2/multi_plane_estimate/output [jsk_recognition_msgs/ClusterPointIndices]
  - /HSI_color_filter_2/multi_plane_estimate/output_polygon [jsk_recognition_msgs/PolygonArray]
  - /HSI_color_filter_2/multi_plane_estimate/output_coefficients [jsk_recognition_msgs/ModelCoefficientsArray]
timeout: 5 seconds
queue_size: 100
slop: 0.1
****************************************************
listening these topics for at most 5 seconds in rostime
synchronized in 0.4 seconds
these topics are: 'synchronized'
```

ykawamura96, 2018-12-06:
- ゼミの進捗
    - 机の上においた箱の検出
 - 流れ
    - 机の点群だけcolor filterでとってくる -> 机の平面を検出 -> 平面の上にある点群をとってくる -> とってきた点群と検出した平面をClusterPointIndicesDecomposerに入れる
 - その他
    - jsk_rviz_pluginsをつかって, bounding boxの場所に "box found ↓" てきなマーカを表示させた
    - 本当はbounding boxの真上にマーカーを出したかったのだけど, scriptはpythonで書いてて, eusのtf的な感じでぱぱっと座標変換できなさそうな感じだったので, bounding boxから諦めて適当に値を決め打ちした. 
    - 余裕があったらeusで書き直すか, pythonにもtf的なものがあるらしいのでそれを使ってみたいと思った. 
[image]

Yu1313321, 2018-12-12:
ちょっと窮屈だけど、箱3つとも腕の可動域に入っているので全部取れると思います。
実際取れるかは今日試してみます。

ykawamura96, 2018-12-12:
fetchのrobot_radiusを変えるスクリプト. 終わったら再起動すれば証拠が残らない(はず)
```
for a in /move_base/global_costmap/inflater/inflation_radius /move_base/global_costmap/inflater/robot_radius /move_base/global_costmap/robot_radius /move_base/local_costmap/inflater/inflation_radius /move_base/local_costmap/inflater/robot_radius /move_base/local_costmap/robot_radius /safe_teleop_base/local_costmap/inflater/inflation_radius /safe_teleop_base/local_costmap/inflater/robot_radius /safe_teleop_base/local_costmap/robot_radius  ; do
    echo $a
	rosparam get $a
	rosparam set $a 0.5
done
rosnode kill move_base
```

Yu1313321, 2018-12-12:
```
for a in /move_base/global_costmap/footprint /move_base/local_costmap/footprint /safe_teleop_base/local_costmap/footprint ; do                 
#/move_base/global_costmap/inflater/inflation_radius /move_base/global_costmap/inflater/robot_radius /move_base/global_costmap/robot_radius /move_base/local_costmap/inflater/inflation_radius /move_base/local_costmap/inflater/robot_radius /move_base/local_costmap/robot_radius /safe_teleop_base/local_costmap/inflater/inflation_radius /safe_teleop_base/local_costmap/inflater/robot_radius /safe_teleop_base/local_costmap/robot_radius  ; do
 echo $a
    rosparam get $a
    rosparam set $a '[[0.141315,-0.25],[0.092705,-0.285317],[-0.092705,-0.285317],[-0.242705,-0.176336],[-0.3,2.842171e-17],[-0.242705,0.176336],[-0.092705,0.285317],[0.092705,0.285317],[0.141315,0.25],[0.5,0.25],[0.5,-0.25],[0.141315,-0.25]]'
done
rosnode kill move_base
```

ykawamura96, 2018-12-13:
メモ
/camera_remote以下のtopicをsubscribeすることで、point cloudのhzが出ていないことが原因で平面検出できていない問題が解決し、実機で箱の認識までできました。
[image]