jsk-ros-pkg/jsk_demos #1255: 2018ゼミ, 春休み中のTODOリスト

ykawamura96, 2019-02-12:
河村です。
期末試験も終わり春休みにはいりましたので、ゼミ活動を再開したいと思います。
自分が担当する箇所ですが

- [ ] 荷物を認識して
- [ ]

708yamaguchi, 2019-02-13:
@ykawamura96 

河村くん

山口です。
僕は2/24以降はかなり暇になるので、手伝えると思います。

k-okada, 2019-02-14:
@ykawamura96 上の４つについてそれぞれ，どういうプログラムが必要か，それはどこにありそうかリストを拡張してもらえればとおもいます．特に認識周りはjsk_recognitionなどから似たようなものがあるはずで，なければなにかおかしいので，自分で作る前に，聞いてみましょう．

また，ストーリー全体で何が残っているか書いてみましょう．

ykawamura96, 2019-02-14:
リストを拡張しました。

また, 今日は 
1. 拾い上げた荷物の各面を目視する動作 (上)
2. 拾い上げた荷物をカゴに落とす動作を作りました (下)

[image]
[image]

ykawamura96, 2019-02-15:
1) ハンドで持った荷物周辺のpoint cloudを抽出するためには, Attention Clipperを使うと良いということを @pazeshun 先輩に教えていただきました. 
    - 荷物周辺のpoint cloudだけ取り出せたので, 後はここにcolor filterをかければ荷物の宛先がわかるようになります. 
    - メモ: Attention Clipperが出力するのは, Point Indices (3次元のマスクのようなもの)で, extract_indicesと組み合わせると, point cloudが取り出せる. よくわからない時は, sampleコードを見るよい. 
2) 荷物のカゴ部分をcollision objectとして, 避けて腕を動かすようにしました. collisionオブジェクトは, 腕を動かす動作を終わらせると, 消えるようになっています. 
[image]

[image]

Yu1313321, 2019-02-20:
戒田です。
自分の担当するところですが、
```
1：部屋までの移動
2：かごを持ち上げる
```
です。
これまでにできていることは
```
「１：部屋までの移動」
「２：かごを持ち上げる」の1つ目のかごにアームを伸ばす
```
です。
かごを持ち上げるためには、「指先を工夫する」か「持つ時のアームの角度を調整する」かが必要そうなので、まずそこを解決したいと思います。

708yamaguchi, 2019-02-21:
ubuntuをインストールする方法
http://wiki.ros.org/kinetic/Installation/Ubuntu

708yamaguchi, 2019-02-21:
@YutoUchimi @Yu1313321 
`LineSegmentDetector`を使おうと思って、`jsk_pcl_ros/sample/sample_line_segment_detector.launch`を立ちあげたのですが、``というtopicの中身が空(?)です。どうしたらいいでしょうか。

Markerが見えず、pointcloudしか見えない図。
[image]

rostopic echoのログ
```
$ rostopic echo /line_segment_detector/debug/line_marker
header: 
  seq: 5407
  stamp: 
    secs: 1486279001
    nsecs:  53697817
  frame_id: right_hand_camera_rgb_optical_frame
ns: ''
id: 0
type: 5
action: 0
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
scale: 
  x: 0.01
  y: 0.0
  z: 0.0
color: 
  r: 1.0
  g: 0.0
  b: 0.0
  a: 1.0
lifetime: 
  secs: 0
  nsecs:         0
frame_locked: False
points: []
colors: []
text: ''
mesh_resource: ''
mesh_use_embedded_materials: False
---
```

708yamaguchi, 2019-02-21:
@ykawamura96 

毎回fetchの実機で認識をするのは大変だし、rosbagを撮って自分のPCでplayするのもめんどくさいことがあると思います。
そういうときは、`dump-loadable-structure`という関数を使うと、euslispの変数をファイルに保存できます。
例:
```lisp
(setq cube (make-cube 100 100 100))
(dump-loadable-structure "cube.l" cube)
```

nakaotatsuya, 2019-02-21:
中尾です。帰省中でゼミになかなか行けていなくてすいません。
自分の担当するところですが、

-  ノックする
-  部屋の前まで行き、ドアを認識する

の予定です。
また、ドアをノックして、留守の場合に何か別の動作をすのも面白いかなと思い、時間があれば行いたいと思います。

YutoUchimi, 2019-02-21:
@Yu1313321 @708yamaguchi 
報告ありがとうございます。
線検出を行った結果、もしその線をサポートする点の数が少なすぎると線としての信頼度が低いので、`LineSegmentDetector`では線を構成する点の数がある一定数`min_indices`より少ない時は除外するという仕組みになっています。

`sample_line_segment_detector.launch`を読むと、RGB-Dカメラから得られる点群(トピックで言うと`/right_hand_camera/depth_registered/points`)を`voxelgrid`というノードによって間引いてから線検出を行っているようですね。
つまり、線検出の入力に使う点群の点の数がそもそも少なくなっていて、線を構成する点の数が`min_indices`を下回りやすくなっています。

で、どうやったら線がちゃんと出てくるのかという話ですが、
```
roslaunch jsk_pcl_ros sample_line_segment_detector.launch
```
を起動した後、別のターミナルで
```
rosrun rqt_reconfigure rqt_reconfigure
```
として`line_segment_detector`のパラメータを変更してみるといいと思います。
たぶん`min_inliers`を小さくするとrvizでマーカーが見えるようになると思います。
他のパラメータもぜひいじってみてください。
[image]

Yu1313321, 2019-02-23:
担当しているところの詳細を考えました。

まず箱を取る動作ですが、これは箱の位置が決まっているため、認識を用いる必要もなく、決め打ちの制御で事足りると思います。
ただ今の状態では箱の手を伸ばして縁をつかめても、うまく持ち上げることができないので、fetchの指か箱に細工をしないといけません。(指の方を変えると他の人の影響がありそうなのでおそらく箱をいじります。)

次に部屋の前までの移動は既に完成しています。

最後にドアを開ける動作を追加の課題で行おうと思います。
ドアノブを認識して、そこに手を伸ばしてドアノブを掴み、ドアノブを回して、ドアを引くという動作をすればいいと考えており、とりあえずドアノブの認識について調べています。
jsk_pcl_rosからいくつかのパッケージを候補にあげており、現在それらを@708yamaguchi さんや @YutoUchimi さんに手伝っていただきながら、動作確認をしている状況です。

また学科PCを返却してしまったため、新しいPCのセットアップも行いました。
以下のURLに沿うとうまくいきました。
https://github.com/jsk-ros-pkg/jsk_demos/issues/1255#issuecomment-465892831

nakaotatsuya, 2019-02-27:
僕の担当するところですが、
- [ ]  ドアを検出する
  - OrganizedMultiPlaneSegmentation(https://jsk-recognition.readthedocs.io/en/latest/jsk_pcl_ros/nodes/organized_multi_plane_segmentation.html) を使って、平面検出する
- [ ] ノックする
  - レンチコントローラ(https://github.com/fetchrobotics/fetch_robots/pull/23#issuecomment-400419673 )を使って、ノックする
- [ ] 音声認識をして、宅配の際に簡単な会話をする
  - SpeechRecognition(https://github.com/jsk-ros-pkg/jsk_3rdparty/tree/master/respeaker_ros ) を用いて簡単な受け答えをし、人間側のYes/Noの答えかたによって動作を変えてみる。

708yamaguchi, 2019-02-27:
@nakaotatsuya 
fetchの中では`/speech_to_text_google`というtopicが出てて、これをsubscribeすれば手軽に音声認識できるのですが、単純にrostopic echoすると表示がおかしくなる(rostopic echoはstring[]型のトピックを表示できない)ので、以下のプログラムを使ってtopicを出し直すといいと思います。

https://gist.github.com/708yamaguchi/b2d82fc7557ba5fc3512c9427df32be2/edit

708yamaguchi, 2019-02-27:
fetchでwrench controlを使いました。
人間とハイタッチするときには手を止めて、人間が手を離すと再びfetchが手を動かし出すデモ動画です。

[image]

708yamaguchi, 2019-02-27:
撮影したビデオは、gifファイルに変換するとgithubに投稿できます。以下のプログラムを見てもらって、
```
./convert_to_gif.sh
```
してもらうと出来ます。
https://gist.github.com/708yamaguchi/d0fbf5a670060bffc865d1bc68d94b67

708yamaguchi, 2019-02-27:
gifファイルを回転したいときは、
```
convert -rotate 90 input.gif output.gif
```
のようにすると良いです。

ykawamura96, 2019-03-09:
- 箱を取り上げて色を確認し、カゴに入れる動作が一応完成しました
    - ikが解けないことが多いので、いい感じの初期位置を与えたり、箱の持つ向きを変の長さにおおじて変えるなど、工夫するべきところはまだ残っています
    - 一連で箱をストアするところまでの動画が取れなかったので、次回来た時はしっかりした動画を撮りたいです
- 箱の色の認識はcolor filterをかけた後、euclidean clusteringをかけ、output の cluster_numが1以上かどうかを調べて行っています。
 - 腕が落ちるまで上手く行った動画↓  
     - https://drive.google.com/open?id=1zzmVh9brybFELss4TI3bdgacCoKguwrR

708yamaguchi, 2019-03-09:
@ykawamura96 
動画、良い感じだと思います。
動画と一緒にrosbagも撮っていたと思うので、自分への備忘録も兼ねてリンクを貼っておきましょう。

k-okada, 2019-03-11:
>> 箱を取り上げて色を確認し、カゴに入れる動作が一応完成しました
>>> ikが解けないことが多いので、いい感じの初期位置を与えたり、箱の持つ向きを変の長さに応じて変えるなど、工夫するべきところはまだ残っています

ikが解けない==ハードウェア的に手が届かない，かどうかはよく確認しないと，もしかしたら
```
(:inverse-kinematics (&rest args) nil)
```
みたいなプログラムになっているのを中身見ずに使って，ikが解けません，といっているとかっこわるいので，解けない時には
1. jskeus がログをダンプしているから，それを取り出しておく
2. ハードウェア的に届かない位置なのか考えてみる
3. 手先の拘束を６軸にする必要があるか，よく考える．（ここは大きな研究のチャンスあり！）
4. fetchの場合は基本moveitを使っているので，eusのIKに依存しなくてよいはず
5. 最近の新しいeusのikを勉強してみる．https://github.com/jsk-ros-pkg/jsk_control/tree/master/eus_qp/optmotiongen

ぐらいは視野に入れておきましょう．

ykawamura96, 2019-03-14:
@k-okada @708yamaguchi 
適当な初期値を与え、角度の制限を緩めることで、ikを解きやすくしてデモの動画をとりましした。
デモの動画は以下です。
robugfileは、
 ```
 roslaunch jsk_2018_10_semi play.launch bagfile:=/home/mech-user/.ros/2019-03-14-19-23-01.bag
(bugfileの場所は適当な場所を指定する)
 ```
別のターミナルで
```
roslaunch jsk_2018_10_semi delivery_box_detection.launch 
```
別のターミナルで
```
roscd jsk_2018_10_semi/
rviz -d ./rviz/fetch_config.rviz 
```
とすれば見れると思います。
rviz : https://drive.google.com/open?id=1Jsq49kxQSDwiapBp6op9HENgM6Lhx23i
カメラ : https://drive.google.com/open?id=1Pf0DzdilXCnfmzHdY82-IHnhgLfykJzi
rosbug: https://drive.google.com/open?id=19ES85EJFZkEjj6u2Ldz6mgTGmsQgVyrK

708yamaguchi, 2019-03-14:
物品2つでデモが出来てよかったですね。

ちょっと気になったんだけど、rvizの動画とカメラの動画は違う実験のものかな？
できれば同じ実験のrviz動画とカメラ動画があるといいですね。

最後に、(もうしてるかもしれないけど)デモをした時のコードをpull request(https://github.com/jsk-ros-pkg/jsk_demos/pull/1252 )に追加しておくと良いと思います。

konoha4303, 2019-03-16:
幕内です。
ゼミに全く参加をしておらず、本当に申し訳ありません。

僕の担当する箇所ですが
- [ ] 持ったカゴを人に渡す
カゴを持った状態で移動したのちに腕を伸ばして所定の位置にカゴを送り、掴んでいたグリップ部分を開く。
一定時間後にグリップを閉じて、その時にriに搭載されているmethod(:gripper)を用いてグリップの状態を調べ、カゴが手から離れたらアームをしまって次の動作に移る。

となっています。
また、部活動の関係で春休み中での今後の参加もかなり厳しいです。
せっかくゼミに参加させてもらっているのにこのような事になってしまい、申し訳ありません。