jsk-ros-pkg/jsk_demos #1272: 2019年度ゼミ

k-okada, 2019-10-16:


Kanazawanaoaki, 2019-10-16:
ゼミは自由。
毎週水曜日
だいたい年度中にビデオを作って終わる。

１２月の末　１８か２５に忘年会。無料。ちやほやされるのは今のうち。
１月以降もやる事ができる。

ロボットで鞄を開けている。宇宙チャレンジ。JAXAとのプロジェクト。できたら缶バッチとかもらえる。

料理を作る。
絵をかけたら、キャラ弁作るロボット。

農業ロボット。
デンソー農業ロボット。トマトロボット。儲かる。
プチトマトを栽培する。

k-okada, 2019-10-16:
- https://github.com/jsk-ros-pkg/jsk_demos/pull/1270/files#diff-bcb78c17a9e39f6f2a95863efdc2cd80
- https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/PR2_joints.pdf

mqcmd196, 2019-10-16:
農家実は儲かっている？ロボットで代替したら勝ち？
エンドエフェクタを工夫することによってロボットの動きを単純化

帰り道にこれは難しいかどうか考えてみよう

１０月中に一度ロボット動かす！！！

MiyabiTane, 2019-10-16:
＜To Do＞
・シナリオ作り
・まずはロボットを動かしてみる（ロボットに何ができるか）
・まず大きい動作から（動く、話す）

k-okada, 2019-10-16:
23日までの宿題
- シナリオをUMLで書く
- room73b2 のプログラムを実行してみる
- send *ri* :angle-vector を使ってみる
- ロボットを決めてeusモデルを生成する

k-okada, 2019-10-16:
CORAL @knorth55 melodic  https://github.com/knorth55/coral_usb_ros

k-okada, 2019-10-16:
ロボットモデルの作り方
```
source /opt/ros/melodic/setup.bash
mkdir -p semi_ws/src
cd semi_ws/src
wstool init
wstool merge https://gist.githubusercontent.com/k-okada/db02de337e957d482ebb63c7a08a218b/raw/20b19f3ad92a8576510c0e0aa02bcd311b347beb/semi.rosinstall
wstool update
rosdep install --from-paths . --ignore-src -y -r
cd ..
catkin build -vi
source devel/setup.bash
```
とすると以下のプログラムでロボットのモデルを作ることが出来ます．
```
(load "package://peppereus/pepper.l")
(setq *pepepr* (pepper))
(objects (list *pepper*))

(load "package://naoeus/nao.l")
(setq *nao* (NaoH25V50))
(objects (list *nao*))

(load "package://baxtereus/baxter.l")
(setq *pepepr* (baxter))
(objects (list *baxter*))

(load "package://fetcheus/fetch.l")
(setq *pepepr* (fetch))
(objects (list *fetch*))

(load "package://pr2eus/pr2.l")
(setq *pepepr* (pr2))
(objects (list *pr2*))
```

knorth55, 2019-10-16:
CORAL環境のセットアップのしかた
https://github.com/knorth55/coral_usb_ros#install-the-edge-tpu-runtime のインストール を行う
```
echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt-get update
sudo apt-get install libedgetpu1-max # If you do not have USB3, install libedgetpu1-std
sudo apt-get install python3-edgetpu
```

https://github.com/knorth55/coral_usb_ros#install-just-the-tensorflow-lite-interpreter のインストール を行う
```
sudo apt-get install python3-pip
wget https://dl.google.com/coral/python/tflite_runtime-1.14.0-cp36-cp36m-linux_x86_64.whl
pip3 install tflite_runtime-1.14.0-cp36-cp36m-linux_x86_64.whl
```

https://github.com/knorth55/coral_usb_ros#workspace-build-melodic
をみてワークスペースを作成しコンパイルする
```
source /opt/ros/melodic/setup.bash
mkdir ~/coral_ws/src
cd ~/coral_ws/src
git clone https://github.com/knorth55/coral_usb_ros.git
wstool init
wstool merge coral_usb_ros/fc.rosinstall.melodic
wstool update
rosdep install --from-paths . --ignore-src -y -r
cd ~/coral_ws
catkin init
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
catkin build -vi
```
https://github.com/knorth55/coral_usb_ros#model-download モデルをダウンロードする
```
source /opt/ros/melodic/setup.bash
source ~/ros/coral_ws/devel/setup.bash
roscd coral_usb/scripts
python download_models.py
```
カメラノードを立ち上げる
```
source /opt/ros/melodic/setup.bash
rosrun usb_cam usb_cam_node
```
認識ノードを立ち上げる
```
source /opt/ros/melodic/setup.bash
source ~/ros/coral_ws/devel/setup.bash
roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/usb_cam/image_raw
```

表示ノードを立ち上げる
```
source /opt/ros/melodic/setup.bash
rosrun image_view image_view image:=/edgetpu_object_detector/output/image
```

knorth55, 2019-10-16:
D1の北川です。Coralの動かし方で困ったらここに投稿してください。対応します。

mqcmd196, 2019-10-21:
> ロボットモデルの作り方
>
> ```
> source /opt/ros/melodic/setup.bash
> mkdir -p semi_ws/src
> cd semi_ws/src
> wstool init
> wstool merge [...]

pr2eusがなくてpepper-interface.lのロードでエラーが出ました

MiyabiTane, 2019-10-21:
pr2eusがないといわれたのでgit cloneしました。
```workspace: /home/mech-user/semi_ws/src

 Localname                     S SCM Version (Spec)    UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                     - --- --------------    -----------  ---------------------------
 pepper_robot                    git master  (-)       efad3979b374 github.com/ros-naoqi/pepper_robot
 pepper_meshes                   git master  (-)       abe1c6ea9d3e github.com/k-okada/pepper_meshes.git
 naoqi_driver                    git kochigami-develop d77dddea5ba8 github.com/k-okada/naoqi_driver
 naoqi_dashboard                 git kochigami-develop 116484fc41bf github.com/kochigami/naoqi_dashboard
 naoqi_bridge_msgs               git kochigami-develop d7417613690e github.com/kochigami/naoqi_bridge_msgs
 naoqi_bridge                    git kochigami-develop 57003d058d22 github.com/kochigami/naoqi_bridge
 nao_robot                       git master  (-)       67476469a137 github.com/ros-naoqi/nao_robot
 nao_meshes                      git master  (-)       fe4c40129dfc github.com/k-okada/nao_meshes.git
 nao_interaction                 git master  (-)       f97ad12f3896 github.com/ros-naoqi/nao_interaction
 libqicore-release               git melodic-devel     fa78dfa2b6ed github.com/k-okada/libqicore-release.git
 libqi-release                   git melodic-devel     2771a10251d1 github.com/k-okada/libqi-release.git
 jsk_robot                       git master            28970fb14919 github.com/jsk-ros-pkg/jsk_robot.git
 fetchrobotics/fetch_ros         git melodic-devel     c10411b495c9 github.com/fetchrobotics/fetch_ros.git
 RethinkRobotics/baxter_common   git kinetic-devel     eabdaba4fbb6 github.com/RethinkRobotics/baxter_common.git

Also detected these repositories in the workspace, add using 'wstool scrape' or 'wstool set':

 Localname  SCM   URI  (Spec) [http(s)://...]
 ---------  ---   ---------------------------
 jsk_pr2eus --git https://github.com/jsk-ros-pkg/jsk_pr2eus.git

```

https://github.com/jsk-ros-pkg/jsk_pr2eus

MiyabiTane, 2019-10-21:
<fetch-interfaceをつかうためにやること>

mech-user@KI00112:~$ cd ~/semi_ws/
mech-user@KI00112:~/semi_ws$ cd src/
ここでjsk_pr2eusが存在することを確認し
mech-user@KI00112:~/semi_ws/src$ rosdep update
mech-user@KI00112:~/semi_ws/src$ rosdep install --from-paths . -i -y -r
mech-user@KI00112:~/semi_ws/src$ cd ..
mech-user@KI00112:~/semi_ws$ catkin build

mqcmd196, 2019-10-21:
```python
print("Hello world.")
```

Kanazawanaoaki, 2019-10-21:
```c
printf("%d")
```

knorth55, 2019-10-21:
# NAOの動かし方

## Nao python SDKのインストール

ここをみて SDKをインストール、アカウントを作る必要が有ります
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment

## nao wsのビルド

```
source /opt/melodic/setup.bash
cd ~
mkdir nao_ws/src -p
cd nao_ws/src
wstool init
wstool merge https://gist.githubusercontent.com/knorth55/c00cba7d7b8ee98ba4c4939c36bdaa6b/raw/dd8ad7552d160f97b873e1d2922b74ac66fe44de/nao.rosinstall
rosdep install --ignore-src --from-path . -y -r -i
cd ~/nao_ws
catkin build
```

## ネットワーク設定

こんな感じで設定して、naoとethernetケーブルでつなげる

[image]

## ROS launchする

```
rossetip
rossetmaster localhost
export NAO_IP=169/254.20.238
source /opt/melodic/setup.bash
source ~/nao_ws/devel/setup.bash
roslaunch jsk_nao_startup jsk_nao_startup.launch
```

## Euslispで動かす

```
rossetip
source /opt/melodic/setup.bash
source ~/nao_ws/devel/setup.bash
roscd naoeus
roseus nao-interface.l
# (nao-init)
# (objects (list *nao*))
# (send *ri* :move-pos 0.1 0 0) # 10cm 前にいく
```

MiyabiTane, 2019-10-21:
ありがとうございます！！

mqcmd196, 2019-10-22:
> # NAOの動かし方
> ## Naoqi python SDKのインストール
> ここをみて SDKをインストール、アカウントを作る必要が有ります
> https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment
> 
> ## nao wsのビルド
> ```
> source /opt/melodic/setup.bash
> cd ~
> mkdir nao_ws/src -p
> cd nao_ws/src
> wstool init
> wstool merge https://gist.githubusercontent.com/knorth55/c00cba7d7b8ee98ba4c4939c36bdaa6b/raw/dd8ad7552d160f97b873e1d2922b74ac66fe44de/nao.rosinstall
> rosdep install --ignore-src --from-path . -y -r -i
> cd ~/nao_ws
> catkin build
> ```
> 
> ## ネットワーク設定
> こんな感じで設定して、naoとethernetケーブルでつなげる
> 
> [image]
> 
> ## ROS launchする
> ```
> rossetip
> rossetmaster localhostI put rviz output for video:
> 
> export NAO_IP=169.254.20.238
> source /opt/melodic/setup.bash
> source ~/nao_ws/devel/setup.bash
> roslaunch jsk_nao_startup jsk_nao_startup.launch
> ```
> 
> ## Euslispで動かす
> ```
> rossetip
> source /opt/melodic/setup.bash
> source ~/nao_ws/devel/setup.bash
> roscd naoeus
> roseus nao-interface.l
> # (nao-init)
> # (objects (list *nao*))
> # (send *ri* :move-pos 0.1 0 0) # 10cm 前にいく
> ```

環境の違いだと思いますが，講義のとおりだと1行目は`source /opt/ros/melodic/setup.bash`かと思われます

mqcmd196, 2019-10-23:
Coral TPUですが@knorth55 さんのコマンド通りにセットアップしたら正常に動作しました．ありがとうございます．

k-okada, 2019-10-23:
nao_ws は　https://github.com/jsk-ros-pkg/jsk_demos/issues/1272#issuecomment-542757033 のサブセットでは？

knorth55, 2019-10-23:
https://gist.github.com/k-okada/db02de337e957d482ebb63c7a08a218b
これですか？そうだとおもいます。

Kanazawanaoaki, 2019-10-23:
[santa_robo.pdf](https://github.com/jsk-ros-pkg/jsk_demos/files/3760451/santa_robo.pdf)

k-okada, 2019-10-23:
30日までの宿題：各動作のアクティビティ図を作る

YoshiaAbe, 2019-10-23:
『SysMLによる組み込みシステムモデリング』
SysML = System Modeling Language・・・HW、SW、ヒトを含むシステムを記述するための言語のこと。（言語といっても図の書き方みたいなもん）
人間って、人それぞれ、抽象的なイラストの使い方がまちまちだけど、
みんながこれを学べば、抽象的な概念をより効率よく、正確に伝えることができるようになる！
例えば、学校で、先生によって、丸、四角、矢印、波線の使い方が違って、たまに解釈に困ることとかあったと思うけど、
そういうのを解決できる。
（必修言語として、小さいうちから教育すべきでは？）
図の種類としては
（１）システムに対する要求を表現・・・要求図
（２−１）階層構造を表現・・・・・・・・・ブロック定義図
（２−２）構造、振る舞いを表現・・・・・・ブロック定義図、ステートマシン図、ユースケース図など
（３）妥当性の検証・・・・・・・・・・パラメトリック図
具体的にどう使うか
（１）まず、システムに対する要求を洗い出す
　　システムに何が必要かを、依存関係も含めた、要求図で表す
　　システムと、その外部（他のシステムや使う人）との関係を、ユースケース図として表す
（２）以下の２つを並行して、複数回繰り返すと良い
　　（２−１）次に、システムの内部構造を考える
　　　ブロック定義図
　　（２−２）それと同時にシステム内部の振る舞いを考える
　　　アクティビティ図・・・大まかな処理の順序、オブジェクトの受け渡しを表す
　　　シーケンス図・・・システム間の処理のタイミングなどを明確に表す
　　　ステートマシン図・・・時間よりも、状態の変化を記述するのに適している
（３）システムに対する制約条件を繋げて書く、パラメトリック図
SysMLでモデルを作るときに気をつけること
・使用される状況からはじめ、徐々に細分化していく
・わかりやすくするためにこの言語を用いるのだから、一枚の図にまとめようとせず、目的に応じて図を分割する
・分割された図同士の関係を、割り当てなどをして明確にすること

k-okada, 2019-10-23:
```
$ git remote -v
origin	https://github.com/jsk-ros-pkg/jsk_robot.git (fetch)
origin	https://github.com/jsk-ros-pkg/jsk_robot.git (push)
$ git branch
  backup_20151106
* master
$ git remote add k-okada http://github.com/k-okada/jsk_robot
$ git remote -v
k-okada	http://github.com/k-okada/jsk_robot (fetch)
k-okada	http://github.com/k-okada/jsk_robot (push)
origin	https://github.com/jsk-ros-pkg/jsk_robot.git (fetch)
origin	https://github.com/jsk-ros-pkg/jsk_robot.git (push)
$ git push k-okada master 
```

k-okada, 2019-10-23:
- add jsk_demos to workspace
- add your repository as remote
- create your_name branch
- add your name to author
 create pull request

MiyabiTane, 2019-10-23:
~/semi_ws/src/jsk_demos/jsk_2019_10_semi/euslisp/demo.l
初めは関数をつくる。中身はだんだん濃くしていくのが良い。

MiyabiTane, 2019-10-23:
naoの動かし方です。
https://github.com/k-okada/jsk_demos/pull/38

MiyabiTane, 2019-10-28:
アクティビティ図を書いてみました。あくまで私が想像するサンタさんなのでみんなの意見を取り入れつつ変えていきたいと思ってます。
[アクティビティ図.pdf](https://github.com/jsk-ros-pkg/jsk_demos/files/3779224/default.pdf)

MiyabiTane, 2019-10-29:
# Coral TPUインストール、sudo apt-get install libedgetpu1-maxができない。

一度インストールに失敗し、やり直したところ、sudo apt-get install libedgetpu1-maxができなくなってしまいました。エラー内容は以下です。

E: ロック /var/lib/dpkg/lock-frontend が取得できませんでした - open (11: リソースが一時的に利用できません)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?

インターネットで調べて 
sudo rm /var/lib/apt/lists/lock
sudo rm /var/cache/apt/archives/lock
を実行したり、
sudo apt autoremove
を実行しようとしました（こちらも同じエラーにより実行できませんでした。）が、同じエラーが返ってきてしまいます。
解決法が分かれば教えて頂きたいです。よろしくお願い致します。

k-okada, 2019-10-29:
リブートして，それでもダメだったら，sudo rm /var/lib/dpkg/lock-frontendしましょう．
その後，sudo dpkg --configure -a が必要かも

https://itsfoss.com/could-not-get-lock-error/
--
◉ Kei Okada

MiyabiTane, 2019-10-29:
再起動してsudo apt-get updateまで行い、
sudo dpkg --configure -a　してから　sudo apt-get install libedgetpu1-max　をすることで解決しました。ありがとうございました！

k-okada, 2019-10-30:
僕も http://plantuml.com/ja/activity-diagram-beta 見て勉強しながら，ですが，書いてくれたものは以下のようなテキストを起こして，
```
@startuml

|子供|

start

:お願い事をつぶやく;                                                            

|サンタロボ|

:変装する;                                                                      

:プレゼントを知る;                                                              

:プレゼント棚に移動;                                                            

if (プレゼントがある？) then (ない)

  :私達が用意 ;     
  
  stop

else (ある)

  :プレゼントを選ぶ;                                                            

  :プレゼントをサンタ袋に入れる;                                                

  :プレゼントを運ぶ;                                                            

  :靴下を見つける;                                                              

  :プレゼントを靴下に入れる;                                                    

endif

|子供|

:プレゼントを受け取る;                                                          

end

@enduml
```
を
https://www.planttext.com/ に貼り付けると，以下が出来ました．

[image]

MiyabiTane, 2019-10-30:
## Gitやることまとめ
[github_semi.txt](https://github.com/jsk-ros-pkg/jsk_demos/files/3787549/github_semi.txt)

Kanazawanaoaki, 2019-10-30:
・動作を作って見る
eusで書いてみる。

認識とかも変数を更新するだけ、
プログラムとしては変数にしておけば、中身は問題で無い。
移動は自動運転をすれば良いだけ、
eusでかけたら、やってみれば良い。

他の人のパートについても、他のロボットでやるとか。
バックアップにもなる。

MiyabiTane, 2019-10-30:
demo.act書き加えました。
#42

MiyabiTane, 2019-10-30:
demo.act書き加えました。
k-okada#42

MiyabiTane, 2019-10-30:
naoで帽子をかぶるポーズを作ってみました。
k-okada#44

mqcmd196, 2019-11-05:
深度カメラの環境構築，及びテストプログラムを動かしたいのですがドキュメントってどこにあがっているでしょうか．機種はIntel Realsense SR300って書いてあります

k-okada, 2019-11-06:
https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/install_realsense_camera.html
を見てみて下さい

https://github.com/jsk-ros-pkg/jsk_recognition/blob/9dc1a95a11bfb3eb0b3c68e27ad6e18f6e3b415a/doc/install_realsense_camera.rst

mqcmd196, 2019-11-06:
`sudo apt-get install ros-indigo-realsense-camera` を実行しましたがパッケージが見つかりませんというエラーがデました

mqcmd196, 2019-11-06:
fetch君にreset-poseを送ったら腕が床に当たってしまった．床の干渉計算はされていない．

Kanazawanaoaki, 2019-11-06:
fetchが挨拶をしてくれるデモを作りました。
k-okada#51

YoshiaAbe, 2019-11-06:
Yoshia abe #47
speakファイルを作っただけですが。

MiyabiTane, 2019-11-06:
### Pullreq
fetchが帽子をかぶる一連の流れを作りました。
k-okada#50

knorth55, 2019-11-06:
https://github.com/k-okada/jsk_demos/pull/45
ここに今日のゼミでメモしたことを書いてください。

Kanazawanaoaki, 2019-11-06:
k-okada#52
今日のゼミのメモを作ってプルリクをしました。

MiyabiTane, 2019-11-06:
k-okada#53
README.mdに今までとっていたメモを載せました。

knorth55, 2019-11-07:
# SR300 RGB-D cameraの使い方
## librealsense2のインストール
```bash
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
sudo apt install librealsense2-dkms librealsense2-utils librealsense2-dev librealsense2-dbg
```

## Workspaceのビルド
```bash
source /opt/ros/melodic/setup.bash
mkdir ~/realsense_ws/src -p
cd ~/realsense_ws/src
git clone https://github.com/IntelRealSense/realsense-ros.git
rosdep install --ignore-src --from-paths . -y -r -i
cd ~/realsense_ws/
catkin build
```

## カメラLaunchの起動

```bash
source /opt/ros/melodic/setup.bash
source ~/realsense_ws/devel/setup.bash
roslaunch realsense_ros rs_rgbd.launch
```

## カメラ画像、点群をみてみる
```bash
rviz
# frameをcamera_linkに変更する
# PointCloud2をrvizでみる
```

Kanazawanaoaki, 2019-11-08:
@k-okada 
来週の水曜日(11/8)は月曜授業ですが午後演習が無いのでゼミに行きたいです。

knorth55, 2019-11-10:
> fetch君にreset-poseを送ったら腕が床に当たってしまった．床の干渉計算はされていない．

この問題ですが、joy controllerでのtuck armで腕が床にぶつかっていました。
PRをだして床の干渉を考慮するようにしました。
今のFetch15にはこれが入っているので、joy controllerでtuck armをしても床にはぶつからないはずです。
https://github.com/fetchrobotics/fetch_ros/pull/129

[image]

cc. @708yamaguchi

k-okada, 2019-11-11:
> 来週の水曜日(11/8)は月曜授業ですが午後演習が無いのでゼミに行きたいです！

はい．演習の表は11/8はゼミになっているので予定しています．

mqcmd196, 2019-11-13:
> # SR300 RGB-D cameraの使い方
> ## librealsense2のインストール
> ```shell
> sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
> sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
> sudo apt install librealsense2-dkms librealsense2-utils librealsense2-dev librealsense2-dbg
> ```
> 
> ## Workspaceのビルド
> ```shell
> source /opt/ros/melodic/setup.bash
> mkdir ~/realsense_ws/src -p
> cd ~/realsense_ws/src
> git clone https://github.com/IntelRealSense/realsense-ros.git
> rosdep install --ignore-src --from-paths . -y -r -i
> cd ~/realsense_ws/
> catkin build
> ```
> 
> ## カメラLaunchの起動
> ```shell
> source /opt/ros/melodic/setup.bash
> source ~/realsense_ws/devel/setup.bash
> roslaunch realsense_ros rs_rgbd.launch
> ```
> 
> ## カメラ画像、点群をみてみる
> ```shell
> rviz
> # frameをcamera_linkに変更する
> # PointCloud2をrvizでみる
> ```

`roslaunch realsense_ros rs_rgbd.launch` がなかったので `roslaunch realsense2_camera rs_rgbd.launch` を実行しましたが `No RealSense devices were found!` というメッセージが返ってきました

knorth55, 2019-11-13:
@Kanazawanaoaki 
今日の`go-to-spot`に行かない問題を治すPRを作りました。
来週はこれで動くと思います。
https://github.com/jsk-ros-pkg/jsk_robot/pull/1172

MiyabiTane, 2019-11-13:
k-okada#58
fetchがサンタ帽をかぶります。

また、明日キーボードを受け取りに行く予定でしたが、予定があり、接着剤が乾く時間に伺えそうにないので金曜日か月曜日に伺います、すみません。

Kanazawanaoaki, 2019-11-14:
第五回ゼミ

・やったこと
ドンキに買い出しにいった。サンタセットは無くて帽子だけ買った。（大きな袋も靴下もアマゾンに売っている。）
帽子かぶる動作、箱をつかんで離す動作

・やること
帽子かぶる用のツノを3Dプリンタで作る。
袋と靴下買うか用意する。（要工夫）
プレゼントをつかむ方法（ちょどよい箱？タケコプター的なアタッチメントをつける？）

・役割分担（暫定）
@MiyabiTane ：変装
@YoshiaAbe ：音声認識
@mqcmd196 ：棚から袋に入れる
@Kanazawanaoaki ：袋から靴下に入れる

knorth55, 2019-11-14:
FetchがJoyコンでTuck armできなかった問題がありましたが、解決したと思います。
これでFetchはJoyコンでTuck armコマンドをしたときに、床にもぶつからず、自分にも腕をぶつけないはずです。
[image]
[image]
https://github.com/fetchrobotics/fetch_ros/pull/131

k-okada, 2019-11-15:
Amazon に売っているものは、リンクを送ってくれれば、
僕の方で買っておきます。
注意点は納期ですぐに来るものと、全然こないものがあるので
よく見ておいてください。

MiyabiTane, 2019-11-15:
fetchのスピーカーが変わらなければ帽子はこのままでいける気がする。
プレゼントを掴む方法→プレゼント箱に（例えば）アルフォートの箱を合体させてそこを掴むとかでいけそう（？）アルフォートのとこだけ色変えとくとか...

> 第五回ゼミ
> 
> ・やったこと
> ドンキに買い出しにいった。サンタセットは無くて帽子だけ買った。（大きな袋も靴下もアマゾンに売っている。）
> 帽子かぶる動作、箱をつかんで離す動作
> 
> ・やること
> 帽子かぶる用のツノを3Dプリンタで作る。
> 袋と靴下買うか用意する。（要工夫）
> プレゼントをつかむ方法（ちょどよい箱？タケコプター的なアタッチメントをつける？）
> 
> ・役割分担（暫定）
> @MiyabiTane ：変装
> @YoshiaAbe ：音声認識
> @mqcmd196 ：棚から袋に入れる
> @Kanazawanaoaki ：袋から靴下に入れる

MiyabiTane, 2019-11-15:
キーボード頂きました、ありがとうございます！
> 
> また、明日キーボードを受け取りに行く予定でしたが、接着剤が乾く時間には伺えないので金曜日か月曜日に伺います、すみません。

knorth55, 2019-11-19:
明日も岡田先生が出張でおられません。
北川も用事で16:30以降に研究室に戻る予定です。
なのでゼミの最初は長谷川（ @pazeshun )に任せようと思います。
よろしくお願いします。

[80 more replies]

closed 2019-12-26 by k-okada