jsk-ros-pkg/jsk_demos #1286: [jsk_2013_jsk_2013_04_pr2_610] demo does not work with downward planner

sktometometo, 2020-04-02:
This demo does not work with downward planner.

- Ubuntu bionic + ROS melodic
- workspace environment: [This PR](https://github.com/jsk-ros-pkg/jsk_demos/pull/1285)

### case 1 ( demo_tray.launch + downward planner )

```
~ $ ROBOT=sim roslaunch jsk_2013_04_pr2_610 demo_tray.launch 
... logging to /home/sktometometo/.ros/log/8c46daa4-74ac-11ea-8134-185e0f6d3cd8/roslaunch-Persing-23432.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Persing:45501/

SUMMARY
========

PARAMETERS
 * /footobject_laser_filter/scan_filter_chain: [{'type': 'laser_...
 * /narrow_stereo/left/laundry_pose_extractor/distanceratio_threshold: 0.49
 * /narrow_stereo/left/laundry_pose_extractor/error_threshold: 80.0
 * /narrow_stereo/left/laundry_pose_extractor/object_height: 0.15
 * /narrow_stereo/left/laundry_pose_extractor/object_width: 0.4
 * /narrow_stereo/left/laundry_pose_extractor/relative_pose: 0.164 0.27 -1.129...
 * /narrow_stereo/left/laundry_pose_extractor/reprojection_threshold: 3.0
 * /narrow_stereo/left/laundry_pose_extractor/template_filename: /home/sktometomet...
 * /narrow_stereo/left/laundry_pose_extractor/viewer_window: False
 * /narrow_stereo/left/laundry_pose_extractor/window_name: laundry_button
 * /narrow_stereo/left/tray_pose_extractor/distanceratio_threshold: 0.49
 * /narrow_stereo/left/tray_pose_extractor/error_threshold: 80.0
 * /narrow_stereo/left/tray_pose_extractor/object_height: 0.38
 * /narrow_stereo/left/tray_pose_extractor/object_width: 0.27
 * /narrow_stereo/left/tray_pose_extractor/relative_pose: 0.19 0.11 0.0 0 0 0
 * /narrow_stereo/left/tray_pose_extractor/reprojection_threshold: 3.0
 * /narrow_stereo/left/tray_pose_extractor/template_filename: /opt/ros/melodic/...
 * /narrow_stereo/left/tray_pose_extractor/viewer_window: False
 * /narrow_stereo/left/tray_pose_extractor/window_name: tray_center
 * /pddl_planner/pddl_planner: downward
 * /pddl_planner/pddl_search_option: --heuristic "hlm=...
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /shadow_laser_filter/scan_filter_chain: [{'type': 'ScanSh...

NODES
  /narrow_stereo/left/
    laundry_pose_extractor (jsk_perception/point_pose_extractor)
    laundry_sift (imagesift/imagesift)
    laundry_throttle (topic_tools/throttle)
    tray_pose_extractor (jsk_perception/point_pose_extractor)
  /
    check_chair_marker (jsk_2013_04_pr2_610/check-publish-marker.l)
    detect_chaire (jsk_2013_04_pr2_610/detect-chair.l)
    detect_table (jsk_2013_04_pr2_610/detect-table.l)
    footobject_laser_filter (laser_filters/scan_to_scan_filter_chain)
    irt_demo (roseus/roseus)
    pddl_planner (pddl_planner/pddl.py)
    shadow_laser_filter (laser_filters/scan_to_scan_filter_chain)

auto-starting new master
process[master]: started with pid [23442]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8c46daa4-74ac-11ea-8134-185e0f6d3cd8
process[rosout-1]: started with pid [23453]
started core service [/rosout]
process[shadow_laser_filter-2]: started with pid [23460]
process[footobject_laser_filter-3]: started with pid [23461]
process[detect_chaire-4]: started with pid [23467]
[... 470 lines cut ...]
The failure of case 1 seems to be because of some kind of syntax error.

```
3. Running search
This is a unit task.
reading input... [t=0s]
Simplifying transitions... done!
done reading input! [t=0.08s]
building causal graph...done! [t=0.09s]
packing state variables...Variables: 37
Bytes per state: 8
done! [t=0.09s]
done initalizing global data [t=0.09s]
Peak memory: 11600 KB

Error:
parse error: 
missing ) at: 
list
Usage error occurred.

Exit code: 2

plan= nil
```

sktometometo, 2020-04-02:
Examples of [pddl_planner](https://github.com/jsk-ros-pkg/jsk_planning/tree/master/pddl/pddl_planner) works well in my environments.

k-okada, 2020-05-05:
@sktometometo do you find why it failing. I think some  text file? return value of service? has wrong syntax. can you debug this?
``
Error:
parse error: 
missing ) at: 
list
Usage er
```

k-okada, 2021-03-03:
https://github.com/jsk-ros-pkg/jsk_planning/pull/88 should fix this issue

k-okada, 2023-12-05:
@sktometometo I encountered same problem and fixed. Please use latest jsk_planning (or deb versin of 0.1.12 for melodic)