jsk-ros-pkg/jsk_demos #1304: Failing in building drc_task_common

mqcmd196, 2020-09-25:
I run
```bash
git clone https://github.com/jsk-ros-pkg/jsk_demos.git
sudo apt install python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential
sudo rosdep init
rosdep update
rosdep install --from-paths . --ignore-src -y -r
catkin build
```
on my `workspace/src`, then `catkin build` returned error below.
```
Errors     << drc_task_common:cmake /home/obinata/research_ws/logs/drc_task_common/build.cmake.000.log
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "jsk_ik_server"
  with any of the following names:

    jsk_ik_serverConfig.cmake
    jsk_ik_server-config.cmake

  Add the installation prefix of "jsk_ik_server" to CMAKE_PREFIX_PATH or set
  "jsk_ik_server_DIR" to a directory containing one of the above files.  If
  "jsk_ik_server" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)
```

I don't use `drc_task_common` but I reported here just in case, not hurring.

knorth55, 2020-09-26:
you need to install jsk-ik-server.
for melodic, jsk_control is not released, so you need to build from source.
https://github.com/jsk-ros-pkg/jsk_control

mqcmd196, 2020-09-28:
Thanks a lot. I cloned jsk_control and succeeded in building jsk_control and drc_task_common.

closed 2020-09-28 by mqcmd196