jsk-ros-pkg/jsk_demos #1309: 2020年度ゼミ

k-okada, 2020-10-14:


knorth55, 2020-10-15:
coralにでていたPRはをマージしました．
なのでcoral本家のREADMEの通りにインストールすれば，Melodicなら動くはずです．
https://github.com/knorth55/coral_usb_ros

明日こられるかたは73B2という部屋にきたら，リビングのような部屋と大きなテレビがあるので，そこを使って作業してもらって結構です．
部屋には @knorth55 @708yamaguchi @sktometometo （@pazeshun） あたりがいるはずなので，困ったらきいてください．
とりあえず realsenseと coralをセットアップして使えればOKという理解であっていますか？

tkmtnt7000, 2020-10-16:
realsenseとcoralのデバイスをいただいていないのが僕なのでその二つをセットアップして使えればOKだと思います

knorth55, 2020-10-16:
@tkmtnt7000 了解しました。何時頃研究室にこられますか？目安でいいので教えてくれるとありがたいです。

tkmtnt7000, 2020-10-16:
@knorth555 演習終了後なので17時ごろになるかと思われます。

knorth55, 2020-10-16:
@tkmtnt7000 了解しました．

knorth55, 2020-10-16:
@k-okada 
今日は @tkmtnt7000 くんが研究室にきてくれたので，Coral TPU USBとrealsense D435iを貸し出しました．
また水曜日にやっていたゼミのとおりにCoral TPUをRealsenseの画像から動かしてみて，動くことを確認しました．
さらに研究室をみてまわったあとにYouBionicのハンドを持って帰って組み上げてもらうようにしました．

tkmtnt7000, 2020-10-16:
[image]

YouBionicハンド製作の進捗です。
M3x5,M3x10,M3x15,M3x20のネジが必要な部分があり、持って帰ったセットの中にM3のネジがなかったので進められるのはここまででした。

k-okada, 2020-10-16:
わすれていました．
ネジとレンチを自宅に送りました．月曜日の午前中に届きます．ロボットシステム中かも．．．
--
◉ Kei Okada

tkmtnt7000, 2020-10-17:
@k-okada 
ありがとうございます。

> わすれていました． ネジとレンチを自宅に送りました．月曜日の午前中に届きます．ロボットシステム中かも．．． -- ◉ Kei Okada 2020年10月16日(金) 22:17 Naoto Tsukamoto <notifications@github.com>:

tkmtnt7000, 2020-10-19:
@k-okada 
午前中にネジとレンチ届きました。少しずつ製作していきます。

k-okada, 2020-10-21:
@yuli-disney　入館，警備， エンターティメント，カードを選ぶ，会計でコートをとってくる，ホテル部屋案内
@softyanija　棚の物を探してくれる
@tenrobo18　除菌，机を拭く，ドアを開ける．　ロボテック
@tkmtnt7000　配達
@ketaro-m　ドアにビニール袋を書ける

tkmtnt7000, 2020-10-21:
@708yamaguchi 
Fetchで`(send *ri* :state :worldcoords)`できないのはこれだとおもいます。
https://github.com/jsk-ros-pkg/jsk_robot/pull/1167

knorth55, 2020-10-21:
@tkmtnt7000 @k-okada 

fetcheus関連ですが，今後のゼミにおいて以下が問題になる可能性があります．
マージしていただけると助かります．

### Fetchの`:go-pose-unsafe`がおかしい
- [fetcheus] fix radian in :go-pos-unsafe https://github.com/jsk-ros-pkg/jsk_robot/pull/1166
- [fetcheus] fix :go-pos-unsafe to add keyword arguments https://github.com/jsk-ros-pkg/jsk_robot/pull/1209

### Fetchの`:move-to`が微修正で余計なことをする
- [fetcheus] override :move-to-wait to set :correction nil https://github.com/jsk-ros-pkg/jsk_robot/pull/1169

### Fetchの腕が床にあたる
- [fetcheus] Add fetch collision default https://github.com/jsk-ros-pkg/jsk_robot/pull/1174

k-okada, 2020-10-21:
皆さん，途中で抜けてすいません＆手伝ってくれた人ありがとう．
マージしました．次回ロボットを動かすまでに，
```
$cd semi_ws/src/jsk_robot
$git pull origin master
```
しておきましょう．

knorth55, 2020-10-21:
マージありがとうございます．
Fetchのユーザ側PCは`jsk_robot`と`jsk_pr2eus`をソースで入れれば問題ないと思います．

tkmtnt7000, 2020-10-25:
[image]
ロボットハンドの進捗です。あとはarduinoを中に入れて蓋をすれば完成です。
arduino周りがよくわからないので次回持っていきます。

k-okada, 2020-10-27:
https://github.com/jsk-ros-pkg/jsk_demos/blob/jsk_2020_10_semi/jsk_2020_10_semi/euslisp/sample.l を追加してみました．
uml を追加した人は，このサンプルをみて，イメージアニメーションを作る事がd家いるはずです．

アームを動かすのは実機とほぼ同じ動きになります．
移動については，建物をすり抜けたりしますが，実機だとちゃんと障害物を避けてくれます．

k-okada, 2020-10-28:
`*ri*` で使えるメンバ関数は https://euslisp-docs.readthedocs.io/en/latest/pr2eus/robot-interface/ にあります．動いている途中でキャンセルにするには
```
(send *ri* :go-pos-no-wait ...)
...
(send *ri* :go-stop)
```
みたいにします．

spotのeusモデルは
https://gist.github.com/k-okada/5bcaecf7b41c534048ac4f82afdf7a81
にあります．(c.f. https://github.com/jsk-ros-pkg/jsk_robot/pull/1263 )

k-okada, 2020-10-29:
皆さんの作っていくれたumlをマージして
https://github.com/jsk-ros-pkg/jsk_demos/pull/1311
に入れました．

また，少しだけコメントをしました．
https://github.com/jsk-ros-pkg/jsk_demos/pull/1311/files#r514116563 音声認識との連携について
https://github.com/jsk-ros-pkg/jsk_demos/pull/1312/files#r514121372 移動コマンドについて
https://github.com/jsk-ros-pkg/jsk_demos/pull/1314/files#r514120599 腕の動かし方について
https://github.com/jsk-ros-pkg/jsk_demos/pull/1316/files#r514125184 顔認識について

また，座標系を動かしながら確認するプログラムを作ってみました．
sample.l をロードした後にこれを実行してみてください．
EmacsのM-x shell 上だとjklm で座標が動きます．ただ毎回リターンキーを押す必要があります．
直接ターミナルから動かすとリターン無しで動いてくれるハズです．
https://gist.github.com/k-okada/78fec94346db94bc08cbbd185d6c20da

= 以下はなにかおかしくなった人向け
もともと僕が作っていた`jsk_2020_10_semi`ブランチがおかしかったのもあって少し強制的に直しました．github上のorign/jsk_2020_10_semi と合わせたかったら以下のようにしてください．特に問題が出ていない人は何もしなくて良いです．
```
git fetch --all
git checkout jsk_2020_10_semi
git branch #  jsk_2020_10_semiにいることを確認
git reset --hard origin/jsk_2020_10_semi
```

tkmtnt7000, 2020-10-30:
73B2近くの座標と関節名リスト
https://gist.github.com/tkmtnt7000/4b816296d9e77e3b2a51730507d76f19

tkmtnt7000, 2020-11-04:
今日学んだことリスト
・git add <追加したいもの>
・git commit -m "コメント" ファイル
・git push アカウント名 ブランチ

・rpy(0 0 pi/2)みたいにロールピッチヨーが表せる
・つかむ動作はwait-interpolationでは待てない(unix:sleep 1)を使う

softyanija, 2020-11-04:
今日の学んだことのメモ
'''
(send *fetch* :rarm :inverse-kinematics (send (send *fetch* :copy-worldcoords) :transform (make-coords :pos #f(600 250 680) **:rpy (float-vector 0 0 pi/2)**)) **:rotation-axis t** :debug-view t)
'''
:rpyで向きを指定し:rotation-axis tでそれを反映させる

'''
(send *fetch* :rarm :wrist-r :joint-angle -90)
'''
手先を指定の角度まで回転させる
'''
(send *ri* :wait-interpolation)
'''
前の動作が終わるまで次の動作を待つ。graspには効かないので注意
'''
(send *ri* :start-grasp :effort 60)
(unix:sleep 1)
'''
graspに同じようなことをする手段

'''
(send *ri* :angle-vector (send *fetch* :angle-vector) 4000 **:use-torso nil**)
'''
腰の上下無しで手を動かす

tenrobo18, 2020-11-04:
git push しようとしたら以下のようなエラーが出たのですがどうすればよいでしょうか

mech-user@ki00102:~/semi_ws/src/jsk_demos/jsk_2020_10_semi$ git add -A
mech-user@ki00102:~/semi_ws/src/jsk_demos/jsk_2020_10_semi$ git commit -m "ドア ノブをつかもうとする"
[jsk_2020_10_semi 2c0ddb64] ドアノブをつかもうとする
 3 files changed, 215 insertions(+)
 create mode 100644 jsk_2020_10_semi/euslisp/.sample.l.swp
 create mode 100644 jsk_2020_10_semi/euslisp/temma.l
 create mode 100644 jsk_2020_10_semi/uml/temmasuzuki.uml
mech-user@ki00102:~/semi_ws/src/jsk_demos/jsk_2020_10_semi$ git branch
* jsk_2020_10_semi
  master
mech-user@ki00102:~/semi_ws/src/jsk_demos/jsk_2020_10_semi$ git push origin jsk_2020_10_semi
Username for 'https://github.com': tenrobo18
Password for 'https://tenrobo18@github.com': 
remote: Permission to jsk-ros-pkg/jsk_demos.git denied to tenrobo18.
fatal: unable to access 'https://github.com/jsk-ros-pkg/jsk_demos/': The requested URL returned error: 403

yuli-disney, 2020-11-04:
上に同じくです。

ketaro-m, 2020-11-04:
多分`git push origin jsk_2020_10_semi`じゃなくて，`git push (フォーク先のリモート名) jsk_2020_10_semi`にしないといけないと思う．
多分，`git push tenrobo18 jsk_2020_10_semi`とか`git push yuli-disney jsk_2020_10_semi`とかだと思うけど，`git remote`で確認して`origin`じゃない方を書けば良かったはず

tenrobo18, 2020-11-04:
mech-user@ki00102:~/semi_ws/src/jsk_demos/jsk_2020_10_semi$ git remote
origin

リモート名がなかったですね
この場合どうすればいいんでしたっけ

tenrobo18, 2020-11-04:
多分フォークしてなさそう

ketaro-m, 2020-11-04:
じゃあフォークした後に，`git remote add (好きな名前) https://github.com/tenrobo18/jsk_demos.git`とかかな
それで`git push (その名前) jsk_2020_10_semi`ってすればいけるはず

tenrobo18, 2020-11-04:
できました！
ありがとうございます

ketaro-m, 2020-11-11:
今日学んだことリスト
・`(send *fetch* :torso :waist-z :joint-angle 100.0)` : fetchの腰を100上げる(reset-poseできないときなどに有効)

tenrobo18, 2020-11-11:
今日学んだこと
ハンドの手先位置を調整
```
(setq *difpos* (list 0 0 0))
(defun move-hand (x y z)
 ;; (let (x)
    (send *fetch* :start-grasp)
    (send *ri* :start-grasp)
   ;; (setq x -100)
    (unix:sleep 1)
 ;; (send *fetch* :rarm :end-coords :parent)
    (send *fetch* :rarm :move-end-pos (float-vector x y z) :local :debug-view t)
    (send *ri* :angle-vector (send *fetch* :angle-vector))
    (send *ri* :wait-interpolation)
  ;;(print (send (send *fetch* :copy-worldcoords) :transform (send *fetch* :rarm-end-coords) ))
    (setf (elt *difpos* 0) (+ (elt *difpos* 0) x))
    (setf (elt *difpos* 1) (+ (elt *difpos* 1) y))
    (setf (elt *difpos* 2) (+ (elt *difpos* 2) z))
   ;; ()
   ;; )
  )
```

softyanija, 2020-11-11:
rvizを立ち上げる方法

```
rossetmaster fetch15
rossetip
roslaunch jsk_fetch_startup rviz.launch
```

ketaro-m, 2020-11-11:
edgetpuで顔認識したトピックをサブスクライブする例
```
#!/usr/bin/env roseus

(ros::roseus-add-msgs "jsk_recognition_msgs")
(ros::roseus-add-msgs "geometry_msgs")

(defclass track-box-to-cmd-vel
    :slots (track cmd_vel))
(defmethod track-box-to-cmd-vel
    (:init ()
        ;; ros::init
        (ros::roseus "face_recognizedColord_client")
        (ros::subscribe "/edgetpu_face_detector/output/rects" jsk_recognition_msgs::RectArray #'send self :cb)
        (ros::advertise "/cmd_vel" geometry_msgs::Twist 5)
        (setq cmd_vel (instance geometry_msgs::Twist :init))
        (setq track nil)
        (ros::ros-info "init")
    self)
    (:cb (msg)
        (setq track (send msg :rects))
    )
    (:loop ()
        (ros::rate 10)
        (setq alpha 0.001)
        (ros::rate 10)
        (while (ros::ok)
            ; (ros::ros-info "hello")
            (setq alpha 0.001)
            (if track
                (progn
                    (if (= (send (car track) :x) 0)
                            (send cmd_vel :angular :z 0.3)
                        (send cmd_vel :angular :z (* (- alpha) (- (send (car track) :x) 315)))
                    )
                    (ros::ros-info "x [~A]" (send (car track) :x))
                    (ros::publish "/cmd_vel" cmd_vel)))
            ; (ros::ros-info "hello2")
            (ros::spin-once)
            (ros::sleep)
        )
        (exit)
    )
)

(setq *ri* (instance track-box-to-cmd-vel :init))
(send *ri* :loop)
```

tenrobo18, 2020-11-11:
PCからundockする方法
"""
2.irteusgl$ load "~/semi_ws/src/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/euslisp/navigation.l"
9.irteusgl$ undock
"""

softyanija, 2020-11-11:
```
$ roseus samp.l
```
で
```
[ INFO] [1603884575.353959970]: waitForService: Service [/apply_planning_scene] has not been advertised, waiting...
```
のエラーが出る人は
https://github.com/jsk-ros-pkg/jsk_robot/pull/1265
をみて
```
$  git remote add k-okada http://github.com/k-okada/jsk_robot
$  git pull k-okada 1174
$  git checkout 1174
```
してください。

softyanija, 2020-11-11:
棚を障害物として設定
```
(defun add-shelf-collision-object ()
  (let ((shelf (make-cube 10 1000 1100)))
    (send shelf :translate #f(795 0 550))
    (send *co* :add-object shelf :frame-id "base_link" :object-id "shelf")))

(defun delete-shelf-collision-object ()
  (send *co* :delete-object-by-id "shelf"))
```

ketaro-m, 2020-11-11:
fetch kinematics simulator 上ではfetch自身の体も含めて障害物が定義されていないので，自分の体をえぐるように腕を動かしてしまうことがある．これは，実機では`fetch-init`をする際に，
```
(setq *co* (instance collision-object-publisher :init))
```
というオブジェクトが作られてそれを避けるように経路を生成するからであるが，ローカルではこれが定義されていないからである．
理由は，[/apply_planning_scene] というサービスがないと` *co*` が作られない仕様になっているからであり，実機だとこのサービスが既に存在しているが，ローカルでは存在しないからである．

ローカル上でも自分で`*co*`を利用して障害物を作りたい場合は，事前に別のターミナルで
```
$ roslaunch fetch_moveit_config demo.launch
```
としておくと上述のサービスが存在するようになるので，その状態で`fetch-init`するとちゃんと`*co*`が定義され，その中の`:add-keepout-collisioin-object`や`:delete-keepout-collision-object`などを使って障害物を生成・削除してシミュレータ上で試すことができる．

k-okada, 2020-11-11:
screen point (https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_pcl_ros/nodes/pointcloud_screenpoint.html ) の使い方．

https://gist.github.com/k-okada/0ec60202cfdff466aa6b0452a90850c1 からファイルをダウンロード，`/home/mech-user/` においてあることを想定している．

1. camera.launch を立ち上げる．カメラの画像が見えないときは，リブートすると良い
2. proc.launch を立ち上げる．image_view2をクリックすると，rvizで赤い球が表示される．これが，画面上の点に対応した三次元点．
   launch ファイルは２つを同じファイルにまとめても良い．
3. プログラムの書き方は，screen-point-example.l を参照．サービスコール(/pointcloud_screenpoint_nodelet/screen_to_point) を呼ぶと三次元点が返ってくる．あるいは，トピック(/camera/color/image_raw/screenpoint) にデータを送ると，赤い球（PointCloud2）が表示される．
　二次元画像上の座標データを連続的に送る場合は，サービスコールではなく，トピックを送って，(/pointcloud_screenpoint_nodelet/outp_point
) 等を取得するのでももよい．

ぜひ，これにedgeTPUの画像処理結果で得られた二次元の座標を３次元情報に変換してみてください．

Tips: edgeTPUは別のワークスペースになるので，launchファイルは別のものが必要になります．

tkmtnt7000, 2020-11-11:
edgetpuで認識した顔の情報をサブスクライブする例を
defclassやdefmethodを使わないパターンで書いてみました.
シンプルにx,y,width,heightを表示します.

プログラム
```
#!/usr/bin/env roseus

(ros::roseus-add-msgs "jsk_recognition_msgs")

(setq *rect* nil)

(defun cb (msg)
  (if msg
      (setq *rect* (send msg :rects))
    )
)

;;main                                                                          
(ros::roseus "face_rect_client")
(ros::subscribe "/edgetpu_face_detector/output/rects" jsk_recognition_msgs::RectArray #'cb)

(ros::rate 10)
(while (ros::ok)
  (if *rect*
      (progn
        ;;(ros::ros-info "hello")                                               
        (ros::ros-info "x: ~A" (send (car *rect*) :x))
        (ros::ros-info "y: ~A" (send (car *rect*) :y))
        (ros::ros-info "width: ~A" (send (car *rect*) :width))
        (ros::ros-info "height: ~A" (send (car *rect*) :height))

	(setq w (send (car *rect*) :width))
	(setq h (send (car *rect*) :height))
	(ros::ros-info "area : ~A" (* w h))
	)
     )
  (ros::spin-once)
  (ros::sleep)
  )

```
```(send (car *rect*) :width)```
のようにcarでメソッドにアクセスできるようになる理由は以下の出力を見ると理解できたかも?
42の出力は中身が1つのリスト
```
42.E1-irteusgl$ *rect*
(#<jsk_recognition_msgs::rect #X55c628ec7028>)
43.E1-irteusgl$ (car *rect*)
#<jsk_recognition_msgs::rect #X55c628ec7028>
```

k-okada, 2020-11-12:
動いているのでこだわらなくても良いですが．．．

```
(setq *difpos* #f(0 0 0))
```
としておくと，
```
(setq *difpos* (v+ *difpos* (float-vector x y z)))
```
とかけます．

```
$ (load "~/semi_ws/src/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/euslisp/navigation-utils.l")
```
```
$ (load "package://jsk_fetch_startup/euslisp/navigation-utils..l")
```
`~/semi_ws/`の人でも`~/catkin_ws`の人でも`source <それぞれのワークスペース>/devel/setup.bash` してあればその環境に合わせてファイルをロードしてくれます
```
roseus '(progn (load "package://jsk_fetch_startup/euslisp/navigation-utils..l")(undock))"
```
とすると，shellからoneliner でundock できます．
```
	(setq w (send (elt *rect* 0) :width))
```
と書くほうが，なんとなく安心かな．あまり理由は何けど，car があるとcdrも欲しくて，再帰で計算するかんじ
```
(if (> (length *rect*) 0)
   (setw w (send (elt *rect* 0) :width)))
```
みたいな感じが見慣れている．ただ，例えばrectのサイズの一番大きいものを知りたい，という場合は
```
(ros::roseus-add-msgs "jsk_recognition_msgs")
(setq a (list (instance jsk_recognition_msgs::Rect :init :width 100 :height 100) (instance jsk_recognition_msgs::Rect :init :width 100 :height 200) (instance jsk_recognition_msgs::Rect :init :width 100 :height 50))) ;; ダミーのデータを作る
(setq b (car (sort a #'(lambda (x y) (< (* (send x :width) (send x :height)) (* (send y :width) (send y :height)))))))
 ;; これが理解できると，だいぶマスターにしてきている．

```
みたいな感じでしょうか．

mqcmd196, 2020-11-24:
uberバッグが届きました[image]

ketaro-m, 2020-11-24:
ありがとうございます

softyanija, 2020-11-25:
メモ
・rossetipとrossetmasterした後rqtを立ち上げ、pluginからimage viewを選択、その後覗きたいノードを選択するとfetch視点の映像を見ることができる

tkmtnt7000, 2020-11-25:
今更ですが相対座標の取得とそこに手を持っていく動きを関数にしました。
```
;; 相対座標の取得                                                                              
(defun make-relative-coord (x y z roll pitch yaw)
  (setq decalt (float-vector x y z))
  (setq euler (float-vector roll pitch yaw))
  (setq relative-coord (send (send *fetch* :copy-worldcoords) :transform
                             (make-coords :pos decalt :rpy euler))
       )
  )
;; 相対位置に動かしたいとき                                                                    
(defun relative-move-hand (x y z roll pitch yaw)
  (let coords
       (setq coords (make-relative-coord x y z roll pitch yaw))
       (send *fetch* :rarm :inverse-kinematics coords :debug-view t)
       (send *ri* :angle-vector (send *fetch* :angle-vector)))
  )

```

k-okada, 2020-11-25:
```
(send *fetch* :rarm :move-end-pos #f(0 0 10))
(send *fetch* :rarm :move-end-rot #f(0 0 0.1))
```
でも動くかと思います．

softyanija, 2020-11-25:
本日の進捗
・引き出しまで手を伸ばし、取っ手を掴んで引き、戻して、取っ手から手を話して初期状態に戻る一連の動作に成功
・ただしこのままでは肘が邪魔で引き出しの中身を見ることができない。手の位置を下げてその分torsoを上げると良さそう

tenrobo18, 2020-11-25:
今日やったこと
- LEGOのハンドを作ってfetchにドアノブを拭かせた
来週までにやること
- ハンドのケースを作る
- ハンドを取り上げるプログラムを書く
- 消毒液に浸す方法を考える

tkmtnt7000, 2020-11-25:
今日やったこと
・ものを受け取る動作と受け渡す動作の確認
・音声認識の確認（ウーバーイーツ）
今後やること
・人が見えたら受け渡すようにする
・言葉をきっかけに動作させてみる
・ウーバーイーツとのつなげ方

softyanija, 2020-12-02:
今日(12/2)やったこと
・keepoutがバグを引き起こすのでしばらく無しでやらなくちゃならない、辛い
・故にtorso上げてからinverse-kinematicsを打つと良い

708yamaguchi, 2020-12-02:
@softyanija 
> ・keepoutがバグを引き起こすのでしばらく無しでやらなくちゃならない、辛い

https://github.com/jsk-ros-pkg/jsk_robot/pull/1268 と https://github.com/jsk-ros-pkg/jsk_robot/pull/1269 で治るかもしれません。
試してみます。

mqcmd196, 2020-12-02:
fetchのCPU使用率を下げるために，立ち上げるノードを制限するlaunchファイルを書いてみました． https://github.com/knorth55/jsk_robot/pull/56

- 使い方
1. fetchの体内でターミナルを立ち上げる
1. `sudo initctl stop jsk-fetch-startup` と実行
1. `roslaunch jsk_fetch_startup fetch_bringup.launch battery_warning:=false nav_speak:=false fetch_tweet:=false fetch_sensors:=false use_safe_teleop:=false` と実行
これによって，ノードの起動を制限することができます．

- 注意点
1. `use_safe_teleop` を無効にしていますが，これによって，コントローラでの移動操作がすべてunsafeになります．つまり**R2を押さなくても，前方の障害物の有無にかかわらず，fetchが動けるようになるので，移動の際には十分に注意してください**．eusの `move-to` などは特に問題ないです．
1. `sudo` コマンドを打つとパスワードが求められますが，ここに書いておくわけにはいかないので，実行する際には73B2の誰かにパスワードを聞いてください．
1. 実験し終わったら，Ctrl+c でlaunchを終了して，`sudo initctl start jsk-fetch-startup` と実行して，もとの状態に戻しておいてください．

tkmtnt7000, 2020-12-02:
今日やったこと
・fetchにUberEatsのリュックの蓋を開けさせる
実行動画：[https://youtu.be/23TQ7wtrp7g](https://youtu.be/23TQ7wtrp7g)

今後やること
・センサやスイッチを使って取っ手を掴んだことを認識させる
・取っ手を考える、探す

tkmtnt7000, 2020-12-02:
シミュレーション上ではそもそもtorsoを使うような動きはしませんが、実機を動かしたときと同様にして
```
(send *ri* :angle-vector (send *fetch* :angle-vector) 5000 :use-torso nil)
```
のようにした場合、:use-torsoの部分がシミュレーション上ではエラーとなるようです。
実機では問題なく動きました。

エラー文
[https://gist.github.com/tkmtnt7000/2f9684c728e129c0efc7165db38ae23e](https://gist.github.com/tkmtnt7000/2f9684c728e129c0efc7165db38ae23e)

knorth55, 2020-12-02:
@tkmtnt7000 
`fetcheus`の`origin/master`では解決しています．
`jsk_robot`を最新版にして試してみてください．

https://github.com/jsk-ros-pkg/jsk_robot/pull/1274

tkmtnt7000, 2020-12-02:
> @tkmtnt7000
> `fetcheus`の`origin/master`では解決しています．
> `jsk_robot`を最新版にして試してみてください．
> 
> [jsk-ros-pkg/jsk_robot#1274](https://github.com/jsk-ros-pkg/jsk_robot/pull/1274)

了解しました. 最新版にして試してみます.

tkmtnt7000, 2020-12-03:
use-torso できました. ありがとうございます.
> > `fetcheus`の`origin/master`では解決しています．
> > `jsk_robot`を最新版にして試してみてください．
> > [jsk-ros-pkg/jsk_robot#1274](https://github.com/jsk-ros-pkg/jsk_robot/pull/1274)
> 
> 了解しました. 最新版にして試してみます.

tkmtnt7000, 2020-12-08:
実機（Kinematics Simulator ）の関節角度を確認する方法
```
(send *ri* :state :potentio-vector)
```

tkmtnt7000, 2020-12-09:
[ss_ds_j_1_13.pdf](https://github.com/jsk-ros-pkg/jsk_demos/files/5663823/ss_ds_j_1_13.pdf)

オムロンのスイッチのデータシート

softyanija, 2020-12-09:
torsoの目安がわかったから後はinverseの方が色々と良い気がする
いい感じの角度引き出しに手をかけたとき
#f(277.216 68.033 76.4115 -124.976 77.5795 5.3212 99.3506 71.1364 -0.390871 27.4847)

ketaro-m, 2020-12-09:
Arduinoのピンインプットをros_msgとしてpublishする例．

- Arduinoにスケッチを書き込む(以下は，デジタルピン7の値をpublishする)
```
#include <math.h>
#include <ros.h>
ros::NodeHandle  nh;

#include <sensor_msgs/Range.h>
#include <std_msgs/Int16.h>

#define DIN_PIN 7
std_msgs::Int16 pin_msg; 
ros::Publisher arduino_pub("arduino", &pin_msg);

void pin_setup()
{
  pin_msg.data = 0;
  nh.advertise(arduino_pub);
}

void pin_loop()
{
  pinMode( DIN_PIN, INPUT );
  pin_msg.data = digitalRead(DIN_PIN);
  arduino_pub.publish(&pin_msg);
}

void setup()
{
  nh.initNode();
  pin_setup();
}

unsigned long timer = 0;
void loop()
{
  unsigned long now = millis();
  // measure and publich pin value every 1000 milliseconds
  if ( (now - timer) > 1000 ) {
    pin_loop();
    timer = now;
  }
  nh.spinOnce();
}
```

- ros_node を立ち上げてros_topicを読める．

```
$ roscore
$ rosrun rosserial_python serial_node.py /dev/ttyUSB0
$ rostopic echo /arduino
```

[131 more replies]