jsk-ros-pkg/jsk_demos #1362: [jsk_maps] start_map_eng2.launchでmachineタグが使われている理由が知りたい

tkmtnt7000, 2022-07-06:
個人的にpublish_spot.lの起動でmachineタグを使っていることが気になりました．
https://github.com/jsk-ros-pkg/jsk_demos/blob/244be9df1bc7b3c711900ce6adb546208e31b207/jsk_maps/launch/start_map_eng2.launch#L27-L33

気になったので調べてみると以下のようにPR2ではc2でこのlaunchが起動されているようで，
https://github.com/jsk-ros-pkg/jsk_robot/blob/3a940abddb31b362c983710ff3ef1e3a78e88e23/jsk_pr2_robot/jsk_pr2_startup/pr2.launch#L62-L66
この辺りは何か歴史的経緯があるのかなと思ったのですが(通信orCPU使用率？と推測しています)，ちょっと自分では推測の域を出なかったのと記録として残ったほうが良いかと思ったのでissueとしておきました．

knorth55, 2022-07-06:
基本的にはPR2はC2でノードを起動します
制御PCがC1なのでできるだけ、そちらの負荷を減らすことをちょいちょいしています

tkmtnt7000, 2022-07-07:
ありがとうございます。
C1の負荷とパケットドロップとが関係あるとすると、常時起動させるノードでC2が選択可能になっているのは重要そうに思いました。