jsk-ros-pkg/jsk_demos #1408: (send *ri* :angle-vector (send *pr2* :angle-vector) time) の補完のバグ？

Michi-Tsubaki, 2024-12-19:
https://github.com/Michi-Tsubaki/jsk_demos/blob/f2dabd87ee818c483fb733d4b4b92d7203e2f027/jsk_2024_10_semi/pr2_surgery/src/pr2_surgery.l#L384-L390

の384行目に
`(send *pr2* :rarm :move-end-pos (float-vector 0 0 (- remain_tmp)) :local)`
でremain_tmp (だいたい400= 40cm) だけ*pr2*に右腕を引かせています．

ここで，385行目のように`(send *ri* :angle-vector (print (send *pr2* :angle-vector)) 10000) `しても10秒書けて動いてくれず，50000とかにしても**速いままです．**

```
(send *ri* :angle-vector-sequence (list (send *pr2* :angle-vector)) (list 10000)
		  :default-controller 0.001
		  :min-time 0.0001
		  :minjerk-interpolation t
		  )
```

にようにして，angle-vector-sequenceに要素1つのlistとして投げ、minjerk補完をtrueにすると所望のゆっくりとした動作になります．

こちらは，補完のバグでしょうか？　ご存知の方がいらっしゃればご教示いただけますとありがたいです．

pazeshun, 2024-12-21:
次回研究室に来た時に、まずは良い動画を撮ることが優先ですが、その後時間があれば、以下の情報をprintしてこのissueに載せておいてもらえると、他の人が簡単に状況を再現できるようになるので、デバッグの足がかりになりそうです。
- 問題のangle-vectorを送る直前の実機のpotentio-vector
- 問題のangle-vector

Michi-Tsubaki, 2024-12-21:
↑了解しました．

Michi-Tsubaki, 2024-12-23:
問題のangle-vectorを送る直前の実機のpotentio-vectorは
`#f(245.558 2.86767 4.836 94.7781 -37.7396 89.1852 -83.2526 234.969 -21.0146 48.4254 -65.465 -80.4161 136.303 -44.8264 167.785 0.207389 65.664)`
問題のangle-vectorは
`#f(245.743 2.86826 4.8349 94.7785 -37.7392 89.1891 -83.2545 234.975 -51.6426 38.334 -74.3802 -70.293 136.59 -13.425 182.547 0.219389 61.578)`
でした．