jsk-ros-pkg/jsk_demos #1411: ikでuse torso nilにしてもtorsoが動く

Michi-Tsubaki, 2025-02-26:
↓のソースコードのコメントアウトしている方のinverse-kinematicsで `:use-torso nil` にしても torsoが動いてしまいます．link-listの指定の仕方が間違っていたりしますか？

link-listを陽に指定した経緯ですが，針先や拡張した指先の座標をtargetにしてikを解いています．

とりあえずtorsoのangele-vectorを保存して復元する方法でうまく行っているので取り急ぎ問題にはなりませんが，気になります．

https://github.com/Michi-Tsubaki/pr2_tubes/blob/ea6c07d5f1978c25a85190687af6632e92d3608b/src/pr2_tubes.l#L378-L407

```
;; Rarm goes to wait position
(setq *larm-angles* (send *pr2* :larm :angle-vector))
(setq *torso-angle* (send *pr2* :torso :angle-vector))
(setq *pre-grasp-coords* (send *needle-tip* :copy-worldcoords))
(send *pre-grasp-coords* :translate #f(-50 10 0))
(send *pr2* :rarm :inverse-kinematics *pre-grasp-coords* 
      :move-target *right-finger-tip*
      :link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent))
      :rotation-axis t)
(send *pr2* :larm :angle-vector *larm-angles*) ;; Recover torso angle
(send *pr2* :torso :angle-vector *torso-angle*)
(send *ri* :angle-vector (send *pr2* :angle-vector) 6000)
(send *ri* :wait-interpolation)
(send *irtviewer* :draw-objects)
(ros::duration-sleep 1)

#| Issue これだとuse-torsoをnilにしても腰が動いてしまうなぜか
(send (send *pr2* :larm :end-coords) :dissoc *needle*)
(setq *pre-grasp-coords* (send *needle-tip* :copy-worldcoords))
(send *pre-grasp-coords* :translate #f(0 -50 0))
(send *pr2* :rarm :inverse-kinematics *pre-grasp-coords* 
      :move-target *right-finger-tip*
      :link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent))
      :use-torso nil
      :rotation-axis t)
(send *ri* :angle-vector (send *pr2* :angle-vector) 5000)
(send *ri* :wait-interpolation)
(send *irtviewer* :draw-objects)
|#
```

Kanazawanaoaki, 2025-02-26:
先程対面で一緒に確認しましたが，
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/2b5af0cd5377c275f7c39975b9d46b574e8ee1e4/pr2eus/pr2-utils.l#L102-L118
を見ると，pr2の:inverse-kinematicsの`:use-torso`は`:link-list`が設定されていない時にtorsoを使うかどうかの引数で，link-listの方が強い指定になっているからだと思います．

問題を再現するサンプルを書いてみると
https://gist.github.com/Kanazawanaoaki/76906c6a062c4585ac822c4e4f75c090#file-use-torso-ik-test-l
```bash
(load "package://pr2eus/pr2-interface.l")
(load "models/arrow-object.l")

(pr2-init)
(setq *arrow* (arrow))

(setq *cube* (make-cube 200 10 10))

(objects (list *pr2* *arrow* *cube*))

(send *arrow* :newcoords (make-coords :pos #f(700 0 1200)))
(send *cube* :newcoords (send (send *pr2* :rarm :end-coords) :copy-worldcoords))
(send *cube* :translate #f(100 0 0) :local)
(send (send *pr2* :rarm :end-coords) :assoc *cube*)

(send *pr2* :reset-pose)
(format t "before ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))

;;;; Comment in only the executing version.

;; (send *pr2* :rarm :inverse-kinematics (send *arrow* :copy-worldcoords)
;;       :move-target *cube*
;;       :link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent))
;;       :use-torso nil
;;       :rotation-axis t) ;; 264.824

(send *pr2* :rarm :inverse-kinematics (send *arrow* :copy-worldcoords)
      :move-target *cube*
      :link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent) (send *pr2* :rarm :root-link))
      :use-torso nil
      :rotation-axis t) ;; 50.0

(format t "after ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
```
こんな感じで，`:link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent) (send *pr2* :rarm :root-link))`のように指定するとlink-listにtorsoが含まれず，torsoを動かさないようにできそうです．

Michi-Tsubaki, 2025-02-27:
:link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent) (send *pr2* :rarm :root-link)) で上手く行きました！
ありがとうございます．

closed 2025-02-27 by Michi-Tsubaki

k-okada, 2025-02-27:
@Kanazawanaoaki の説明で尽くされているかとおもいます． 💯 

敢えて追加すると，そもそもは片手のIKだけを考えていて，かつ，`:move-target`も`:end-coords`が一般的だろう，ということで，`(send *pr2* :rarm :inverse-kinematics <coords>) ` をデフォルトの使い方として，`(send *pr2* :rarm :inverse-kinematics <coords> :use-torso <t/nil>) ` で腰の利用のon/offを切り替えるようにしていました．

その中で，`:move-target`で制御対象を切り替えたい，としたときに一つだけ何処にも書いていない制約があって（多分これでエラーになって，今のプログラムになっている？）:link-list をセットしない時は，`:move-target`の:parentがlinkでないとダメ，というのがあります．こうすると，片手IKで，`:move-target`を指定しつつ，`:use-torso`で腰の利用のon/off を指定することが出来ます．

また，`(send *pr2* :inverse-kinematics ...)` と呼ぶ事が出来て，これがオールマイティで，人形だと全身IKになります．で，この時は，全身の中でどのリンクを使うか指定するために`:link-list` がマスト（片手IKのときは，片手全部動かすのが想定される）で，もはや`:use-torso` を使う必要はない，という考え方になっていたんだとおもいます．

```
(load "package://pr2eus/pr2-interface.l")
(load "models/arrow-object.l")

(pr2-init)
(setq *arrow* (arrow))

(setq *cube* (make-cube 200 10 10))

(objects (list *pr2* *arrow* *cube*))

(send *arrow* :newcoords (make-coords :pos #f(800 -400 1200)))
(send *cube* :newcoords (send (send *pr2* :rarm :end-coords) :copy-worldcoords))
(send *cube* :translate #f(100 0 0) :local)
(setq *cube-tip-pos* (make-cascoords :coords (send *cube* :copy-worldcoords)))
(send *cube-tip-pos* :translate #f(100 0 0) :local)
(send *cube-tip-pos* :assoc *cube*)
;; when :move-arm is set and :link-list is nil, '(send move-target :parent)' must be link
;; https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/2b5af0cd5377c275f7c39975b9d46b574e8ee1e4/pr2eus/pr2-utils.l#L117
;; (send (send *pr2* :rarm :end-coords) :assoc *cube*)
(send (send *pr2* :rarm :end-coords :parent) :assoc *cube-tip-pos*)

;; default usage 1, set only target-coords, use :use-torso :t for option
(send *pr2* :reset-pose)
;;(format t "before ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
(format t "before ik torso joint is ~A~%" (send *pr2* :torso :waist-z :joint-angle))
(send *pr2* :rarm :angle-vector #f(0 0 0 -90 -180 -90 180.0)) ;; use nice initial pose
(send *pr2* :rarm :inverse-kinematics (send *arrow* :copy-worldcoords)
      :use-torso t
      :rotation-axis nil
      :debug-view :no-message) ;; 50.0
(send *irtviewer* :draw-objects)
(format t "after ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))

(send *pr2* :reset-pose)
(format t "before ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
(send *pr2* :rarm :angle-vector #f(0 0 0 -90 -180 -90 180.0)) ;; use nice initial pose
(send *pr2* :rarm :inverse-kinematics (send *arrow* :copy-worldcoords)
      :use-torso nil
      :rotation-axis t
      :debug-view :no-message) ;; 50.0
(send *irtviewer* :draw-objects)
(format t "after ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))

;; default usage 2, use move-target. we expect '(send move-target :parent)' is a link
(send *pr2* :reset-pose)
;;(format t "before ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
(format t "before ik torso joint is ~A~%" (send *pr2* :torso :waist-z :joint-angle))
(send *pr2* :rarm :angle-vector #f(0 0 0 -90 -180 -90 180.0)) ;; use nice initial pose
(send *pr2* :rarm :inverse-kinematics (send *arrow* :copy-worldcoords)
      :move-target *cube-tip-pos*
      :use-torso t
      :rotation-axis nil
      :debug-view :no-message) ;; 50.0
(send *irtviewer* :draw-objects)
(format t "after ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))

(send *pr2* :reset-pose)
(format t "before ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
(send *pr2* :rarm :angle-vector #f(0 0 0 -90 -180 -90 180.0)) ;; use nice initial pose
(send *pr2* :rarm :inverse-kinematics (send *arrow* :copy-worldcoords)
      :move-target *cube-tip-pos*
[... 31 lines cut ...]
;; advanced usage 2, you do not need call from  :rarm object, call :ik from *pr2* object
(send *pr2* :reset-pose)
(format t "before ik torso joint is ~A~%" (send *pr2* :torso :waist-z :joint-angle))
(send *pr2* :rarm :angle-vector #f(0 0 0 -90 -180 -90 180.0)) ;; use nice initial pose
(send *pr2* :inverse-kinematics (send *arrow* :copy-worldcoords)
      :move-target *cube-tip-pos*
      :link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent))
      ;; :use-torso t
      :rotation-axis t
      :debug-view :no-message) ;; 50.0
(send *irtviewer* :draw-objects)
(format t "after ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))

(send *pr2* :reset-pose)
(format t "before ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
(send *pr2* :rarm :angle-vector #f(0 0 0 -90 -180 -90 180.0)) ;; use nice initial pose
(send *pr2* :inverse-kinematics (send *arrow* :copy-worldcoords)
      :move-target *cube-tip-pos*
      :link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords :parent) (send *pr2* :rarm :root-link))
      ;; :use-torso nil
      :rotation-axis t
      :debug-view :no-message) ;; 50.0
(send *irtviewer* :draw-objects)
(format t "after ik torso joint is ~A~%" (send *pr2* :torso_lift_joint :joint-angle))
```